NVlabs/curobo #3: pull_repos.sh script is referenced but not included in the repo

blooop, 2023-10-27:
The comments in docker/build_dev_docker.sh says:

# Make sure you have pulled all required repos into pkgs folder (see pull_repos.sh script)

But I can not find that script in the repo.

As a result the build_dev_docker.sh script fails to find curobo and usd in the pkgs folder.

balakumar-s, 2023-10-27:
I will push a fix for this today. In the mean time you can comment out line 87: https://github.com/NVlabs/curobo/blob/b53d79ebb5a08096ed914b6fabc6c2d36d03d5e0/docker/x86.dockerfile#L87
and add the below instead:
```
RUN mkdir /pkgs && cd /pkgs && git clone https://github.com/NVlabs/curobo.git
```

blooop, 2023-10-27:
Thanks, I found a similar workaround so was able to get things running today.

balakumar-s, 2023-11-04:
Added new dockerfiles.

closed 2023-11-04 by balakumar-s