NVlabs/curobo #6: cannot run issac sim example

lianghongzhuo, 2023-10-30:
Hi, thank you for the package. The standalone example works well. But the issac sim examples fail to run on 2023 version. Can you support issac sim 2023?

balakumar-s, 2023-10-30:
Are you unable to install isaac sim 2022.2? It's available from the omniverse launcher.

balakumar-s, 2023-10-30:
You can have two versions of cuda installed at the same time by changing the cuda path. Instructions are here: https://curobo.org/source/getting_started/1_install_instructions.html#install-for-use-in-isaac-sim

We are looking at supporting isaac sim 2023, but don't have a timeline yet. 

We are also working on a docker container that will install isaac sim 2022.2 and the examples so that it's easier to run. This will be available end of this week.

lianghongzhuo, 2023-10-30:
Install two versions of cuda is not so easy, I used the above command and got cuda version 12.3 instead. (even if the downloaded file is called cuda 11.7)

lianghongzhuo, 2023-10-30:
I usually use conda two maintain different cuda versions, it would be great if using conda with issac sim is supported.

balakumar-s, 2023-10-30:
> Install two versions of cuda is not so easy, I used the above command and got cuda version 12.3 instead. (even if the downloaded file is called cuda 11.7)

Can you try ‘sudo apt install cuda-11.7’

lianghongzhuo, 2023-10-30:
```
sudo apt install cuda-11.7
[sudo] password for *: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package cuda-11.7
E: Couldn't find any package by glob 'cuda-11.7'

```
There is no such command.

balakumar-s, 2023-10-30:
I was mistaken, try this `sudo apt-get install cuda-11-7`

lianghongzhuo, 2023-10-30:
That works! Thank you!

closed 2023-10-30 by balakumar-s