NVlabs/curobo #7: how to run collision check IK for custom robots.

lianghongzhuo, 2023-10-30:
Hi, is it possible to infer collsion models directly from URDF?
From the IK example, the ik without collision model works well, but for the collision check, it seems not support collision mesh defined in the urdf? Could you support this?

blooop, 2023-10-30:
For the moment I think they only support collision spheres:

https://curobo.org/source/tutorials/1_robot_configuration.html

lianghongzhuo, 2023-10-30:
> For the moment I think they only support collision spheres:
> 
> https://curobo.org/source/tutorials/1_robot_configuration.html

Thank you for the link. That leads to my other question, it seems they only support issac sim 2022, which I only have Isaac 2023

balakumar-s, 2023-10-30:
There is a tutorial you can build a collision model from a urdf here: https://curobo.org/source/tutorials/1_robot_configuration.html

blooop, 2023-10-31:
This code could be modified to automatically create sphere models from meshes, but would requite a bit of integration work:  https://github.com/mlund/spheretree

balakumar-s, 2023-11-04:
We looked at similar methods, we could not get it to work robustly on arbitrary meshes. Hence, went the route of using Isaac Sim's sphere utility to build the sphere model, where users can visually inspect the spheres. I do agree that having an automatic way will make it easy to use even if it is not an exact method. 

I  will close this issue as there is no timeline on getting this working. Feel free to re-open the issue if there is an easy way to create spheres.

closed 2023-11-04 by balakumar-s