NVlabs/curobo #10: Isaac Sim Examples not running due to broken urdf import

patorrad, 2023-11-01:
Running Curobo examples in Isaac Sim 2022.2.1 result in broken simulation. Simulation runs but Franka Panda is not imported. For example, running motion_gen_reacher.py gives the following warnings and errors:
```
2023-10-31 05:16:49 [152,960ms] [Warning] [omni.isaac.urdf] Asset convert failed as asset file is broken.
2023-10-31 05:16:49 [152,960ms] [Warning] [omni.isaac.urdf] Prim /World/panda/panda_link0/visuals not created
2023-10-31 05:16:49 [152,971ms] [Warning] [omni.isaac.urdf] Asset convert failed as asset file is broken.
2023-10-31 05:16:49 [152,971ms] [Warning] [omni.isaac.urdf] Prim /World/panda/panda_link1/visuals not created
2023-10-31 05:16:49 [152,984ms] [Warning] [omni.isaac.urdf] Asset convert failed as asset file is broken.
2023-10-31 05:16:49 [152,984ms] [Warning] [omni.isaac.urdf] Prim /World/panda/panda_link2/visuals not created
2023-10-31 05:16:49 [152,998ms] [Warning] [omni.isaac.urdf] Asset convert failed as asset file is broken.
2023-10-31 05:16:49 [152,998ms] [Warning] [omni.isaac.urdf] Prim /World/panda/panda_link3/visuals not created
2023-10-31 05:16:49 [153,010ms] [Warning] [omni.isaac.urdf] Asset convert failed as asset file is broken.
2023-10-31 05:16:49 [153,010ms] [Warning] [omni.isaac.urdf] Prim /World/panda/panda_link4/visuals not created
2023-10-31 05:16:49 [153,022ms] [Warning] [omni.isaac.urdf] Asset convert failed as asset file is broken.
2023-10-31 05:16:49 [153,022ms] [Warning] [omni.isaac.urdf] Prim /World/panda/panda_link5/visuals not created
2023-10-31 05:16:49 [153,033ms] [Warning] [omni.isaac.urdf] Asset convert failed as asset file is broken.
2023-10-31 05:16:49 [153,033ms] [Warning] [omni.isaac.urdf] Prim /World/panda/panda_link6/visuals not created
2023-10-31 05:16:49 [153,045ms] [Warning] [omni.isaac.urdf] Asset convert failed as asset file is broken.
2023-10-31 05:16:49 [153,045ms] [Warning] [omni.isaac.urdf] Prim /World/panda/panda_link7/visuals not created
2023-10-31 05:16:49 [153,057ms] [Warning] [omni.isaac.urdf] Asset convert failed as asset file is broken.
2023-10-31 05:16:49 [153,057ms] [Warning] [omni.isaac.urdf] Prim /World/panda/panda_hand/visuals not created
2023-10-31 05:16:49 [153,070ms] [Warning] [omni.isaac.urdf] Asset convert failed as asset file is broken.
2023-10-31 05:16:49 [153,070ms] [Warning] [omni.isaac.urdf] Prim /World/panda/panda_leftfinger/visuals not created
2023-10-31 05:16:49 [153,082ms] [Warning] [omni.isaac.urdf] Asset convert failed as asset file is broken.
2023-10-31 05:16:49 [153,082ms] [Warning] [omni.isaac.urdf] Prim /World/panda/panda_rightfinger/visuals not created
2023-10-31 05:17:02 [166,582ms] [Error] [omni.physicsschema.plugin] Provided mesh geom with a PhysicsCollisionAPI does not have points, collision will not be created. Prim: /World/panda/panda_hand/collisions
2023-10-31 05:17:02 [166,582ms] [Error] [omni.physicsschema.plugin] Provided mesh geom with a PhysicsCollisionAPI does not have points, collision will not be created. Prim: /World/panda/panda_link7/collisions
2023-10-31 05:17:02 [166,583ms] [Error] [omni.physicsschema.plugin] Provided mesh geom with a PhysicsCollisionAPI does not have points, collision will not be created. Prim: /World/panda/panda_link6/collisions
2023-10-31 05:17:02 [166,583ms] [Error] [omni.physicsschema.plugin] Provided mesh geom with a PhysicsCollisionAPI does not have points, collision will not be created. Prim: /World/panda/panda_link5/collisions
2023-10-31 05:17:02 [166,583ms] [Error] [omni.physicsschema.plugin] Provided mesh geom with a PhysicsCollisionAPI does not have points, collision will not be created. Prim: /World/panda/panda_rightfinger/collisions
2023-10-31 05:17:02 [166,583ms] [Error] [omni.physicsschema.plugin] Provided mesh geom with a PhysicsCollisionAPI does not have points, collision will not be created. Prim: /World/panda/panda_link2/collisions
2023-10-31 05:17:02 [166,583ms] [Error] [omni.physicsschema.plugin] Provided mesh geom with a PhysicsCollisionAPI does not have points, collision will not be created. Prim: /World/panda/panda_leftfinger/collisions
2023-10-31 05:17:02 [166,583ms] [Error] [omni.physicsschema.plugin] Provided mesh geom with a PhysicsCollisionAPI does not have points, collision will not be created. Prim: /World/panda/panda_link3/collisions
2023-10-31 05:17:02 [166,583ms] [Error] [omni.physicsschema.plugin] Provided mesh geom with a PhysicsCollisionAPI does not have points, collision will not be created. Prim: /World/panda/panda_link1/collisions
2023-10-31 05:17:02 [166,583ms] [Error] [omni.physicsschema.plugin] Provided mesh geom with a PhysicsCollisionAPI does not have points, collision will not be created. Prim: /World/panda/panda_link4/collisions
2023-10-31 05:17:02 [166,583ms] [Error] [omni.physicsschema.plugin] Provided mesh geom with a PhysicsCollisionAPI does not have points, collision will not be created. Prim: /World/panda/panda_link0/collisions
```

balakumar-s, 2023-11-01:
Do you have git lfs installed? The urdf meshes are stored as lfs files. I will add a note in the installation instructions.

``` 
sudo apt install git-lfs 
cd curobo && git lfs pull
```

patorrad, 2023-11-01:
That was it. Thank you!

closed 2023-11-01 by patorrad