NVlabs/curobo #11: cannot run curobo with issac sim

lianghongzhuo, 2023-11-01:
In isaac sim 2023, the blow code will leads to a crash.
```
from omni.isaac.kit import SimulationApp
from curobo.util.usd_helper import UsdHelper
simulation_app = SimulationApp(
    {"headless": False, "width": 1920, "height": 1080}
)
```
If curobo is not imported, then the above code will run without an error.

balakumar-s, 2023-11-01:
Did you follow instructions to install curobo here: https://curobo.org/source/getting_started/1_install_instructions.html#install-for-use-in-isaac-sim

Note that you will need isaac sim 2022.2.1

lianghongzhuo, 2023-11-01:
As I mentioned, I am using isaac sim 2023
In the link it says:
> Isaac Sim 2023.1: Isaac Sim examples have not been tested on 2023.1 and may not work. However, python functions will work in Isaac Sim 2023.1. We found these additional packages to be missing in 2023.1, install them with omni_python -m pip install tomli wheel before installing curobo.

so I expect some functions can work on Isaac sim 2023, but the code as simple as above is not working. I would like to help port this to Isaac 2023 if you could give me some more instruction on the different between these two versions.

balakumar-s, 2023-11-01:
The isaac sim examples will not run but the python examples should. 

What's the error you are getting? Can you paste the text here?

lianghongzhuo, 2023-11-01:
```
[Warning] [omni.isaac.kit.simulation_app] Modules: ['pxr', 'pxr.Tf._tf', 'pxr.Tf', 'pxr.Gf._gf', 'pxr.Gf', 'pxr.Trace._trace', 'pxr.Trace', 'pxr.Work._work', 'pxr.Work', 'pxr.Plug._plug', 'pxr.Plug', 'pxr.Vt._vt', 'pxr.Vt', 'pxr.Ar._ar', 'pxr.Ar', 'pxr.Sdf._sdf', 'pxr.Sdf', 'pxr.Kind._kind', 'pxr.Kind', 'pxr.Pcp._pcp', 'pxr.Pcp', 'pxr.Usd._usd', 'pxr.Usd', 'pxr.UsdGeom._usdGeom', 'pxr.UsdGeom', 'pxr.Ndr._ndr', 'pxr.Ndr', 'pxr.Sdr._sdr', 'pxr.Sdr', 'pxr.UsdShade._usdShade', 'pxr.UsdShade', 'pxr.UsdPhysics._usdPhysics', 'pxr.UsdPhysics'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
[Info] [carb] Logging to file: /home//.local/share/ov/pkg/isaac_sim-2023.1.0/kit/logs/Kit/Isaac-Sim/2023.1/kit_20231101_154207.log
2023-11-01 14:42:07 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.055s] [ext: omni.kit.async_engine-0.0.0] startup
[0.307s] [ext: omni.activity.core-1.0.1] startup
[0.310s] [ext: omni.assets.plugins-0.0.0] startup
[0.311s] [ext: omni.stats-0.0.0] startup
[0.311s] [ext: omni.client-1.0.1] startup
[0.316s] [ext: omni.activity.profiler-1.0.2] startup
[0.318s] [ext: omni.gpu_foundation-0.0.0] startup
[0.324s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.325s] [ext: carb.windowing.plugins-1.0.0] startup
[0.330s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 535.86.05     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 3090          | Yes: 1 |     | 24576   MB | 10de      | 0          |
|     |                                  |        |     |            | 2204      | 51f8ac5c.. |
|     |                                  |        |     |            | 2d        |            |
|---------------------------------------------------------------------------------------------|
| 1   | NVIDIA GeForce RTX 3090          | Yes: 0 |     | 24576   MB | 10de      | 0          |
|     |                                  |        |     |            | 2204      | 3bb2c37c.. |
|     |                                  |        |     |            | 2e        |            |
|=============================================================================================|
| OS: 20.04.6 LTS (Focal Fossa) ubuntu, Version: 20.04.6, Kernel: 5.15.0-78-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: AMD Ryzen 9 5950X 16-Core Processor | Cores: 16 | Logical: 32
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 64215 | Free Memory: 55361
| Total Page/Swap (MB): 16383 | Free Page/Swap: 14856
|---------------------------------------------------------------------------------------------|
2023-11-01 14:42:08 [1,028ms] [Warning] [gpu.foundation.plugin] IOMMU is enabled. Found 30 items in /sys/kernel/iommu_groups/.
2023-11-01 14:42:08 [1,028ms] [Warning] [gpu.foundation.plugin] An input-output memory management unit (IOMMU) appears to be enabled on this system.
On bare-metal Linux systems, CUDA and the display driver do not support IOMMU-enabled PCIe peer to peer memory copy.
If you are on a bare-metal Linux system, please disable the IOMMU. Otherwise you risk image corruption and program instability.
This typically can be controlled via BIOS settings (Intel Virtualization Technology for Directed I/O (VT-d) or AMD I/O Virtualization Technology (AMD-Vi)) and kernel parameters (iommu, intel_iommu, amd_iommu).
Note that in virtual machines with GPU pass-through (vGPU) the IOMMU needs to be enabled.
Since we can not reliably detect whether this system is bare-metal or a virtual machine, we show this warning in any case when an IOMMU appears to be enabled.
2023-11-01 14:42:08 [1,028ms] [Warning] [gpu.foundation.plugin]  
2023-11-01 14:42:08 [1,028ms] [Warning] [gpu.foundation.plugin] -----------------------------------------------------------------------
2023-11-01 14:42:08 [1,028ms] [Warning] [gpu.foundation.plugin] !!!!! Local system validation failed! Incorrect configuration detected.
2023-11-01 14:42:08 [1,028ms] [Warning] [gpu.foundation.plugin]       Summary below. Details above.
2023-11-01 14:42:08 [1,028ms] [Warning] [gpu.foundation.plugin] -----------------------------------------------------------------------
2023-11-01 14:42:08 [1,028ms] [Warning] [gpu.foundation.plugin]  
2023-11-01 14:42:08 [1,028ms] [Warning] [gpu.foundation.plugin]  - ECC:                         OK
2023-11-01 14:42:08 [1,028ms] [Warning] [gpu.foundation.plugin]  - IOMMU:                       FAILED
2023-11-01 14:42:08 [1,028ms] [Warning] [gpu.foundation.plugin]  
2023-11-01 14:42:08 [1,028ms] [Warning] [gpu.foundation.plugin] -----------------------------------------------------------------------
[... cut ...]
[5.390s] [ext: omni.kit.manipulator.viewport-104.0.8] startup
[5.391s] [ext: omni.kit.viewport.window-105.1.9] startup
[5.517s] [ext: omni.physics.tensors-0.1.0] startup
[5.524s] [ext: omni.kit.widget.graph-1.8.4] startup
[5.528s] [ext: omni.kit.usd_undo-0.1.2] startup
[5.528s] [ext: omni.kit.viewport.utility-1.0.16] startup
[5.529s] [ext: omni.physx.tensors-0.1.0] startup
[5.543s] [ext: omni.kit.window.extensions-1.3.0] startup
[5.547s] [ext: omni.graph-1.106.0] startup
There was an error running python
```

balakumar-s, 2023-11-01:
Let's check if curobo works in your python environment first. 

Can you run `omni_python curobo/examples/kinematics_example.py`

lianghongzhuo, 2023-11-01:
> Let's check if curobo works in your python environment first.
> 
> Can you run `omni_python curobo/examples/kinematics_example.py`

this works

balakumar-s, 2023-11-01:
Now add the below lines to the start of `curobo/examples/ik_example.py`:

```
import torch

a = torch.zeros(4, device="cuda:0")

from omni.isaac.kit import SimulationApp

simulation_app = SimulationApp(
    {
        "headless": True,
        "width": "1920",
        "height": "1080",
    }
)
```

This will load isaac sim followed by running the IK example in python. This should print some success rate in the terminal and exit. You might get a seg fault after finishing IK, but that's fine. 

You will not get any GUI robot to work with as importing a robot in 2023 is not yet setup. This would only confirm that curobo's python examples run with isaac sim 2023.

lianghongzhuo, 2023-11-01:
It would be great if you could support the work with GUI in the 2023 version.

lianghongzhuo, 2023-11-03:
Before isacc sim 2023 is supported, I will use isacc sim 2022 instead. Thank you for the great work again!

balakumar-s, 2023-11-03:
I have isaac sim 2023 working, will push today and then close this issue.

balakumar-s, 2023-11-04:
Just pushed changes to support Isaac Sim 2023

closed 2023-11-04 by balakumar-s

lianghongzhuo, 2023-11-04:
Great! Thanks!