NVlabs/curobo #14: error when run "python3 -m pytest ."

Spphire, 2023-11-02:
self = Mesh(name='mesh_1', pose=[0.0, 2, 0.5, 0.043, -0.471, 0.284, 0.834], scale=[0.5, 0.5, 0.5], color=[0.38343539824735295...file_string=None, urdf_path=None, vertices=None, faces=None, vertex_colors=None, vertex_normals=None, face_colors=None)
process = True

    def get_trimesh_mesh(self, process: bool = True):
        # load mesh from filepath or verts and faces:
        if self.file_path is not None:
            m = trimesh.load(self.file_path, process=process, force="mesh")
            if isinstance(m, trimesh.Scene):
                m = m.dump(concatenate=True)

>           if isinstance(m.visual, trimesh.visual.texture.TextureVisuals):
E           AttributeError: 'list' object has no attribute 'visual'

src/curobo/geom/types.py:385: AttributeError
-------------------------------------------------- Captured log call ---------------------------------------------------
WARNING  curobo:logger.py:34 Creating new Mesh cache: 4
=================================================== warnings summary ===================================================
<unknown>:3
  <unknown>:3: DeprecationWarning: invalid escape sequence \p

tests/pose_test.py::test_pose_transform_point_grad
  /home/apri/anaconda3/envs/curobo/lib/python3.8/site-packages/torch/autograd/gradcheck.py:652: UserWarning: Input #0 requires gradient and is not a double precision floating point or complex. This check will likely fail if all the inputs are not of double precision floating point or complex.
    warnings.warn(

tests/pose_test.py::test_pose_transform_point_grad
  /home/apri/anaconda3/envs/curobo/lib/python3.8/site-packages/torch/autograd/gradcheck.py:652: UserWarning: Input #1 requires gradient and is not a double precision floating point or complex. This check will likely fail if all the inputs are not of double precision floating point or complex.
    warnings.warn(

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
=============================================== short test summary info ================================================
FAILED tests/warp_mesh_test.py::test_sdf_pose - AttributeError: 'list' object has no attribute 'visual'
FAILED tests/warp_mesh_test.py::test_swept_sdf_speed_pose - AttributeError: 'list' object has no attribute 'visual'
FAILED tests/warp_mesh_test.py::test_swept_sdf_pose - AttributeError: 'list' object has no attribute 'visual'
FAILED tests/world_config_test.py::test_world_modify - AttributeError: 'list' object has no attribute 'visual'
=========================== 4 failed, 125 passed, 2 skipped, 3 warnings in 65.17s (0:01:05) ============================

balakumar-s, 2023-11-02:
Can you check if the lfs files loaded correctly by running:

```
sudo apt install git-lfs 
cd curobo && git lfs pull
```

Spphire, 2023-11-02:
> Can you check if the lfs files loaded correctly by running:
> 
> ```
> sudo apt install git-lfs 
> cd curobo && git lfs pull
> ```

Error updating the git index: (206/206), 248 MB | 828 KB/s
error: src/curobo/content/assets/robot/franka_description/meshes/visual/link6.dae: cannot add to the index - missing --add option?
fatal: Unable to process path src/curobo/content/assets/robot/franka_description/meshes/visual/link6.dae

Errors logged to /home/apri/workspace/curobo/.git/lfs/logs/20231102T190326.726506706.log
Use `git lfs logs last` to view the log.

Spphire, 2023-11-02:
is it possible to publish a release pack?

Spphire, 2023-11-02:
> Error updating the git index: (206/206), 248 MB | 828 KB/s error: src/curobo/content/assets/robot/franka_description/meshes/visual/link6.dae: cannot add to the index - missing --add option? fatal: [...]

the log is below:

git-lfs/2.9.2 (GitHub; linux amd64; go 1.13.5)
git version 2.25.1

$ git-lfs pull
Error updating the git index:
error: src/curobo/content/assets/robot/franka_description/meshes/visual/link6.dae: cannot add to the index - missing --add option?
fatal: Unable to process path src/curobo/content/assets/robot/franka_description/meshes/visual/link6.dae

exit status 128

Current time in UTC:
2023-11-02 11:03:26

ENV:
LocalWorkingDir=/home/apri/workspace/curobo
LocalGitDir=/home/apri/workspace/curobo/.git
LocalGitStorageDir=/home/apri/workspace/curobo/.git
LocalMediaDir=/home/apri/workspace/curobo/.git/lfs/objects
LocalReferenceDirs=
TempDir=/home/apri/workspace/curobo/.git/lfs/tmp
ConcurrentTransfers=3
TusTransfers=false
BasicTransfersOnly=false
SkipDownloadErrors=false
FetchRecentAlways=false
FetchRecentRefsDays=7
FetchRecentCommitsDays=0
FetchRecentRefsIncludeRemotes=true
PruneOffsetDays=3
PruneVerifyRemoteAlways=false
PruneRemoteName=origin
LfsStorageDir=/home/apri/workspace/curobo/.git/lfs
AccessDownload=none
AccessUpload=none
DownloadTransfers=basic,lfs-standalone-file
UploadTransfers=basic,lfs-standalone-file
GIT_EXEC_PATH=/usr/lib/git-core

Client IP addresses:
172.28.90.87 fe80::215:5dff:fe28:3f26

balakumar-s, 2023-11-02:
Does this work? https://github.com/NVlabs/curobo/archive/refs/heads/main.zip

Spphire, 2023-11-03:
> Does this work? https://github.com/NVlabs/curobo/archive/refs/heads/main.zip

nope

balakumar-s, 2023-11-04:
Releases are now available: https://github.com/NVlabs/curobo/releases

Spphire, 2023-11-04:
I got it successfully with another PC, maybe its a network problem

Spphire, 2023-11-04:
> Releases are now available: https://github.com/NVlabs/curobo/releases

thanks!

closed 2023-11-04 by Spphire