NVlabs/curobo #16: Error when testing a new robot: extension class wrapper for base class pxrInternal_v0_22__pxrReserved__::UsdGeomBoundable has not been created yet

cedricgoubard, 2023-11-02:
Good morning,

I wanted to test curobo with a robot we are using (ur10e arm with a gripper on a mobile base). I followed the [tutorial](https://curobo.org/source/tutorials/1_robot_configuration.html) and everything worked until the last test, when I run `omni_python curobo/examples/isaac_sim/motion_gen_reacher.py --robot /{PATH}/ur5e.yml --visualize_spheres`

This might be related with the IsaacSim setup I am using:
- **Hardware**: Ubuntu20, driver 535, GeForce RTX 3060
- **In docker**: IsaacSim 2023.1.0 with CUDA 11.3 based on [the official image](https://github.com/NVIDIA-Omniverse/IsaacSim-dockerfiles/blob/main/Dockerfile.2023.1.0-ubuntu20.04), but with `nvidia/cudagl:$11.3.1-devel-ubuntu20.04` as a base image so that `nvcc` is accessible to build curobo and nvblox

As mentioned, everything builds correctly, and I am able to run IsaacSim to generate the USD and the spheres.
Here is [the yaml file](https://gist.github.com/cedricgoubard/11f6bfe9897196d0e095be995b7f1e10), and here is full log of the error I get:
<details>
<summary>Full Log</summary>

```
# /isaac-sim/python.sh curobo/examples/isaac_sim/motion_gen_reacher.py --robot /root/ros_ws/src/curobo_ros/cfg/amiga.yaml --visualize_spheres 

[Warning] [omni.isaac.kit.simulation_app] Modules: ['pxr', 'pxr.Tf._tf', 'pxr.Tf', 'pxr.Trace._trace', 'pxr.Trace', 'pxr.Work._work', 'pxr.Work', 'pxr.Plug._plug', 'pxr.Plug', 'pxr.Gf._gf', 'pxr.Gf', 'pxr.Vt._vt', 'pxr.Vt', 'pxr.Ar._ar', 'pxr.Ar', 'pxr.Kind._kind', 'pxr.Kind', 'pxr.Sdf._sdf', 'pxr.Sdf', 'pxr.Pcp._pcp', 'pxr.Pcp', 'pxr.Usd._usd', 'pxr.Usd', 'pxr.UsdGeom._usdGeom', 'pxr.UsdGeom', 'pxr.Ndr._ndr', 'pxr.Ndr', 'pxr.Sdr._sdr', 'pxr.Sdr', 'pxr.UsdShade._usdShade', 'pxr.UsdShade', 'pxr.UsdPhysics._usdPhysics', 'pxr.UsdPhysics'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/2023.1/kit_20231102_114656.log
2023-11-02 11:46:56 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.137s] [ext: omni.kit.async_engine-0.0.0] startup
[1.409s] [ext: omni.activity.core-1.0.1] startup
[1.418s] [ext: omni.assets.plugins-0.0.0] startup
[1.423s] [ext: omni.stats-0.0.0] startup
[1.426s] [ext: omni.client-1.0.1] startup
[1.453s] [ext: omni.activity.profiler-1.0.2] startup
[1.458s] [ext: omni.gpu_foundation-0.0.0] startup
[1.502s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[1.505s] [ext: carb.windowing.plugins-1.0.0] startup
2023-11-02 11:46:58 [2,064ms] [Warning] [carb.windowing-glfw.gamepad] Joystick with unknown remapping detected (will be ignored):  Corsair CORSAIR K60 RGB PRO Low Profile Mechanical Gaming Keyboard [030000001c1b0000ad1b000011010000]
[2.102s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 535.113.01    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 3060          | Yes: 0 |     | 12534   MB | 10de      | 0          |
|     |                                  |        |     |            | 2503      | f55b400e.. |
|     |                                  |        |     |            | 65        |            |
|=============================================================================================|
| OS: 20.04.4 LTS (Focal Fossa) ubuntu, Version: 20.04.4, Kernel: 5.15.0-87-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: Intel(R) Core(TM) i9-10920X CPU @ 3.50GHz | Cores: 12 | Logical: 24
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31768 | Free Memory: 15125
| Total Page/Swap (MB): 2047 | Free Page/Swap: 530
|---------------------------------------------------------------------------------------------|
[2.810s] [ext: omni.kit.pipapi-0.0.0] startup
[2.812s] [ext: omni.kit.pip_archive-0.0.0] startup
[2.812s] [ext: omni.pip.compute-1.2.0] startup
[2.813s] [ext: omni.pip.torch-2_0_1-2.0.2] startup
[2.845s] [ext: omni.pip.cloud-1.0.1] startup
[2.847s] [ext: omni.isaac.core_archive-2.2.1] startup
[2.847s] [ext: omni.kit.loop-isaac-1.0.2] startup
[2.849s] [ext: omni.kit.test-0.0.0] startup
[2.851s] [ext: omni.kit.telemetry-0.4.0] startup
[2.902s] [ext: omni.isaac.ml_archive-1.1.3] startup
[2.904s] [ext: omni.mtlx-0.1.0] startup
[2.906s] [ext: omni.usd.config-1.0.3] startup
[2.919s] [ext: omni.gpucompute.plugins-0.0.0] startup
[2.922s] [ext: omni.usd.libs-1.0.0] startup
[3.180s] [ext: omni.appwindow-1.1.5] startup
[3.496s] [ext: omni.kit.renderer.core-0.0.0] startup
[4.599s] [ext: omni.kit.renderer.capture-0.0.0] starWarning: in Tf_PyLoadScriptModule at line 123 of /buildAgent/work/ac88d7d902b57417/USD/pxr/base/tf/pyUtils.cpp -- Import failed for module 'pxr.UsdSkel'!
tup
[4.606s] [ext: omni.kit.renderer.imgui-0.0.0] startup
[4.823s] [ext: carb.audio-0.1.0] startup
[... 180 lines cut ...]

2023-11-02 11:47:04 [8,716ms] [Error] [omni.ext.plugin] [ext: omni.graph-1.106.0] Failed to startup python extension.
Fatal Python error: Segmentation fault

Current thread 0x00007f4e0a4cfb80 (most recent call first):
  File "/isaac-sim/kit/exts/omni.debugdraw/omni/debugdraw/__init__.py", line 6 in _get_interface
  File "/isaac-sim/kit/exts/omni.debugdraw/omni/debugdraw/__init__.py", line 10 in get_debug_draw_interface
  File "/isaac-sim/kit/exts/omni.debugdraw/omni/debugdraw/scripts/extension.py", line 7 in on_startup
  File "/isaac-sim/kit/kernel/py/omni/ext/_impl/_internal.py", line 166 in _startup_ext
  File "/isaac-sim/kit/kernel/py/omni/ext/_impl/_internal.py", line 224 in startup
  File "/isaac-sim/kit/kernel/py/omni/ext/_impl/_internal.py", line 328 in startup_extension
  File "PythonExtension.cpp::startup()", line 2 in <module>
  File "/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 303 in _start_app
  File "/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 192 in __init__
  File "/curobo/curobo/examples/isaac_sim/motion_gen_reacher_nvblox.py", line 52 in <module>

Extension modules: numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, torch._C, torch._C._fft, torch._C._linalg, torch._C._nested, torch._C._nn, torch._C._sparse, torch._C._special, xxhash._xxhash, scipy._lib._ccallback_c, scipy.sparse._sparsetools, _csparsetools, scipy.sparse._csparsetools, scipy.sparse.linalg._isolve._iterative, scipy.linalg._fblas, scipy.linalg._flapack, scipy.linalg._cythonized_array_utils, scipy.linalg._flinalg, scipy.linalg._solve_toeplitz, scipy.linalg._matfuncs_sqrtm_triu, scipy.linalg.cython_lapack, scipy.linalg.cython_blas, scipy.linalg._matfuncs_expm, scipy.linalg._decomp_update, scipy.sparse.linalg._dsolve._superlu, scipy.sparse.linalg._eigen.arpack._arpack, scipy.sparse.csgraph._tools, scipy.sparse.csgraph._shortest_path, scipy.sparse.csgraph._traversal, scipy.sparse.csgraph._min_spanning_tree, scipy.sparse.csgraph._flow, scipy.sparse.csgraph._matching, scipy.sparse.csgraph._reordering, scipy.spatial._ckdtree, scipy._lib.messagestream, scipy.spatial._qhull, scipy.spatial._voronoi, scipy.spatial._distance_wrap, scipy.spatial._hausdorff, scipy.special._ufuncs_cxx, scipy.special._ufuncs, scipy.special._specfun, scipy.special._comb, scipy.special._ellip_harm_2, scipy.spatial.transform._rotation, scipy.optimize._minpack2, scipy.optimize._group_columns, scipy.optimize._trlib._trlib, numpy.linalg.lapack_lite, scipy.optimize._lbfgsb, _moduleTNC, scipy.optimize._moduleTNC, scipy.optimize._cobyla, scipy.optimize._slsqp, scipy.optimize._minpack, scip [... cut ...] scipy.stats._qmc_cy, scipy.stats._rcont.rcont, pydantic.typing, pydantic.validators, pydantic.networks, pydantic.types, pydantic.json, pydantic.error_wrappers, pydantic.fields, pydantic.parse, pydantic.main, pydantic.dataclasses, pydantic.decorator, omni.mdl.pymdlsdk._pymdlsdk (total: 146)
/isaac-sim/python.sh: line 41:  1111 Segmentation fault      (core dumped) $python_exe "$@" $args
There was an error running python

```
</details>

Do you have any ideas where this might come from?
Thank you!

balakumar-s, 2023-11-02:
You would need IsaacSim 2022.2.1 to run the example. We are working on getting the examples to run on 2023.1.  I am glad that the sphere tutorial worked in 2023.

cedricgoubard, 2023-11-02:
Thank you for your reply; I am building the 2022 image now, will let you know if it fixes the issue.

Also, here are a couple of typos I noticed in [the tutorial](https://curobo.org/source/tutorials/1_robot_configuration.html):
- The yaml file given as an example at the end (just after "Your final configuration file should now look as below") is wrong, the collision spheres should be a dictionary but are currently a list
- In [this section](https://curobo.org/source/tutorials/1_robot_configuration.html#robot-collision-representation), `curobo/examples/isaac_sim/convert_urdf_to_usd.py` should be replaced by `curobo/examples/isaac_sim/utils/convert_urdf_to_usd.py`

cedricgoubard, 2023-11-02:
I got past the error with 2022.2.1 :partying_face:
I am now having a problem with my yaml file, but I should be able to fix this myself. I'll close the issue when I manage to finish the tutorial.

FYI, I have also identified the problem in 2023.1, it was caused by this line in my Dockerfile:
```dockerfile
RUN /isaac-sim/python.sh -m pip install tomli tqdm usd-core wheel cloudpickle farama-notifications jax-jumpy botocore==1.19.52  boto3==1.16.52
```
I had to add this to fix a `ModuleNotFoundError: No module named 'tomli'` error.

balakumar-s, 2023-11-03:
Thanks, yes you do have to install tomli. You actually shouldn't install `usd-core` as isaac sim comes with it's own `usd-core`. pip installing `usd-core` will break isaac sim.

cedricgoubard, 2023-11-03:
That is weird; I am still stuck even with version 2022.2.1, but I am not really getting an error message this time... The error is:
```bash
...
/isaac-sim/python.sh: line 41:    36 Killed                  $python_exe "$@" $args
There was an error running python
```

I am getting a warning about usd-core at the beginning though:
```bash
USDParser failed to import, install curobo with pip install .[usd] or pip install usd-core
```

To be clear, this is **without** installing it with `pip`. 

<details>
<summary>Full log</summary>

```bash
/isaac-sim/python.sh curobo/examples/isaac_sim/motion_gen_reacher.py --robot /root/ros_ws/src/curobo_ros/cfg/amiga.yaml --visualize_spheres

torchtyping could not be imported, falling back to basic types
USDParser failed to import, install curobo with pip install .[usd] or pip install usd-core
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.isaac.kit.app_framework'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/isaac-sim/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/isaac-sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1920', '--/app/renderer/resolution/height=1080', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/isaac-sim/exts', '--ext-folder', '/isaac-sim/apps', '--/physics/cudaDevice=0', '--portable', '--allow-root']
Passing the following args to the base kit application:  ['--robot', '/root/ros_ws/src/curobo_ros/cfg/amiga.yaml', '--visualize_spheres']
[Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/2022.2/kit_20231102_190109.log
2023-11-02 19:01:09 [35ms] [Warning] [omni.ext.plugin] [ext: omni.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/isaac-sim/exts/omni.sensors.nv.lidar' or '/isaac-sim/exts/omni.sensors.nv.lidar/config'
2023-11-02 19:01:09 [35ms] [Warning] [omni.ext.plugin] [ext: omni.sensors.nv.radar] Extensions config 'extension.toml' doesn't exist '/isaac-sim/exts/omni.sensors.nv.radar' or '/isaac-sim/exts/omni.sensors.nv.radar/config'
[0.236s] [ext: omni.stats-0.0.0] startup
[0.252s] [ext: omni.rtx.shadercache-1.0.0] startup
[0.258s] [ext: omni.assets.plugins-0.0.0] startup
[0.259s] [ext: omni.gpu_foundation-0.0.0] startup
[0.266s] [ext: carb.windowing.plugins-1.0.0] startup
2023-11-02 19:01:09 [716ms] [Warning] [carb.windowing-glfw.gamepad] Joystick with unknown remapping detected (will be ignored):  Corsair CORSAIR K60 RGB PRO Low Profile Mechanical Gaming Keyboard [030000001c1b0000ad1b000011010000]
[0.726s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 535.113.01    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 3060          | Yes: 0 |     | 12534   MB | 10de      | 0          |
|     |                                  |        |     |            | 2503      | e405bf50.. |
|=============================================================================================|
| OS: Linux gladOS, Version: 5.15.0-87-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: Intel(R) Core(TM) i9-10920X CPU @ 3.50GHz | Cores: Unknown | Logical: 24
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31768 | Free Memory: 19545
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
[1.295s] [ext: omni.kit.pipapi-0.0.0] startup
[1.309s] [ext: omni.kit.pip_archive-0.0.0] startup
[1.331s] [ext: omni.kit.loop-isaac-1.0.0] startup
[1.332s] [ext: omni.kit.async_engine-0.0.0] startup
[1.333s] [ext: omni.kit.test-0.0.0] startup
[1.418s] [ext: omni.usd.config-1.0.0] startup
[1.422s] [ext: omni.usd.libs-1.0.0] startup
[1.609s] [ext: omni.isaac.core_archive-2.0.1] startup
[1.612s] [ext: omni.pip.torch-1_13_1-0.1.4] startup
[1.614s] [ext: omni.isaac.ml_archive-1.1.0] startup
[1.677s] [ext: omni.client-0.1.1] startup
[1.689s] [ext: omni.appwindow-1.0.1] startup
[1.691s] [ext: omni.kit.renderer.core-0.0.0] startup
[1.813s] [ext: omni.kit.renderer.capture-0.0.0] startup
[1.818s] [ext: omni.kit.renderer.imgui-0.0.0] startup
[2.012s] [ext: carb.audio-0.1.0] startup
[... 351 lines cut ...]

ENV TORCH_CUDA_ARCH_LIST "8.6+PTX"

RUN git clone https://github.com/nvlabs/nvblox_torch.git
RUN cd nvblox_torch && \
[... cut ...]
############################################## curobo ##############################################
####################################################################################################
RUN git clone https://github.com/NVlabs/curobo.git 

ENV PATH="${PATH}:/opt/hpcx/ompi/bin"
ENV LD_LIBRARY_PATH="${LD_LIBRARY_PATH}:/opt/hpcx/ompi/lib"

ENV TORCH_CUDA_ARCH_LIST "8.6+PTX"
ENV LD_LIBRARY_PATH="/usr/local/lib:${LD_LIBRARY_PATH}"

RUN /isaac-sim/python.sh -m pip install wheel 

RUN cd curobo && /isaac-sim/python.sh -m pip install -e . --no-build-isolation 

ENTRYPOINT [ "" ]
```

balakumar-s, 2023-11-03:
I think this is because the path to your urdf is not getting loaded correctly in Isaac Sim's urdf importer. Is it possible for me to get access to the urdf folder and the configuration file you generated?

lianghongzhuo, 2023-11-03:
> Thanks, yes you do have to install tomli. You actually shouldn't install `usd-core` as isaac sim comes with it's own `usd-core`. pip installing `usd-core` will break isaac sim.

I also mislead by the terminal output and install usd-core mannully using pip. I have to remove isacc sim and reinstall it. Hope that isacc sim can provide a clean python binding that make isacc sim independent. Do you have the opputunity to forward this wish? Thank you!

lianghongzhuo, 2023-11-03:
The down side also includes, currently using isacc sim means binding the torch version ,python version to the isaac provided version.

balakumar-s, 2023-11-04:
> I also mislead by the terminal output and install usd-core mannully using pip. I have to remove isacc sim and reinstall it. Hope that isacc sim can provide a clean python binding that make isacc sim [...]

You can leave feedback for Isaac Sim here: https://forums.developer.nvidia.com/t/feedback-request-from-isaac-sim-users/267919

cedricgoubard, 2023-11-06:
Hi @balakumar-s,

I found the error thanks to your feedback; it was a typo in the YAML file (I had declared a sphere with a centre but no radius).

The simulation starts now, but the robot gets immediately stuck and I see a lot of PhysX errors like this one:

```
2023-11-06 14:59:30 [519,081ms] [Error] [omni.physx.plugin] PhysX error: Illegal BroadPhaseUpdateData 
, FILE /buildAgent/work/16dcef52b68a730f/source/lowlevelaabb/src/BpBroadPhaseABP.cpp, LINE 4024
2023-11-06 14:59:30 [519,083ms] [Warning] [omni.physx.plugin] Invalid PhysX transform detected for /World/amiga/castor_back_left.
2023-11-06 14:59:30 [519,083ms] [Warning] [omni.physx.plugin] Invalid PhysX transform detected for /World/amiga/castor_back_right.
2023-11-06 14:59:30 [519,083ms] [Warning] [omni.physx.plugin] Invalid PhysX transform detected for /World/amiga/castor_front_left.
2023-11-06 14:59:30 [519,083ms] [Warning] [omni.physx.plugin] Invalid PhysX transform detected for /World/amiga/castor_front_right.
2023-11-06 14:59:30 [519,083ms] [Warning] [omni.physx.plugin] Invalid PhysX transform detected for /World/amiga/back_wheel_left.
2023-11-06 14:59:30 [519,083ms] [Warning] [omni.physx.plugin] Invalid PhysX transform detected for /World/amiga/back_wheel_right.
2023-11-06 14:59:30 [519,083ms] [Warning] [omni.physx.plugin] Invalid PhysX transform detected for /World/amiga/front_wheel_left.
2023-11-06 14:59:30 [519,083ms] [Warning] [omni.physx.plugin] Invalid PhysX transform detected for /World/amiga/front_wheel_right.
```

The links mentioned are all the ones with passive joints, and they seem to be colliding with the ground in the simulation... any idea why?

[image]

balakumar-s, 2023-11-06:
The examples will put the base link at `0,0,0` so that is definitely a problem. You can add an extra link that will have an offset on z axis and make it the root. Have a 5cm gap between the floor and the base.

Regarding passive joints, I'm not sure if they work in sim. You can make them as foxed just to check if everything else works.

cedricgoubard, 2023-11-07:
Hi,

I have changed the passive joints to fixed, and I am spawning the robot above the floor (I added `position=np.array([0.4, 0.4, 0.4])` to [this line](https://github.com/NVlabs/curobo/blob/e0e804e9062da33c13aa262a1da253d364b8768a/examples/isaac_sim/motion_gen_reacher.py#L130)).

The arm is still not moving when I move the red cube - I am suppose to do something to make it move other than starting the simulation?

I think this could be linked to 2 things:
 - Wrong definition of `base_link` or `ee_link` in my config file, but I would expect some weird movement or error; otherwise, how can I check this?
 - I am getting some warnings like below; they seem to appear for each "virtual link" (links that have no physical properties)
 
```
2023-11-07 13:53:10 [33,058ms] [Warning] [omni.physx.plugin] The rigid body at /World/amiga/amiga_arm_base has a possibly invalid inertia tensor of {1.0, 1.0, 1.0} and a negative mass, small sphere approximated inertia was used. Either specify correct values in the mass properties, or add collider(s) to any shape(s) that you wish to automatically compute mass properties for. If you do not want the objects to collide, add colliders regardless then disable the 'enable collision' property.
```

Also, interestingly, the robot does not fall to the floor when I start the simulation...

Any advice would be greatly appreciated, thank you for your help :pray:

balakumar-s, 2023-11-07:
Regarding position offset, you need to convert the pose from the world frame to the position offset as it's not being done in the example. 

Regarding self collisions, did you add the colliding links to the ignore list?

Can you paste your t robot configuration file ?

cedricgoubard, 2023-11-07:
<details>
<summary>Here is the configuration file</summary>

```yaml
##
## Copyright (c) 2023 NVIDIA CORPORATION & AFFILIATES. All rights reserved.
##
## NVIDIA CORPORATION, its affiliates and licensors retain all intellectual
## property and proprietary rights in and to this material, related
## documentation and any modifications thereto. Any use, reproduction,
## disclosure or distribution of this material and related documentation
## without an express license agreement from NVIDIA CORPORATION or
## its affiliates is strictly prohibited.
##

robot_cfg:
  kinematics:
    usd_path: "/ros_ws/src/amiga_description/robots/usd/amiga.usd"
    usd_robot_root: "/robot"
    isaac_usd_path: ""
    usd_flip_joints: {}
    usd_flip_joint_limits: []

    urdf_path: "/ros_ws/src/amiga_description/robots/urdf/amiga_curobo.urdf"
    asset_root_path: "/ros_ws/src/"

    base_link: "base_link"
    ee_link: "amiga_gripper_palm"
    link_names: null
    lock_joints: null
    extra_links: null

    collision_link_names: null # List[str]
    collision_sphere_buffer: 0.005
    extra_collision_spheres: {}
    self_collision_ignore: 
      amiga_arm_base_link:
      - amiga_arm_shoulder_link
      - amiga_arm_upper_arm_link
      - amiga_arm_wrist_1_link
      - back_wheel_left
      - back_wheel_right
      - base_link
      - castor_back_left
      - castor_back_right
      - castor_front_left
      - castor_front_right
      - drive_wheel_left
      - drive_wheel_right
      - front_wheel_left
      - front_wheel_right
      amiga_arm_base_link_inertia:
      - amiga_arm_shoulder_link
      - amiga_arm_upper_arm_link
      - amiga_arm_wrist_1_link
      - back_wheel_left
      - back_wheel_right
      - base_link
      - castor_back_left
      - castor_back_right
      - castor_front_left
      - castor_front_right
      - drive_wheel_left
      - drive_wheel_right
      - front_wheel_left
      - front_wheel_right
      amiga_arm_flange:
      - amiga_arm_wrist_1_link
      - amiga_arm_wrist_2_link
      - amiga_arm_wrist_3_link
[... 743 lines cut ...]
          "radius": 0.034
        - "center": [0.009, -0.015, 0.032]
          "radius": 0.037
        - "center": [0.029, 0.031, -0.035]
          "radius": 0.029
        - "center": [0.034, 0.029, 0.036]
          "radius": 0.03
        - "center": [0.036, 0.005, 0.037]
          "radius": 0.034
        - "center": [-0.041, 0.036, -0.0]
          "radius": 0.025
    
    cspace:
      joint_names: ['amiga_arm_shoulder_pan_joint', 'amiga_arm_shoulder_lift_joint', 'amiga_arm_elbow_joint', 'amiga_arm_wrist_1_joint', 'amiga_arm_wrist_2_joint', 'amiga_arm_wrist_3_joint']
      retract_config:  [-0.00, 0.00, 0.00, 0.00, 0.00, 0.00]
      null_space_weight:  [1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
      cspace_distance_weight: [1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
      max_jerk: 500.0
      max_acceleration: 30.0
```
</details>

I generated the `self_collision_ignore` list from my previous MoveIt configuration file; it includes likes that are both always and never colliding.

In the picture, the links colliding are `amiga_arm_upper_arm_link` and `amiga_gripper_finger_[1 or 2]_link_[probably 2 and 3]`.

balakumar-s, 2023-11-07:
I missed a step in the tutorial. 

In your configuration file, can you add the names of all links you want to enable collision checking as a list to `collision_link_names` ?

cedricgoubard, 2023-11-07:
I have added them, and I also had to add `self_collision_buffer` as a dictionary with each link and `0.02` as value.

<details>
<summary>Updated file</summary>

```yaml
##
## Copyright (c) 2023 NVIDIA CORPORATION & AFFILIATES. All rights reserved.
##
## NVIDIA CORPORATION, its affiliates and licensors retain all intellectual
## property and proprietary rights in and to this material, related
## documentation and any modifications thereto. Any use, reproduction,
## disclosure or distribution of this material and related documentation
## without an express license agreement from NVIDIA CORPORATION or
## its affiliates is strictly prohibited.
##

robot_cfg:
  kinematics:
    usd_path: "/ros_ws/src/amiga_description/robots/usd/amiga.usd"
    usd_robot_root: "/robot"
    isaac_usd_path: ""
    usd_flip_joints: {}
    usd_flip_joint_limits: []

    urdf_path: "/ros_ws/src/amiga_description/robots/urdf/amiga_curobo.urdf"
    asset_root_path: "/ros_ws/src/"

    base_link: "base_link"
    ee_link: "amiga_gripper_palm"
    link_names: null
    lock_joints: null
    extra_links: null

    collision_link_names: 
      - amiga_arm_shoulder_link
      - amiga_arm_upper_arm_link
      - amiga_arm_forearm_link
      - amiga_arm_wrist_1_link
      - amiga_arm_wrist_2_link
      - amiga_arm_wrist_3_link
      - back_wheel_left
      - back_wheel_right
      - front_wheel_left
      - front_wheel_right
      - base_link
      - castor_back_left
      - castor_back_right
      - castor_front_left
      - castor_front_right
      - drive_wheel_left
      - drive_wheel_right
      - amiga_gripper_finger_1_link_0
      - amiga_gripper_finger_1_link_1
      - amiga_gripper_finger_1_link_2
      - amiga_gripper_finger_1_link_3
      - amiga_gripper_finger_2_link_0
      - amiga_gripper_finger_2_link_1
      - amiga_gripper_finger_2_link_2
      - amiga_gripper_finger_2_link_3
      - amiga_gripper_finger_middle_link_0
      - amiga_gripper_finger_middle_link_1
      - amiga_gripper_finger_middle_link_2
      - amiga_gripper_finger_middle_link_3
      - amiga_gripper_palm
      - zed2_camera_center
      - amiga_grip_camera_mount_base_link
      - amiga_arm_base_link_inertia

    self_collision_buffer:
[... 838 lines cut ...]
    main()
  File "curobo/examples/isaac_sim/motion_gen_reacher.py", line 163, in main
    velocity_scale=[0.25, 1, 1, 1, 1.0, 1.0, 1.0, 1.0, 1.0],
  File "/isaac-sim/kit/python/lib/python3.7/contextlib.py", line 74, in inner
    return func(*args, **kwds)
  File "/curobo/curobo/src/curobo/wrap/reacher/motion_gen.py", line 338, in load_from_robot_config
    robot_cfg = RobotConfig.from_dict(robot_cfg, tensor_args)
  File "/curobo/curobo/src/curobo/types/robot.py", line 49, in from_dict
    CudaRobotGeneratorConfig(**data_dict_in["kinematics"], tensor_args=tensor_args)
  File "/curobo/curobo/src/curobo/cuda_robot_model/cuda_robot_model.py", line 128, in from_config
    generator = CudaRobotGenerator(config)
  File "/curobo/curobo/src/curobo/cuda_robot_model/cuda_robot_generator.py", line 217, in __init__
    self.initialize_tensors()
  File "/isaac-sim/kit/python/lib/python3.7/contextlib.py", line 74, in inner
    return func(*args, **kwds)
  File "/curobo/curobo/src/curobo/cuda_robot_model/cuda_robot_generator.py", line 296, in initialize_tensors
    self.cspace.inplace_reindex(self.joint_names)
  File "/curobo/curobo/src/curobo/cuda_robot_model/types.py", line 128, in inplace_reindex
    new_index = [self.joint_names.index(j) for j in joint_names]
  File "/curobo/curobo/src/curobo/cuda_robot_model/types.py", line 128, in <listcomp>
    new_index = [self.joint_names.index(j) for j in joint_names]
ValueError: 'amiga_gripper_palm_finger_1_joint' is not in list
```

Should I add them? Would that not complicate the planning, since I am only interested in moving the arm?

balakumar-s, 2023-11-07:
Add all actuated joints to joint names and then you can add joints to lock as below in the configuration file:

https://github.com/NVlabs/curobo/blob/e0e804e9062da33c13aa262a1da253d364b8768a/src/curobo/content/configs/robot/franka.yml#L102

cedricgoubard, 2023-11-07:
Thank you; collisions spheres are now appearing, but the plan does not converge. I am guessing this is either because my starting configuration already has collisions, or because my parameters are wrong 

- Is there a simple way to see which links are colliding?
- Did I add too many spheres?
- Are my parameters ok?

```yaml
cspace:
      joint_names: 
        - amiga_arm_shoulder_pan_joint 
        - amiga_arm_shoulder_lift_joint 
        - amiga_arm_elbow_joint 
        - amiga_arm_wrist_1_joint 
        - amiga_arm_wrist_2_joint 
        - amiga_arm_wrist_3_joint 
        - amiga_gripper_palm_finger_1_joint 
        - amiga_gripper_finger_1_joint_1 
        - amiga_gripper_finger_1_joint_2 
        - amiga_gripper_finger_1_joint_3 
        - amiga_gripper_palm_finger_2_joint 
        - amiga_gripper_finger_2_joint_1 
        - amiga_gripper_finger_2_joint_2 
        - amiga_gripper_finger_2_joint_3 
        - amiga_gripper_finger_middle_joint_1 
        - amiga_gripper_finger_middle_joint_2 
        - amiga_gripper_finger_middle_joint_3

      retract_config:  [0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00]
      null_space_weight:  [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
      cspace_distance_weight: [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
      max_jerk: 500.0
      max_acceleration: 30.0
```

<details>
<summary>Full file</summary>

```yaml
##
## Copyright (c) 2023 NVIDIA CORPORATION & AFFILIATES. All rights reserved.
##
## NVIDIA CORPORATION, its affiliates and licensors retain all intellectual
## property and proprietary rights in and to this material, related
## documentation and any modifications thereto. Any use, reproduction,
## disclosure or distribution of this material and related documentation
## without an express license agreement from NVIDIA CORPORATION or
## its affiliates is strictly prohibited.
##

robot_cfg:
  kinematics:
    usd_path: "/ros_ws/src/amiga_description/robots/usd/amiga.usd"
    usd_robot_root: "/robot"
    isaac_usd_path: ""
    usd_flip_joints: {}
    usd_flip_joint_limits: []

    urdf_path: "/ros_ws/src/amiga_description/robots/urdf/amiga_curobo.urdf"
    asset_root_path: "/ros_ws/src/"

    base_link: "base_link"
    ee_link: "amiga_gripper_palm"
    link_names: null
    lock_joints:
      amiga_gripper_palm_finger_1_joint: 0.0 
      amiga_gripper_finger_1_joint_1: 0.0 
      amiga_gripper_finger_1_joint_2: 0.0 
      amiga_gripper_finger_1_joint_3: 0.0 
      amiga_gripper_palm_finger_2_joint: 0.0 
      amiga_gripper_finger_2_joint_1: 0.0 
[... 873 lines cut ...]
        - amiga_arm_wrist_2_joint 
        - amiga_arm_wrist_3_joint 
        - amiga_gripper_palm_finger_1_joint 
        - amiga_gripper_finger_1_joint_1 
        - amiga_gripper_finger_1_joint_2 
        - amiga_gripper_finger_1_joint_3 
        - amiga_gripper_palm_finger_2_joint 
        - amiga_gripper_finger_2_joint_1 
        - amiga_gripper_finger_2_joint_2 
        - amiga_gripper_finger_2_joint_3 
        - amiga_gripper_finger_middle_joint_1 
        - amiga_gripper_finger_middle_joint_2 
        - amiga_gripper_finger_middle_joint_3

      retract_config:  [0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00]
      null_space_weight:  [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
      cspace_distance_weight: [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
      max_jerk: 500.0
      max_acceleration: 30.0
```
</details>

<p align="center">
[image]
</p>

balakumar-s, 2023-11-07:
Does the IK example work? That would not consider the start state and would teleport the robot to a collision-free IK solution.

Also the self collision buffer can  all be set to 0.0. They can also be an empty dictionary.

cedricgoubard, 2023-11-07:
When I run `/isaac-sim/python.sh curobo/examples/isaac_sim/ik_reachability.py --robot /ros_ws/src/curobo_ros/cfg/amiga.yaml --visualize_spheres`, the arm does not move, but there is no error message.

You might want to fix the `yaml` file from the tutorial from `self_collision_buffer: null` to `self_collision_buffer: {}`; it raised an error before, complaining about `self_collision_buffer.keys()`.

balakumar-s, 2023-11-07:
And in the `ik_reachability` example, moving the target cube doesn't do anything?

I will add a check for empty dictionary in the code. For now `null` should work.

balakumar-s, 2023-11-07:
One way to debug if links are colliding:

1. Remove the links from `collision_link_names` tag and see when the plan starts to converge. Then you can debug just the link that is causing the plan to fail. 
2. You can also check MotionGenResult.invalid_query to see if the start state is in collision.

cedricgoubard, 2023-11-07:
I followed your advice, it looks like the problem happens because some links' spheres collide with the plane/table obstacle.

I manually added an offset to the robot, the spheres and the target in the example code, and it works! I will refine my spheres and start testing this on the real robot next, but I think I can close this issue now.

Thank you very much for all your help :pray:

closed 2023-11-07 by cedricgoubard

balakumar-s, 2023-11-07:
Your robot looks cool. Share any videos of your robot (sim or real) working with cuRobo here: https://github.com/NVlabs/curobo/discussions/categories/community-demos

Thanks for using the tutorial and catching bugs. I have updated the tutorial to incorporating the changes from your use.