NVlabs/curobo #17: Segmentation fault when pytest

wzf2022, 2023-11-04:
First thanks for your wonderful work!
I follow the instruction  and run `python3 -m pytest . ` and get Segmentation fault:

platform linux -- Python 3.8.16, pytest-7.4.3, pluggy-1.3.0
rootdir: /share/haoran/HRI/curobo
configfile: pyproject.toml
plugins: dash-2.11.1, typeguard-4.1.5, torchtyping-0.1.4
collecting 23 items                                                                                                                                          
Fatal Python error: Segmentation fault

Current thread 0x00007fa6eb87c4c0 (most recent call first):
  File "/share/haoran/HRI/curobo/src/curobo/types/math.py", line 207 in from_list
  File "/share/haoran/HRI/curobo/src/curobo/cuda_robot_model/kinematics_parser.py", line 43 in from_dict
  File "/share/haoran/HRI/curobo/src/curobo/cuda_robot_model/cuda_robot_generator.py", line 204 in __post_init__
  File "<string>", line 31 in __init__
  File "/share/haoran/HRI/curobo/src/curobo/types/robot.py", line 49 in from_dict
  File "/share/haoran/HRI/curobo/tests/ik_config_test.py", line 35 in ik_base_config
  File "/share/haoran/HRI/curobo/tests/ik_config_test.py", line 134 in <module>
  File "/home/haoran/anaconda3/envs/cvpr2023/lib/python3.8/site-packages/_pytest/assertion/rewrite.py", line 186 in exec_module
  File "<frozen importlib._bootstrap>", line 671 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 975 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 991 in _find_and_load
  File "<frozen importlib._bootstrap>", line 1014 in _gcd_import
  File "/home/haoran/anaconda3/envs/cvpr2023/lib/python3.8/importlib/__init__.py", line 127 in import_module
  File "/home/haoran/anaconda3/envs/cvpr2023/lib/python3.8/site-packages/_pytest/pathlib.py", line 567 in import_path
...

I wonder what might be the cause of this. My CUDA Version: 11.6.

balakumar-s, 2023-11-04:
How did install curobo? With `pip intall -e . --no-build-isolation` ?

balakumar-s, 2023-11-04:
Looking at your logs, I am not sure what's happening. 
1. You can try to run pytest with `python -m pytest tests/` to see if that works. 
2. If that doesn't work, you can try installing cuRobo on a base conda environment with only pytorch installed. 
3. If that also doesn't work, try with a pytorch docker using instructions here: https://curobo.org/source/getting_started/5_docker_development.html

balakumar-s, 2023-11-17:
@wzf2022 Does the issue still exist? If not, can you close this issue as I cannot reproduce it on my machine.

balakumar-s, 2023-11-21:
Closing as this cannot be reproduced.

closed 2023-11-21 by balakumar-s