NVlabs/curobo #19: wierd performance when running with Isaac-sim 2023.1.0

Spphire, 2023-11-04:
Isaac-sim 2023.1.0 with cu118
GPU 4090 Ubuntu22.04

https://github.com/NVlabs/curobo/assets/56157591/3c5a26ae-83ca-47b7-8d03-83ad920e709a

https://github.com/NVlabs/curobo/assets/56157591/76b80647-e281-48e3-84b4-97e98d2aeaf0

balakumar-s, 2023-11-04:
Can you run it now? I forgot to push a file.

Spphire, 2023-11-05:
i also switch to Isaac-sim 2022.2.1, but still same wierd

balakumar-s, 2023-11-05:
Can you explain what issue you are seeing?

Spphire, 2023-11-05:
when running "omni_python examples/isaac_sim/motion_gen_reacher.py"
start isaac-sim, the first frame the robot is set to a init pose, but then frozen

balakumar-s, 2023-11-05:
Can you click the red cube and the move it to make the robot track it.

balakumar-s, 2023-11-05:
An example is in first video in this section: https://curobo.org/source/getting_started/2b_isaacsim_examples.html#motion-generation

Spphire, 2023-11-05:
> Can you click the red cube and the move it to make the robot track it.

Oh i see, my fault

closed 2023-11-05 by Spphire