NVlabs/curobo #24: Question about the MotionGen

pangxincheng, 2023-11-05:
Hello, thanks for your great work.
When I try to use the MotionGen, I get some errors:
```
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1105 20:22:12.898658 31063 layer_cake_impl.h:32] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9TsdfVoxelEEE, voxel_size: 0.03, and memory_type: kDevice to LayerCake.
I1105 20:22:12.898676 31063 layer_cake_impl.h:32] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_10ColorVoxelEEE, voxel_size: 0.03, and memory_type: kDevice to LayerCake.
I1105 20:22:12.898679 31063 layer_cake_impl.h:32] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_14OccupancyVoxelEEE, voxel_size: 0.03, and memory_type: kDevice to LayerCake.
I1105 20:22:12.898682 31063 layer_cake_impl.h:32] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9EsdfVoxelEEE, voxel_size: 0.03, and memory_type: kDevice to LayerCake.
I1105 20:22:12.898684 31063 layer_cake_impl.h:32] Adding Layer with type: N6nvblox10BlockLayerINS_9MeshBlockEEE, voxel_size: 0.03, and memory_type: kDevice to LayerCake.
[1]    31063 segmentation fault (core dumped)
```
The code is as following:
```python
from curobo.types.base import TensorDeviceType
from curobo.geom.sdf.world import CollisionCheckerType
from curobo.wrap.reacher.motion_gen import MotionGen, MotionGenConfig

def main():

    robot_config = "franka.yml"
    tensor_args = TensorDeviceType()
    interpolation_dt = 0.001
    world_model = {
        "blox": {
            "world": {
                "pose": [0, 0, 0, 1, 0, 0, 0],
                "integrator_type": "occupancy",
                "voxel_size": 0.03,
            }   
        },
    }
    motion_gen_cfg = MotionGenConfig.load_from_robot_config(
        robot_config,
        world_model,
        tensor_args,
        collision_checker_type=CollisionCheckerType.BLOX,
        interpolation_dt=interpolation_dt,
        collision_activation_distance=0.4,
    )
    motion_gen = MotionGen(motion_gen_cfg)
    motion_gen.warmup(warmup_js_trajopt=False)

if __name__ == "__main__":
    main()
```
I try to get a full traceback, the problem lies in [L501](https://github.com/NVlabs/curobo/blob/21d434f608ab48371b952d138d7f5d1bb4e37495/src/curobo/util/sample_lib.py#L501)

balakumar-s, 2023-11-06:
If you are trying to use the motion generation on obstacles that are represented as meshes or primitives, you can use `CollisionCheckerType.Mesh` instead of `CollisionCheckerType.BLOX`. Note that when changing to a different collision checker type, change the world config to `collision_table.yml` to load an appropriate set of obstacles.

If you are trying to use obstacles from a depth camera, try one of the below:
1. you might need to compile glog from source following instructions from : https://curobo.org/source/getting_started/1_install_instructions.html#installing-nvblox-with-isaac-sim
2. If that doesn't work, you can follow instructions from dockerfile here: https://github.com/NVlabs/curobo/blob/21d434f608ab48371b952d138d7f5d1bb4e37495/docker/isaac_sim.dockerfile#L186

We found nvblox to work reliably inside the provided dockerfiles. When using outside docker, you might have to make sure the glog library is correctly linked to avoid getting a segfault. 

We found compiling nvblox with statically linked glog following these instructions (https://github.com/NVlabs/curobo/blob/21d434f608ab48371b952d138d7f5d1bb4e37495/docker/isaac_sim.dockerfile#L186) worked well when running with isaac sim.

mfinean, 2023-11-06:
I have also encountered the same problem described above and it happens when the warmup() is called. You can remove warmup() and continue to use the motion_gen planning but without the speedup that the warmup provides. I'll update this thread if I get to the bottom of this

balakumar-s, 2023-11-17:
We have a fix for this issue, will push an update before Monday.

balakumar-s, 2023-12-15:
We just pushed a new release that works better with nvblox. Can you try it now and close this issue if it's resolved?

pangxincheng, 2023-12-15:
> We just pushed a new release that works better with nvblox. Can you try it now and close this issue if it's resolved?
Thanks! @balakumar-s 
I pulled the new code and reinstalled it, but it seems that the issue has not been resolved.
But this issue did not occur when running in the docker you provided(Even in the old version curobo). 
Is this an issue with the PyTorch version? The pytorch version in the docker is `2.1.0a0+29c30b1`, but in my conda environment, the pytorch version is `1.13.1+cu117`
I will try installing a new version of pytorch later to see if the problem is resolved.😄

balakumar-s, 2023-12-15:
Did you also remove nvblox (uninstall any files from system path as well) and reinstall it with the new instructions?

pangxincheng, 2023-12-16:
> Did you also remove nvblox (uninstall any files from system path as well) and reinstall it with the new instructions?

Thank you. I have reinstalled Curobo according to the latest [installation documentation](https://curobo.org/source/getting_started/1_install_instructions.html) and it can now run successfully. I will close this issue.😄

closed 2023-12-16 by pangxincheng