NVlabs/curobo #30: No GUI when Installing the Python version

ismarou, 2023-11-06:
Hello,

I followed the installation instructions and installed the Python version of the library.

However, when I run the examples, I don't have access to some visualization of the planned trajectories or any kind of GUI. Should this be the case? How do I see the planned trajectories if so? Is this the only way to have Isaac Sim installed?

Best,
Isidoros

balakumar-s, 2023-11-06:
Yes, there is no GUI for the python version. You have to link it to your own simulation pipeline or use isaac sim examples for a GUI.

ismarou, 2023-11-06:
What do you mean link it with your own simulation pipeline? From what I understand, only IsaacSim is supported, no?

balakumar-s, 2023-11-06:
It's a python library, so it could integrated with any simulator. We only integrate with Isaac Sim.