NVlabs/curobo #33: isaac-sim crashed when opening usd file for configuring new robot

denzelzheng, 2023-11-07:
isaac-sim 2022.2.1 pass all tests
i tried to configuring new robot
urdf file:
[flexiv.json](https://github.com/NVlabs/curobo/files/13281223/flexiv.json)

and successfully generate a usd file:
[flexiv_usd.json](https://github.com/NVlabs/curobo/files/13281237/flexiv_usd.json)

but when opening usd by isaac-sim, it crashed

balakumar-s, 2023-11-07:
What is printed in the terminal?

Also is this urdf available online with the meshes?

balakumar-s, 2023-11-07:
Also I noticed that your usd is being names with `.json` while it should be `flexiv_usd.usd`. Can you try renaming it to `.usd` to see if that's the issue?

denzelzheng, 2023-11-08:
> Also I noticed that your usd is being names with `.json` while it should be `flexiv_usd.usd`. Can you try renaming it to `.usd` to see if that's the issue?

in issue it cant upload usd file, so i rename it to json

balakumar-s, 2023-11-08:
The usd file you shared loads on my machine: 
[image]

Are you not able to drag and drop the usd file into Isaac Sim?

denzelzheng, 2023-11-09:
> The usd file you shared loads on my machine on Isaac Sim 2023.1.0 : [image]
> 
> Are you not able to drag and drop the usd file into Isaac Sim?
> 
> The file did not open for me in Isaac Sim 2022.2.1. Can you update to Isaac Sim 2023.1.0 and try?

yeah cant open it by 2022, ill try 2023

balakumar-s, 2023-11-17:
Closing issue as it's fixed in 2023. Re-open if you find it not working in 2023

closed 2023-11-17 by balakumar-s