NVlabs/curobo #41: Error  when run "motion_gen_reacher.py"

peter05010402, 2023-11-15:
Hi,
my env info:
1. cuRobo installation mode:     python, isaac sim
2. python version:                      3.7.13
3. Isaac Sim version (if using):  2022.2.1

------
Issue Details:
When run motion_gen_reacher.py on my robot without griper collision spheres,  It's workable.
But with griper collision spheres, it got Error:
[image]

peter05010402, 2023-11-15:
Is this lead to the error:
[image]

balakumar-s, 2023-11-17:
Does the standard examples work (with the franka.yml robot)  in the same python environment?
I am asking because I see `Failed to load CUDA module` in your log which is not related to  a robot configuration.

peter05010402, 2023-11-20:
yes, standard example work.   
It failed ,because of wrong robot configuration!
Thank you for ur reply

closed 2023-11-20 by peter05010402