NVlabs/curobo #43: Invalid requirements: '.[dev,' when running build_docker.sh for isaac_sim_2022.2.1

RishavGH, 2023-11-16:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]):
2. python version:
3. Isaac Sim version (if using): 

# Installation
Docker Isaac Sim

# Python version
[Did not change the default provided by isaac_sim.dockerfile]

# Isaac Sim Version
2022.2.1

------
Issue Details

From /docker/, I ran the command `bash build_docker.sh isaac_sim_2022.2.1`. Here is the terminal output
```
Building Isaac Sim docker
isaac_sim.dockerfile
[+] Building 1.4s (23/35)                                       docker:default
 => [internal] load .dockerignore                                         0.0s
 => => transferring context: 60B                                          0.0s
 => [internal] load build definition from isaac_sim.dockerfile            0.0s
 => => transferring dockerfile: 8.50kB                                    0.0s
 => [internal] load metadata for [nvcr.io/nvidia/cudagl:11.4.2-devel-ubun](http://nvcr.io/nvidia/cudagl:11.4.2-devel-ubun)  0.5s
 => [internal] load metadata for [nvcr.io/nvidia/isaac-sim:2022.2.1](http://nvcr.io/nvidia/isaac-sim:2022.2.1)        0.5s
 => [auth] nvidia/cudagl:pull,push token for [nvcr.io](http://nvcr.io/)                      0.0s
 => [auth] nvidia/isaac-sim:pull,push token for [nvcr.io](http://nvcr.io/)                   0.0s
 => [stage-1  1/28] FROM [nvcr.io/nvidia/cudagl:11.4.2-devel-ubuntu20.04@](http://nvcr.io/nvidia/cudagl:11.4.2-devel-ubuntu20.04@)  0.0s
 => [isaac-sim 1/1] FROM [nvcr.io/nvidia/isaac-sim:2022.2.1@sha256:48ccef](http://nvcr.io/nvidia/isaac-sim:2022.2.1@sha256:48ccef)  0.0s
 => CACHED [stage-1  2/28] RUN echo 'debconf debconf/frontend select Non  0.0s
 => CACHED [stage-1  3/28] RUN apt-get update && apt-get install -y   tz  0.0s
 => CACHED [stage-1  4/28] RUN apt-get update && apt-get install -y --no  0.0s
 => CACHED [stage-1  5/28] RUN wget -q --show-progress     --progress=ba  0.0s
 => CACHED [stage-1  6/28] COPY --from=isaac-sim /isaac-sim /isaac-sim    0.0s
 => CACHED [stage-1  7/28] RUN mkdir -p /root/.nvidia-omniverse/config    0.0s
 => CACHED [stage-1  8/28] COPY --from=isaac-sim /root/.nvidia-omniverse  0.0s
 => CACHED [stage-1  9/28] COPY --from=isaac-sim /etc/vulkan/icd.d/nvidi  0.0s
 => CACHED [stage-1 10/28] COPY --from=isaac-sim /etc/vulkan/icd.d/nvidi  0.0s
 => CACHED [stage-1 11/28] WORKDIR /isaac-sim                             0.0s
 => CACHED [stage-1 12/28] RUN echo "alias omni_python='/isaac-sim/pytho  0.0s
 => CACHED [stage-1 13/28] RUN /isaac-sim/python.sh -m pip install "robo  0.0s
 => CACHED [stage-1 14/28] RUN mkdir /pkgs && cd /pkgs && git clone http  0.0s
 => CACHED [stage-1 15/28] RUN /isaac-sim/python.sh -m pip install ninja  0.0s
 => ERROR [stage-1 16/28] RUN cd /pkgs/curobo && /isaac-sim/python.sh -m  0.8s
------
 > [stage-1 16/28] RUN cd /pkgs/curobo && /isaac-sim/python.sh -m pip install .[dev, isaac_sim] --no-build-isolation:
0.765 ERROR: Invalid requirement: '.[dev,'
0.769 WARNING: You are using pip version 21.2.1+nv1; however, version 23.3.1 is available.
0.769 You should consider upgrading via the '/isaac-sim/kit/python/bin/python3 -m pip install --upgrade pip' command.
0.831 There was an error running python
------
isaac_sim.dockerfile:174
--------------------
 172 |
 173 |
 174 | >>> RUN cd /pkgs/curobo && $omni_python -m pip install .[dev, isaac_sim] --no-build-isolation
 175 |
 176 |     # Optionally install nvblox:
--------------------
ERROR: failed to solve: process "/bin/sh -c cd /pkgs/curobo && $omni_python -m pip install .[dev, isaac_sim] --no-build-isolation" did not complete successfully: exit code: 1
```
 As you can see it's failing on [this](https://github.com/NVlabs/curobo/blob/e0e804e9062da33c13aa262a1da253d364b8768a/docker/isaac_sim.dockerfile#L174) line of isaac_sim.dockerfile.

balakumar-s, 2023-11-16:
In this line: https://github.com/NVlabs/curobo/blob/e0e804e9062da33c13aa262a1da253d364b8768a/docker/isaac_sim.dockerfile#L174

Change `[dev, isaac_sim]` to `[dev,isaac_sim`. 

I will push a fix for this soon.

RishavGH, 2023-11-16:
Thank you @balakumar-s . That worked. Also, `.[dev, isaac-sim]` works as well.

blooop, 2023-11-16:
I faced the same issue here:
https://github.com/NVlabs/curobo/pull/27

I fixed by removing the dev requirement, but will update with your improved fix.

balakumar-s, 2023-11-17:
#27 fixes this issue. Thanks @blooop!

closed 2023-11-17 by balakumar-s