NVlabs/curobo #44: ModuleNotFoundError: No module named 'pxr'

simanxu, 2023-11-16:
$ omni_python torch_layers_example.py 

----output-----
Warning: running in conda env, please deactivate before executing this script
If conda is desired please source setup_conda_env.sh in your python 3.7 conda env and run python normally
Traceback (most recent call last):
  File "torch_layers_example.py", line 28, in <module>
    from curobo.util.usd_helper import UsdHelper
  File "/home/xu2/Work/curobo/src/curobo/util/usd_helper.py", line 19, in <module>
    from pxr import Gf, Sdf, Usd, UsdGeom, UsdPhysics, UsdShade
ModuleNotFoundError: No module named 'pxr'
There was an error running python
----output-----

balakumar-s, 2023-11-16:
When running python examples in an isaac sim environment, you need to add the below to the start of the file to load python packages from isaac sim:

```

import torch

a = torch.zeros(4, device="cuda:0")

from omni.isaac.kit import SimulationApp

simulation_app = SimulationApp(
    {
        "headless": True,
        "width": "1920",
        "height": "1080",
    }
)

```

simanxu, 2023-11-17:
thanks for your reply, I add the follow code at the start of torch_layers_example.py and it becomes
`import torch

a = torch.zeros(4, device="cuda:0")

from omni.isaac.kit import SimulationApp

simulation_app = SimulationApp(
    {
        "headless": True,
        "width": "1920",
        "height": "1080",
    }
)

# Standard Library
import uuid
from typing import Optional

# Third Party
import numpy as np
import torch
from torch import nn
from torch.utils.tensorboard import SummaryWriter
from tqdm import tqdm
...
`
omni_python examples/torch_layers_example.py get another error as
`
...
[4.395s] Simulation App Starting
[7.738s] app ready
[7.857s] RTX ready
[7.965s] Simulation App Startup Complete
t: 73.81147003173828: 100%|██████████████████████████████████████████████████████████| 500/500 [00:54<00:00,  9.13it/s]
2023-11-17 05:20:05 [83,963ms] [Warning] [carb.audio.context] 1 contexts were leaked
/home/xu2/.local/share/ov/pkg/isaac_sim-2022.2.1/python.sh: line 41: 184761 Segmentation fault      (core dumped) $python_exe "$@" $args
There was an error running python
`

balakumar-s, 2023-11-17:
That last segfault happens after completion of the script so that's fine.

simanxu, 2023-11-17:
> That last segfault happens after completion of the script so that's fine.

Thank you very much. Just to confirm whether this means that the neural network training demo has been completed. Please forgive me if it is a stupid question.

balakumar-s, 2023-11-17:
Yes it's completed. This example is only to show how to use with a neural network so don't expect the trained network to be very good.

simanxu, 2023-11-17:
> Yes it's completed. This example is only to show how to use with a neural network so don't expect the trained network to be very good.

Problem solved, thanks again, I will close this comment right now!

closed 2023-11-17 by simanxu

balakumar-s, 2023-11-17:
Feel free to ask questions here: https://github.com/NVlabs/curobo/discussions/categories/software-q-a