NVlabs/curobo #50: Sphere model is fitting joints instead of links (for custom robot)

jayaramreddy10, 2023-11-22:
Hi, I was trying to build sphere model for xarm7 (which is not provided by default in curobo). Spheres are fitting joints instead of links. 

URDF : https://drive.google.com/file/d/15mpdO20_NXHT8zGIXS_X6gOnTpc5biYx/view?usp=sharing
USD :  https://drive.google.com/file/d/179Jmx-5bW3uxbX5I8bk_qkOby5i7vSJ7/view?usp=drive_link

Screenshots of the sphere model for first 3 links. Could you please help me understand why is it modelling that way?

[image]
[image]
[image]

Let me know if you need further information.

balakumar-s, 2023-11-27:
Not exactly sure what's happening, but you can manually move the spheres after they are generated following this video: https://curobo.org/source/tutorials/1_robot_configuration.html#robot-collision-representation

One thing to try would be to select `collisions` in `Select Mesh` drop down.

jayaramreddy10, 2023-11-28:
Thank you, will try that.