NVlabs/curobo #53: Install failed on Win11

Juzhan, 2023-11-27:
1. cuRobo installation mode: python
2. python version: python 3.7.16
-- setuptool: 65.0.0
-- cuda 11.7
-- pytorch: 1.13.1
3. Isaac Sim version (if using): I have not tried Isaac sim yet.

### Issue Details

I try to install curobo on my win11 PC, with RTX 3090, but it failed when I executed the command:
```
pip install -e  . --no-build-isolation
```

I also follow the suggestion printed from the log, run the pip install without edit mode, but it still not work:
```
pip install . --no-build-isolation
```
Is it possible to install the curobo on a Windows PC? Thanks in advance!

Here is the log output: 
[error_log.txt](https://github.com/NVlabs/curobo/files/13470614/error_log.txt)

balakumar-s, 2023-11-27:
Windows is not supported but it might work in wsl as mentioned here: https://github.com/NVlabs/curobo/discussions/37#discussioncomment-7571229

Juzhan, 2023-11-27:
Thanks for your quick reply! I will have a try in WSL.

closed 2023-11-27 by Juzhan