NVlabs/curobo #58: Undefined symbol issue when running nvblox related examples

jayaramreddy10, 2023-11-28:
Hi, examples given in isaac-sim folder are working except nvblox ones. I have followed instructions given in curobo documentation to install nvblox correctly in isaac-sim and was able to import nvblox_torch. But when running nvblox related examples, I am getting below issue. Can you help me resolve this? Is this some version mismatch issue?

Command: omni_python mpc_nvblox_example.py
Issue: 
OSError: /usr/local/lib/libglog.so.0: undefined symbol: _ZN6google14FlagRegistererC1INSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEPKcS9_S9_PT_SB_
2023-11-28 18:23:36 [29,361ms] [Warning] [carb.audio.context] 1 contexts were leaked
/home2/$USER$/.local/share/ov/pkg/isaac_sim-2022.2.1/python.sh: line 41: 3414963 Segmentation fault      (core dumped) $python_exe "$@" $args
There was an error running python

balakumar-s, 2023-12-15:
We pushed a new release that should fix this issue. Follow instructions here to install: https://curobo.org/source/getting_started/1_install_instructions.html

balakumar-s, 2023-12-22:
Closing this due to inactivity. Feel free to open this issue again if the latest release didn't fix it.

closed 2023-12-22 by balakumar-s