NVlabs/curobo #61: Nvidia WARP does not build on Jetson orin

nivetaiyer, 2023-12-02:
1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): Python
2. python version: 3.8
3. Isaac Sim version (if using): 

------
Issue Details
Curobo builds fine, but unit tests fail because it can't find warp.so. I'm trying to build WARP on arm64 Ubuntu (jetson orin) but it fails because llvm is not installed. Can curobo work without warp? If not, what are the steps to successfully compile warp on arm64?

balakumar-s, 2023-12-02:
cuRobo requires warp. We have only been running cuRobo within a docker on the jetson. You can follow instructions to build a docker for jetson here: https://curobo.org/source/getting_started/5_docker_development.html

Compiling warp on jetson can be done following: https://curobo.org/source/getting_started/5_docker_development.html#build-warp-for-nvidia-jetson

nivetaiyer, 2023-12-02:
When I try to build warp, it only builds for x86 on a Linux platform - https://github.com/NVIDIA/warp/blob/main/build_llvm.py#L30 

The instructions for building warp here  https://curobo.org/source/getting_started/5_docker_development.html#build-warp-for-nvidia-jetson don't actually seem to work.

balakumar-s, 2023-12-02:
You need to run all docker build commands on a jetson to build an image for the jetson.

> The instructions for building warp here https://curobo.org/source/getting_started/5_docker_development.html#build-warp-for-nvidia-jetson don't actually seem to work.

What issue are you facing in building warp on a jetson? 
You should install cuda 11.5+ on a jetson device (outside docker) and then compile warp (outside docker).

nivetaiyer, 2023-12-02:
Are the instructions here https://curobo.org/source/getting_started/5_docker_development.html#build-warp-for-nvidia-jetson for aarch64 or x86?

balakumar-s, 2023-12-02:
They are for aarch64.

balakumar-s, 2023-12-02:
Also, can you check out an older branch of warp. I tested the instructions with `v0.10.0` and `v0.9.0`

nivetaiyer, 2023-12-05:
Thanks! I managed to build warp but the tests fail with this error -

```
============================= test session starts ==============================
 platform linux -- Python 3.8.10, pytest-7.4.3, pluggy-1.3.0
 rootdir: /home/user/curobo
 configfile: pyproject.toml
 plugins: typeguard-4.1.5, torchtyping-0.1.4
 collected 121 items / 2 errors / 2 skipped
 ==================================== ERRORS ====================================
 ___________________ ERROR collecting tests/ik_config_test.py ___________________
 ../.local/lib/python3.8/site-packages/warp/codegen.py:620: in build
     adj.eval(adj.tree.body[0])
 ../.local/lib/python3.8/site-packages/warp/codegen.py:1957: in eval
     return emit_node(adj, node)
 ../.local/lib/python3.8/site-packages/warp/codegen.py:1158: in emit_FunctionDef
     adj.eval(f)
 ../.local/lib/python3.8/site-packages/warp/codegen.py:1957: in eval
     return emit_node(adj, node)
 ../.local/lib/python3.8/site-packages/warp/codegen.py:1786: in emit_Assign
     value = adj.eval(node.value)
 ../.local/lib/python3.8/site-packages/warp/codegen.py:1957: in eval
     return emit_node(adj, node)
 ../.local/lib/python3.8/site-packages/warp/codegen.py:1674: in emit_Call
     out = adj.add_call(func=func, args=args, kwds=kwds, templates=templates, min_outputs=min_outputs)
 ../.local/lib/python3.8/site-packages/warp/codegen.py:895: in add_call
     func = adj.resolve_func(func, args, min_outputs, templates, kwds)
 ../.local/lib/python3.8/site-packages/warp/codegen.py:890: in resolve_func
     raise WarpCodegenError(
 E   warp.codegen.WarpCodegenError: Couldn't find function overload for 'index' that matched inputs with types: [warp.types.quatf, int32]
 During handling of the above exception, another exception occurred:
 ../.local/lib/python3.8/site-packages/_pytest/runner.py:341: in from_call
     result: Optional[TResult] = func()
 ../.local/lib/python3.8/site-packages/_pytest/runner.py:372: in <lambda>
     call = CallInfo.from_call(lambda: list(collector.collect()), "collect")
 ../.local/lib/python3.8/site-packages/_pytest/python.py:531: in collect
     self._inject_setup_module_fixture()
 ../.local/lib/python3.8/site-packages/_pytest/python.py:545: in _inject_setup_module_fixture
     self.obj, ("setUpModule", "setup_module")
 ../.local/lib/python3.8/site-packages/_pytest/python.py:310: in obj
     self._obj = obj = self._getobj()
 ../.local/lib/python3.8/site-packages/_pytest/python.py:528: in _getobj
     return self._importtestmodule()
 ../.local/lib/python3.8/site-packages/_pytest/python.py:617: in _importtestmodule
     mod = import_path(self.path, mode=importmode, root=self.config.rootpath)
 ../.local/lib/python3.8/site-packages/_pytest/pathlib.py:567: in import_path
     importlib.import_module(module_name)
 /usr/lib/python3.8/importlib/__init__.py:127: in import_module
     return _bootstrap._gcd_import(name[level:], package, level)
 <frozen importlib._bootstrap>:1014: in _gcd_import
     ???
 <frozen importlib._bootstrap>:991: in _find_and_load
     ???
 <frozen importlib._bootstrap>:975: in _find_and_load_unlocked
     ???
 <frozen importlib._bootstrap>:671: in _load_unlocked
     ???
 ../.local/lib/python3.8/site-packages/_pytest/assertion/rewrite.py:186: in exec_module
     exec(co, module.__dict__)
 tests/ik_config_test.py:134: in <module>
     (ik_base_config(), True),
 tests/ik_config_test.py:35: in ik_base_config
     robot_cfg = RobotConfig.from_dict(
 src/curobo/types/robot.py:49: in from_dict
     CudaRobotGeneratorConfig(**data_dict_in["kinematics"], tensor_args=tensor_args)
 <string>:31: in __init__
     ???
 src/curobo/cuda_robot_model/cuda_robot_generator.py:203: in __post_init__
     self.extra_links[k] = LinkParams.from_dict(self.extra_links[k])
 src/curobo/cuda_robot_model/kinematics_parser.py:43: in from_dict
[... 141 lines cut ...]
 ../.local/lib/python3.8/site-packages/warp/codegen.py:1957: in eval
     return emit_node(adj, node)
 ../.local/lib/python3.8/site-packages/warp/codegen.py:1786: in emit_Assign
     value = adj.eval(node.value)
 ../.local/lib/python3.8/site-packages/warp/codegen.py:1957: in eval
     return emit_node(adj, node)
 ../.local/lib/python3.8/site-packages/warp/codegen.py:1674: in emit_Call
     out = adj.add_call(func=func, args=args, kwds=kwds, templates=templates, min_outputs=min_outputs)
 ../.local/lib/python3.8/site-packages/warp/codegen.py:895: in add_call
     func = adj.resolve_func(func, args, min_outputs, templates, kwds)
 ../.local/lib/python3.8/site-packages/warp/codegen.py:890: in resolve_func
     raise WarpCodegenError(
 E   warp.codegen.WarpCodegenError: Error while parsing function "compute_matrix_to_quat" at /home/user/curobo/src/curobo/geom/transform.py:401:
 E       out_v[0] = wp.index(out_q, 3)
 E   ;Couldn't find function overload for 'index' that matched inputs with types: [warp.types.quatf, int32]
 =============================== warnings summary ===============================
 <unknown>:3
   <unknown>:3: DeprecationWarning: invalid escape sequence \p
 -- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
 =========================== short test summary info ============================
 ERROR tests/ik_config_test.py - warp.codegen.WarpCodegenError: Error while pa...
 ERROR tests/trajopt_config_test.py - warp.codegen.WarpCodegenError: Error whi...
 !!!!!!!!!!!!!!!!!!! Interrupted: 2 errors during collection !!!!!!!!!!!!!!!!!!!!
 =================== 2 skipped, 1 warning, 2 errors in 12.56s ===================
```

balakumar-s, 2023-12-05:
What version of warp are you using? Can you install warp 0.10.0 or lower?

waldezjr, 2024-01-04:
I was able to compile libwarp in my Orin Nx (not AGX).
I used tag v0.9.0 and the option to build llvm from source.

`python3 build.py -build_llvm`

Without this option, warp uses the already built files for x64, which (maybe obviously in hindsight xD) don't work in aarch64

c0d1f1ed, 2024-01-05:
I've been working on this recently. The next release of Warp will include Jetson support, and we'll publish `pip` packages.

c0d1f1ed, 2024-01-11:
[Warp release 1.0.0-beta.6](https://github.com/NVIDIA/warp/releases/tag/v1.0.0-beta.6) adds Jetson support. It can be installed using `pip install warp-lang==1.0.0b6`.

Hermanye996, 2024-01-18:
`pip install warp-lang==1.0.0b6` works for me, but error occurred when I run curobo fk. What should I do? Thanks. @c0d1f1ed @balakumar-s 
```bash
[INFO]  Creating curobo robot config instance...
Traceback (most recent call last):
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/codegen.py", line 620, in build
    adj.eval(adj.tree.body[0])
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/codegen.py", line 1975, in eval
    return emit_node(adj, node)
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/codegen.py", line 1162, in emit_FunctionDef
    adj.eval(f)
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/codegen.py", line 1975, in eval
    return emit_node(adj, node)
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/codegen.py", line 1803, in emit_Assign
    value = adj.eval(node.value)
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/codegen.py", line 1975, in eval
    return emit_node(adj, node)
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/codegen.py", line 1691, in emit_Call
    out = adj.add_call(func=func, args=args, kwds=kwds, templates=templates, min_outputs=min_outputs)
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/codegen.py", line 896, in add_call
    func = adj.resolve_func(func, args, min_outputs, templates, kwds)
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/codegen.py", line 891, in resolve_func
    raise WarpCodegenError(
warp.codegen.WarpCodegenError: Couldn't find function overload for 'index' that matched inputs with types: [warp.types.quatf, int32]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "herman_curobo_warpper.py", line 1482, in <module>
    main()
  File "herman_curobo_warpper.py", line 1468, in main
    demo_compute_fk(command_line_args)
  File "herman_curobo_warpper.py", line 1297, in demo_compute_fk
    herman_curobo_warpper = hermanCuroboWarpper(
  File "herman_curobo_warpper.py", line 52, in __init__
    self.curobo_init()
  File "herman_curobo_warpper.py", line 91, in curobo_init
    self.curobo_kinemetics_init()
  File "herman_curobo_warpper.py", line 239, in curobo_kinemetics_init
    self.curobo_robot_config_instance = RobotConfig.from_dict(
  File "/home/herman_orin_240118/gmp_ws/curobo/src/curobo/types/robot.py", line 49, in from_dict
    CudaRobotGeneratorConfig(**data_dict_in["kinematics"], tensor_args=tensor_args)
  File "<string>", line 33, in __init__
  File "/home/herman_orin_240118/gmp_ws/curobo/src/curobo/cuda_robot_model/cuda_robot_generator.py", line 208, in __post_init__
    self.extra_links[k] = LinkParams.from_dict(self.extra_links[k])
  File "/home/herman_orin_240118/gmp_ws/curobo/src/curobo/cuda_robot_model/kinematics_parser.py", line 43, in from_dict
    Pose.from_list(dict_data["fixed_transform"], tensor_args=TensorDeviceType())
  File "/home/herman_orin_240118/gmp_ws/curobo/src/curobo/types/math.py", line 281, in get_numpy_matrix
    return self.get_matrix().cpu().numpy()
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/contextlib.py", line 75, in inner
    return func(*args, **kwds)
  File "/home/herman_orin_240118/gmp_ws/curobo/src/curobo/types/math.py", line 277, in get_matrix
    full_mat = pose_to_matrix(self.position, self.quaternion, out_matrix)
  File "/home/herman_orin_240118/gmp_ws/curobo/src/curobo/geom/transform.py", line 163, in pose_to_matrix
    out_matrix[..., :3, :3] = quaternion_to_matrix(quaternion)
  File "/home/herman_orin_240118/gmp_ws/curobo/src/curobo/geom/transform.py", line 110, in quaternion_to_matrix
    out_mat = QuatToMatrix.apply(quaternions, out_mat, adj_quaternion)
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/torch/autograd/function.py", line 506, in apply
    return super().apply(*args, **kwargs)  # type: ignore[misc]
  File "/home/herman_orin_240118/gmp_ws/curobo/src/curobo/geom/transform.py", line 1067, in forward
    init_warp()
  File "/home/herman_orin_240118/gmp_ws/curobo/src/curobo/util/warp.py", line 27, in init_warp
    wp.force_load(wp.device_from_torch(tensor_args.device))
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/context.py", line 3698, in force_load
    m.load(d)
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/context.py", line 1534, in load
    builder = ModuleBuilder(self, self.options)
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/context.py", line 1166, in __init__
    self.build_kernel(kernel)
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/context.py", line 1194, in build_kernel
    kernel.adj.build(self)
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/codegen.py", line 634, in build
    raise e
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/codegen.py", line 620, in build
    adj.eval(adj.tree.body[0])
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/codegen.py", line 1975, in eval
    return emit_node(adj, node)
[... cut ...]
    func = adj.resolve_func(func, args, min_outputs, templates, kwds)
  File "/home/herman_orin_240118/anaconda3/envs/gmp/lib/python3.8/site-packages/warp/codegen.py", line 891, in resolve_func
    raise WarpCodegenError(
warp.codegen.WarpCodegenError: Error while parsing function "compute_pose_inverse" at /home/herman_orin_240118/gmp_ws/curobo/src/curobo/geom/transform.py:356:
    out_v[0] = wp.index(out_q, 3)
;Couldn't find function overload for 'index' that matched inputs with types: [warp.types.quatf, int32]
```

c0d1f1ed, 2024-01-18:
Sorry about that. I made a breaking change by adding `wp.extract()` to replace `wp.index()` when used to extract elements from a vector or quaternion. `wp.index()` now only handles array indexing. I should have documented that change better.

balakumar-s, 2024-01-18:
We will push a fix on the curobo repo to remove use of wp.index().

balakumar-s, 2024-01-25:
We pushed a fix to curobo, there are now no instances of wp.index and it should work with the latest warp.

closed 2024-01-25 by balakumar-s