NVlabs/curobo #64: nvrtc: error: invalid value for --gpu-architecture (-arch)

pmj110119, 2023-12-05:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]):     python
2. python version:       python3.7.13
3. Isaac Sim version (if using):       not used

------
Issue Details

When , I meet the error when run `example/motion_gen_api_example.py`, the main error code is :
```
Traceback of TorchScript (most recent call last):
  File "/home/panmingjie/gsnet/curobo/src/curobo/util/sample_lib.py", line 511, in gaussian_transform
    uniform_samples: torch.Tensor, proj_mat: torch.Tensor, i_mat: torch.Tensor, variance: float
):
    gaussian_halton_samples = proj_mat * torch.erfinv(2 * uniform_samples - 1)
                                         ~~~~~~~~~~~~ <--- HERE
    i_mat = i_mat * variance
    gaussian_halton_samples = torch.matmul(gaussian_halton_samples, i_mat)

nvrtc: error: invalid value for --gpu-architecture (-arch)

```
pytorch version: 1.12.0
CUDA version: 11.3

The point that confuses me is that my pytorch and cuda version can run other repos and perform model training normally. I tried running the `torch.erfinv()` function alone but no error was reported. Why is it reporting an error when running curobo?

I'm not very familiar with CUDA, I hope you can help me .

balakumar-s, 2023-12-05:
I am assuming this is not inside a docker. 
Can you run the unit tests: `cd curobo && python -m pytest tests/`

pmj110119, 2023-12-06:
Yes, it's not inside the docker. I start to use dockerfile and meet new [issue](https://github.com/NVlabs/curobo/issues/70)

closed 2023-12-06 by pmj110119