NVlabs/curobo #76: Error when use shadowhand robot

Spphire, 2023-12-12:
Isaac Sim version 2023

The reason might be too many sphere collisions?
[image]

[config.zip](https://github.com/NVlabs/curobo/files/13642958/config.zip)

balakumar-s, 2023-12-15:
We fixed a bug that could have been the cause of this issue. Can you try with the latest commit on main branch? Once you download the latest commit, recompile the package with `omni_python -m pip install -e . --no-build-isolation`

Spphire, 2023-12-16:
> We fixed a bug that could have been the cause of this issue. Can you try with the latest commit on main branch? Once you download the latest commit, recompile the package with `omni_python -m pip install -e . --no-build-isolation`

cant install it
[image]

balakumar-s, 2023-12-16:
It looks like git didn't clone the repo correctly. Can you clone the repo into a new folder and try again?
Also make sure you have the following installed `omni_python -m pip install tomli wheel ninja`

Spphire, 2023-12-18:
It's fixed!

closed 2023-12-18 by Spphire