NVlabs/curobo #82: Possible Error When Motion_Gen's Plan Fails to Converge to a Solution

zoctipus, 2023-12-15:
I think there is some bug with batch_motion_gen_reacher.py with n_envs >=5( n_envs = 2 - 4 seems working fine)

To reproduce the bug, go

"batch_motion_gen_reacher.py" 

and change few lines:

1.
```
n_envs = 2 ->
n_envs = 5
```

2.
```
world_file = ["collision_test.yml", "collision_thin_walls.yml"] ->
world_file = ["collision_test.yml" for i in range(n_envs)]
```

3.
cmd_plan = [None, None] ->
cmd_plan = [None for i in range(n_envs)]

then running the file will cause no planned trajectory being success, if we commented out the check ``if result.success[s]:``
we can observe
[image]

In contrast, if we just change the value of n_envs from 5 to 4
```n_envs = 4```
all the issue are gone and can be seens from
[image]

the updated batch_motiuon_gen_reacher.py that reproduces error can be downloaded from here
https://drive.google.com/drive/folders/12Ph7JvX9RleSdUkNCYJS6Mu96byihg9l

Thank you for your hard work.

balakumar-s, 2023-12-15:
Can you try with the latest commit on main branch? We fixed some indexing bugs which could have also removed this issue. Once you download the latest commit, recompile the package with `python3 -m pip install -e . --no-build-isolation`

zoctipus, 2023-12-15:
Hello, Thank you for getting back to me,

I have tested the latest branch, I think the same issue but different bug persists, when n_envs = 2, 3, or 4, the motion_gen will be successfull. But when n_envs = 5, the warm up will fail. The Trace Stack can be found at the end.

The python file reproduces the bug can be downloaded here(remember to change n_envs from 4 to 5 to reproduce the error):
https://drive.google.com/drive/folders/1H5VQtwpbcy1TQrhW03QkFkFO-qQIAYrJ

modification of the file are:
```
world_file = ["collision_test.yml", "collision_thin_walls.yml"] ->
world_file = ["collision_test.yml" for i in range(n_envs)]
```
```
cmd_plan = [None, None] ->
world_file = ["collision_test.yml", "collision_thin_walls.yml"]
```

trace stack:
Stack (most recent call last):
  File "~/Projects/Orbit/curobo/examples/isaac_sim/batch_motion_gen_reacher.py", line 298, in <module>
    main()
  File "~/Projects/Orbit/curobo/examples/isaac_sim/batch_motion_gen_reacher.py", line 165, in main
    enable_graph=False, warmup_js_trajopt=False, batch=n_envs, batch_env_mode=True
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/motion_gen.py", line 2149, in warmup
    max_attempts=1, enable_finetune_trajopt=True, enable_graph=enable_graph
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1318, in plan_batch_env
    plan_config,
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1143, in _plan_batch_attempts
    solve_state, start_state, goal_pose, plan_config
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1983, in _plan_from_solve_state_batch
    return_all_solutions=True,
  File "~/anaconda3/envs/orbit/lib/python3.7/contextlib.py", line 74, in inner
    return func(*args, **kwds)
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/motion_gen.py", line 920, in _solve_trajopt_from_solve_state
    newton_iters=newton_iters,
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/trajopt.py", line 537, in solve_any
    newton_iters=newton_iters,
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/trajopt.py", line 737, in solve_batch_env
    newton_iters=newton_iters,
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/trajopt.py", line 578, in solve_from_solve_state
    self.solver.reset()
  File "~/Projects/Orbit/curobo/src/curobo/wrap/wrap_base.py", line 106, in reset
    opt.reset()
  File "~/Projects/Orbit/curobo/src/curobo/opt/particle/parallel_mppi.py", line 457, in reset
    self.update_samples()  # this helps in restarting optimization
  File "~/Projects/Orbit/curobo/src/curobo/opt/particle/parallel_mppi.py", line 530, in update_samples
    base_seed=self.seed,
  File "~/Projects/Orbit/curobo/src/curobo/util/sample_lib.py", line 378, in get_samples
    samples = self.sample_fns[k](sample_shape=s_shape)
  File "~/Projects/Orbit/curobo/src/curobo/util/sample_lib.py", line 331, in get_samples
    log_error("Nan values found in samplelib, installation could have been corrupted")
  File "~/Projects/Orbit/curobo/src/curobo/util/logger.py", line 44, in log_error
    logger.error(txt, exc_info=exc_info, stack_info=stack_info, *args, **kwargs)
Traceback (most recent call last):
  File "~/Projects/Orbit/curobo/examples/isaac_sim/batch_motion_gen_reacher.py", line 298, in <module>
    main()
  File "~/Projects/Orbit/curobo/examples/isaac_sim/batch_motion_gen_reacher.py", line 165, in main
    enable_graph=False, warmup_js_trajopt=False, batch=n_envs, batch_env_mode=True
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/motion_gen.py", line 2149, in warmup
    max_attempts=1, enable_finetune_trajopt=True, enable_graph=enable_graph
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1318, in plan_batch_env
    plan_config,
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1143, in _plan_batch_attempts
    solve_state, start_state, goal_pose, plan_config
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1983, in _plan_from_solve_state_batch
    return_all_solutions=True,
  File "~/anaconda3/envs/orbit/lib/python3.7/contextlib.py", line 74, in inner
    return func(*args, **kwds)
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/motion_gen.py", line 920, in _solve_trajopt_from_solve_state
    newton_iters=newton_iters,
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/trajopt.py", line 537, in solve_any
    newton_iters=newton_iters,
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/trajopt.py", line 737, in solve_batch_env
    newton_iters=newton_iters,
  File "~/Projects/Orbit/curobo/src/curobo/wrap/reacher/trajopt.py", line 578, in solve_from_solve_state
    self.solver.reset()
  File "~/Projects/Orbit/curobo/src/curobo/wrap/wrap_base.py", line 106, in reset
    opt.reset()
  File "~/Projects/Orbit/curobo/src/curobo/opt/particle/parallel_mppi.py", line 457, in reset
    self.update_samples()  # this helps in restarting optimization
  File "~/Projects/Orbit/curobo/src/curobo/opt/particle/parallel_mppi.py", line 530, in update_samples
    base_seed=self.seed,
  File "~/Projects/Orbit/curobo/src/curobo/util/sample_lib.py", line 378, in get_samples
    samples = self.sample_fns[k](sample_shape=s_shape)
  File "~/Projects/Orbit/curobo/src/curobo/util/sample_lib.py", line 331, in get_samples
    log_error("Nan values found in samplelib, installation could have been corrupted")
  File "~/Projects/Orbit/curobo/src/curobo/util/logger.py", line 45, in log_error
    raise
RuntimeError: No active exception to reraise

Thank you for your hard work XD!

balakumar-s, 2023-12-16:
We found an issue where matmul leads to nans in stomp covariance computation and added a error as printed in your terminal. We are not yet sure why this happens. 

Here is a workaround that I tested in your script and it worked.

Go to this line : https://github.com/NVlabs/curobo/blob/256227b0b6ce2f06892aef24bf7d933bb017806a/src/curobo/content/configs/task/particle_trajopt.yml#L106

change it to `sample_ratio: {'halton':0.0, 'halton-knot':1.0, 'random':0.0, 'random-knot':0.0, "stomp": 0.0}`

We are effectively disabling stomp sampling and using spline samples.

zoctipus, 2023-12-16:
It worked!
Thank you!!

balakumar-s, 2023-12-19:
We pushed a fix for this. You can revert back to stomp sampling.

closed 2023-12-19 by balakumar-s