NVlabs/curobo #86: Motion_gen fail to plan with multi_arm_reacher.py and custom robot

Spphire, 2023-12-18:
Isaac Sim version: 2022.2.1

In multi_arm_reacher.py:
https://github.com/NVlabs/curobo/assets/56157591/0ecf78ea-437c-438e-8a4b-b316d178fe87

I use mpc instead of motion_gen, it performs well, so i think the reason is not my custom robot:

https://github.com/NVlabs/curobo/assets/56157591/8d5d6426-2e37-4d45-b0a9-5b94e731040d

balakumar-s, 2023-12-18:
Is it possible for you to share the robot configuration file and the urdf?

The reason could be that the robot configuration file did not specify all self collision ignores. Mpc will return a solution even when in self collision.

Spphire, 2023-12-18:
> Is it possible for you to share the robot configuration file and the urdf?
> 
> The reason could be that the robot configuration file did not specify all self collision ignores. Mpc will return a solution even when in self collision.

sure
[content_flexiv.zip](https://github.com/NVlabs/curobo/files/13705595/content_flexiv.zip)

balakumar-s, 2023-12-18:
I found the issue. The spheres for the base link were colliding with the ground. We will add a note in the robot configuration tutorial about this.

Reduce the radius of the spheres for base_link and base_link_1 to 0.05. This worked for me.

Spphire, 2023-12-19:
> I found the issue. The spheres for the base link were colliding with the ground. We will add a note in the robot configuration tutorial about this.
> 
> Reduce the radius of the spheres for base_link and base_link_1 to 0.05. This worked for me.

Thx!

closed 2023-12-19 by Spphire

WangYong-999, 2024-01-30:
Hello, I would like to ask how to use mpc in two robotic arms, similar to multi_arm_reacher.py, but I do not find the `link_poses` parameter in mpc similar to the plan_single() function. @balakumar-s