NVlabs/curobo #87: nvrtc: error: invalid value for --gpu-architecture (-arch)

im-renpei, 2023-12-20:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): python
2. python version: 3.8.10
3. Isaac Sim version (if using): N/A
4. Ubuntu version: 20.04

Hello Balakumar @balakumar-s ,

I am getting an error when trying to install cuRobo (Python version without Isaac Sim). Following is the installation details. Only a few example can be executed normally, and the pytest cannot pass normally.

------
Issue Details

```
(base) ➜  ~ conda create -n curobo-env python=3.8.10
Collecting package metadata (current_repodata.json): done
Solving environment: unsuccessful attempt using repodata from current_repodata.json, retrying with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
  current version: 23.7.4
  latest version: 23.11.0

Please update conda by running

    $ conda update -n base -c defaults conda

Or to minimize the number of packages updated during conda update use

     conda install conda=23.11.0

## Package Plan ##

  environment location: /home/pren/anaconda3/envs/curobo-env

  added / updated specs:
    - python=3.8.10

The following NEW packages will be INSTALLED:

  _libgcc_mutex      pkgs/main/linux-64::_libgcc_mutex-0.1-main 
  _openmp_mutex      pkgs/main/linux-64::_openmp_mutex-5.1-1_gnu 
  ca-certificates    pkgs/main/linux-64::ca-certificates-2023.12.12-h06a4308_0 
  ld_impl_linux-64   pkgs/main/linux-64::ld_impl_linux-64-2.38-h1181459_1 
  libffi             pkgs/main/linux-64::libffi-3.3-he6710b0_2 
  libgcc-ng          pkgs/main/linux-64::libgcc-ng-11.2.0-h1234567_1 
  libgomp            pkgs/main/linux-64::libgomp-11.2.0-h1234567_1 
  libstdcxx-ng       pkgs/main/linux-64::libstdcxx-ng-11.2.0-h1234567_1 
  ncurses            pkgs/main/linux-64::ncurses-6.4-h6a678d5_0 
  openssl            pkgs/main/linux-64::openssl-1.1.1w-h7f8727e_0 
  pip                pkgs/main/linux-64::pip-23.3.1-py38h06a4308_0 
  python             pkgs/main/linux-64::python-3.8.10-h12debd9_8 
  readline           pkgs/main/linux-64::readline-8.2-h5eee18b_0 
  setuptools         pkgs/main/linux-64::setuptools-68.2.2-py38h06a4308_0 
  sqlite             pkgs/main/linux-64::sqlite-3.41.2-h5eee18b_0 
  tk                 pkgs/main/linux-64::tk-8.6.12-h1ccaba5_0 
  wheel              pkgs/main/linux-64::wheel-0.41.2-py38h06a4308_0 
  xz                 pkgs/main/linux-64::xz-5.4.5-h5eee18b_0 
  zlib               pkgs/main/linux-64::zlib-1.2.13-h5eee18b_0 

Proceed ([y]/n)? Y

Downloading and Extracting Packages

Preparing transaction: done
Verifying transaction: done
[... 1402 lines cut ...]
tests/ik_config_test.py:149
  /home/pren/Space/GitHub-Codes/curobo_ws/curobo/curobo/tests/ik_config_test.py:149: SyntaxWarning: "is not" with a literal. Did you mean "!="?
    if expected is not -100:

src/curobo/types/math.py:83
src/curobo/types/math.py:83
  /home/pren/Space/GitHub-Codes/curobo_ws/curobo/curobo/src/curobo/types/math.py:83: UserWarning: FALLBACK path has been taken inside: runCudaFusionGroup. This is an indication that codegen Failed for some reason.
  To debug try disable codegen fallback path via setting the env variable `export PYTORCH_NVFUSER_DISABLE=fallback`
   (Triggered internally at  ../torch/csrc/jit/codegen/cuda/manager.cpp:329.)
    self.quaternion = normalize_quaternion(self.quaternion)

src/curobo/opt/particle/particle_opt_utils.py:67
  /home/pren/Space/GitHub-Codes/curobo_ws/curobo/curobo/src/curobo/opt/particle/particle_opt_utils.py:67: UserWarning: FALLBACK path has been taken inside: runCudaFusionGroup. This is an indication that codegen Failed for some reason.
  To debug try disable codegen fallback path via setting the env variable `export PYTORCH_NVFUSER_DISABLE=fallback`
   (Triggered internally at  ../torch/csrc/jit/codegen/cuda/manager.cpp:329.)
    cov, scale_tril, scaled_M = get_stomp_cov_jit(

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
======================================== short test summary info ========================================
ERROR tests/ik_config_test.py - RuntimeError: The following operation failed in the TorchScript interp...
ERROR tests/trajopt_config_test.py - RuntimeError: The following operation failed in the TorchScript i...
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 2 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
=============================== 2 skipped, 5 warnings, 2 errors in 3.49s ================================
(curobo-env) ➜  curobo git:(main) 
```

balakumar-s, 2023-12-20:
What gpu are you testing this on? Also what's your version of the nvidia driver?

im-renpei, 2023-12-20:
> What gpu are you testing this on? Also what's your version of the nvidia driver?

Hi Balakumar @balakumar-s , thx for your timely reply. I just update the above issue description. The GPU is RTX 4090 and the driver is 525.

```
➜  ~ nvidia-smi
Wed Dec 20 10:09:58 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.125.06   Driver Version: 525.125.06   CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  Off |
| 32%   31C    P8    12W / 450W |   2567MiB / 24564MiB |      1%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1210      G   /usr/lib/xorg/Xorg                151MiB |
|    0   N/A  N/A      2038      G   /usr/lib/xorg/Xorg               1448MiB |
|    0   N/A  N/A      2167      G   /usr/bin/gnome-shell              257MiB |
|    0   N/A  N/A      2508      G   ...RendererForSitePerProcess       18MiB |
|    0   N/A  N/A      4024    C+G   ...369779272449050828,262144      595MiB |
|    0   N/A  N/A      5735      G   ...RendererForSitePerProcess       15MiB |
|    0   N/A  N/A     12806      G   ...RendererForSitePerProcess       60MiB |
+-----------------------------------------------------------------------------+
➜  ~ 
➜  ~ nvcc -V   
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Tue_May__3_18:49:52_PDT_2022
Cuda compilation tools, release 11.7, V11.7.64
Build cuda_11.7.r11.7/compiler.31294372_0
➜  ~ 
```

balakumar-s, 2023-12-20:
Can you try with cuda 11.8?

Pytorch provides conda installation for cuda 11.8 here: 
https://pytorch.org/get-started/locally/

balakumar-s, 2023-12-20:
Looks like it's a known bug in pytorch 1.13 that was fixed. 

https://github.com/pytorch/pytorch/issues/87595

Install the newer pytorch 2.0 and this should be fixed.

im-renpei, 2023-12-20:
> Can you try with cuda 11.8?
> 
> Pytorch provides conda installation for cuda 11.8 here: https://pytorch.org/get-started/locally/

I will try it now.

im-renpei, 2023-12-20:
Hi Yibo @Spphire , I see that you can successfully run CuRobo demo in Isaac Sim. May I ask what your environment (Ubuntu, Python, CUDA, PyTorch, Isaac Sim) is?

im-renpei, 2023-12-20:
> Looks like it's a known bug in pytorch 1.13 that was fixed.
> 
> [pytorch/pytorch#87595](https://github.com/pytorch/pytorch/issues/87595)
> 
> Install the newer pytorch 2.0 and this should be fixed.

My setup is

```
conda create -n curobo-env python=3.8.10
conda activate curobo-env
conda install pytorch torchvision torchaudio pytorch-cuda=11.8 -c pytorch -c nvidia
conda install pip
mkdir -p /home/pren/Space/GitHub-Codes/curobo_ws/curobo
cd /home/pren/Space/GitHub-Codes/curobo_ws/curobo
git clone https://github.com/NVlabs/curobo.git
cd curobo
git lfs pull
pip install -e . --no-build-isolation
python examples/collision_check_example.py  # Passed
python examples/ik_example.py  # Passed
pip install pluggy
python3 -m pytest .  # 1 failed, 135 passed, 3 skipped, 5 warnings
```

```
➜  ~ nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Wed_Sep_21_10:33:58_PDT_2022
Cuda compilation tools, release 11.8, V11.8.89
Build cuda_11.8.r11.8/compiler.31833905_0
➜  ~    
➜  ~ nvidia-smi
Wed Dec 20 14:33:51 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.125.06   Driver Version: 525.125.06   CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  Off |
| 32%   34C    P8    12W / 450W |   1663MiB / 24564MiB |      1%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1210      G   /usr/lib/xorg/Xorg                151MiB |
|    0   N/A  N/A      2038      G   /usr/lib/xorg/Xorg                865MiB |
|    0   N/A  N/A      2167      G   /usr/bin/gnome-shell              134MiB |
|    0   N/A  N/A      2508      G   ...RendererForSitePerProcess       14MiB |
|    0   N/A  N/A      4024    C+G   ...369779272449050828,262144      414MiB |
|    0   N/A  N/A     12806      G   ...RendererForSitePerProcess       62MiB |
+-----------------------------------------------------------------------------+
➜  ~ 
```

Issue Details

```
(curobo-env) ➜  curobo git clone https://github.com/NVlabs/curobo.git
Cloning into 'curobo'...
remote: Enumerating objects: 854, done.
remote: Counting objects: 100% (74/74), done.
remote: Compressing objects: 100% (34/34), done.
[... 190 lines cut ...]

tests/ik_config_test.py:149
  /home/pren/Space/GitHub-Codes/curobo_ws/curobo/curobo/tests/ik_config_test.py:149: SyntaxWarning: "is not" with a literal. Did you mean "!="?
    if expected is not -100:

tests/pose_reaching_test.py::test_pose_sequence_ur5e[True-False-12]
  /home/pren/Space/GitHub-Codes/curobo_ws/curobo/curobo/src/curobo/graph/graph_base.py:696: UserWarning: FALLBACK path has been taken inside: runCudaFusionGroup. This is an indication that codegen Failed for some reason.
  To debug try disable codegen fallback path via setting the env variable `export PYTORCH_NVFUSER_DISABLE=fallback`
   (Triggered internally at  ../torch/csrc/jit/codegen/cuda/manager.cpp:329.)
    return compute_distance_norm_jit(pt, batch_pts, self.distance_weight)

tests/pose_test.py::test_pose_transform_point_grad
  /home/pren/.local/lib/python3.8/site-packages/torch/autograd/gradcheck.py:652: UserWarning: Input #0 requires gradient and is not a double precision floating point or complex. This check will likely fail if all the inputs are not of double precision floating point or complex. 
    warnings.warn(

tests/pose_test.py::test_pose_transform_point_grad
  /home/pren/.local/lib/python3.8/site-packages/torch/autograd/gradcheck.py:652: UserWarning: Input #1 requires gradient and is not a double precision floating point or complex. This check will likely fail if all the inputs are not of double precision floating point or complex. 
    warnings.warn(

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
===================================================================================== short test summary info =====================================================================================
FAILED tests/mpc_test.py::test_mpc_single[mpc_single_env-True] - assert False == True
================================================================= 1 failed, 135 passed, 3 skipped, 5 warnings in 74.77s (0:01:14) =================================================================
(curobo-env) ➜  curobo git:(main)
```

balakumar-s, 2023-12-20:
It's working correctly, that one test which failed is an outlier.

im-renpei, 2023-12-21:
> It's working correctly, that one test which failed is an outlier.

By the way, I would like to ask under which environments has CuRobo been tested and is feasible? Have you used CuRobo under Isaac Sim 2023.1.0 (currently 2023.1.0-hotfix.1; and 2023.1.1 is now available) and Ubuntu 22.04? What are the minimum CUDA and pytorch versions required by CuRobo?

balakumar-s, 2023-12-22:
cuRobo will work on pytorch>=1.10 and CUDA>=11.0.

However there might be bugs in pytorch like the one you encountered that can cause issues for newer gpus working on older pytorch. 

As long as you can run pytorch with your gpu, curobo will also work. 

Regarding Ubuntu, it doesn't matter which one you use.

We have tested isaac sim 2022.2.1 and 2023.1.0. Newer versions should also work.

im-renpei, 2023-12-25:
Hi @balakumar-s , have you encountered these two problems?

- python version: 3.10.12
- Isaac Sim version (if using): N/A
- Ubuntu version: 22.04
- GPU: NVIDIA GeForce RTX A6000
- Driver Version: 525.147.05

```
pip install torch==2.1.1 torchvision==0.16.1 torchaudio==2.1.1 --index-url https://download.pytorch.org/whl/cu121
```

```
eai@eai-Z790-UD-AX:~/rp/curobo$ nvidia-smi
Mon Dec 25 20:33:36 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.147.05   Driver Version: 525.147.05   CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA RTX A6000    Off  | 00000000:01:00.0  On |                  Off |
| 30%   41C    P8    28W / 300W |    365MiB / 49140MiB |     12%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1876      G   /usr/lib/xorg/Xorg                146MiB |
|    0   N/A  N/A      1999      G   /usr/bin/gnome-shell               35MiB |
|    0   N/A  N/A      2480      G   ...RendererForSitePerProcess       13MiB |
|    0   N/A  N/A      2798      G   ...7/usr/lib/firefox/firefox      165MiB |
+-----------------------------------------------------------------------------+
eai@eai-Z790-UD-AX:~/rp/curobo$ nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Feb__7_19:32:13_PST_2023
Cuda compilation tools, release 12.1, V12.1.66
Build cuda_12.1.r12.1/compiler.32415258_0
```

# Issue 1

```
eai@eai-Z790-UD-AX:~/rp/curobo$ python3 examples/ik_example.py
Traceback (most recent call last):
  File "/home/eai/rp/curobo/examples/ik_example.py", line 232, in <module>
    demo_basic_ik()
  File "/home/eai/rp/curobo/examples/ik_example.py", line 62, in demo_basic_ik
    result = ik_solver.solve_batch(goal)
  File "/home/eai/rp/curobo/src/curobo/wrap/reacher/ik_solver.py", line 537, in solve_batch
    return self.solve_from_solve_state(
  File "/home/eai/rp/curobo/src/curobo/wrap/reacher/ik_solver.py", line 675, in solve_from_solve_state
    result = self.solver.solve(goal_buffer, coord_position_seed)
  File "/home/eai/rp/curobo/src/curobo/wrap/wrap_base.py", line 139, in solve
    act_seq = self.optimize(seed, shift_steps=0)
  File "/home/eai/rp/curobo/src/curobo/wrap/wrap_base.py", line 71, in optimize
    act_seq = opt.optimize(act_seq, shift_steps)
  File "/home/eai/rp/curobo/src/curobo/opt/opt_base.py", line 96, in optimize
    out = self._optimize(opt_tensor, shift_steps, n_iters)
  File "/home/eai/rp/curobo/src/curobo/opt/newton/newton_base.py", line 144, in _optimize
    best_q, best_cost, q, grad_q = self._call_opt_iters_graph(q, grad_q)
  File "/home/eai/rp/curobo/src/curobo/opt/newton/newton_base.py", line 521, in _call_opt_iters_graph
    return self._opt_iters(q, grad_q)
  File "/home/eai/rp/curobo/src/curobo/opt/newton/newton_base.py", line 170, in _opt_iters
    cost_n, q, grad_q = self._opt_step(q.detach(), grad_q.detach())
[... 42 lines cut ...]
    raise ex[1].with_traceback(ex[2])
  File "/usr/lib/python3/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/usr/lib/python3/dist-packages/_pytest/config/__init__.py", line 1003, in pytest_cmdline_parse
    self.parse(args)
  File "/usr/lib/python3/dist-packages/_pytest/config/__init__.py", line 1283, in parse
    self._preparse(args, addopts=addopts)
  File "/usr/lib/python3/dist-packages/_pytest/config/__init__.py", line 1172, in _preparse
    self.pluginmanager.load_setuptools_entrypoints("pytest11")
  File "/usr/lib/python3/dist-packages/pluggy/manager.py", line 297, in load_setuptools_entrypoints
    plugin = ep.load()
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "/usr/lib/python3/dist-packages/_pytest/assertion/rewrite.py", line 170, in exec_module
    exec(co, module.__dict__)
  File "/home/eai/.local/lib/python3.10/site-packages/typeguard/_pytest_plugin.py", line 6, in <module>
    from pytest import Config, Parser
ImportError: cannot import name 'Config' from 'pytest' (/usr/lib/python3/dist-packages/pytest/__init__.py)
```

balakumar-s, 2023-12-26:
Your driver only supports up to CUDA 12.0. Can you upgrade your nvidia driver and test?

im-renpei, 2023-12-27:
Hi @balakumar-s , thx for your timely reply.

I have updated my CUDA and NVIDIA driver version, and seems that CuRobo works well with Python but seems to have encountered problems in Isaac Sim.

Ubuntu version: 22.04.3
Yython version: 3.10.12
Isaac Sim version (if using): N/A
GPU: NVIDIA GeForce RTX A6000
Driver version: 545.29.06
CUDA version: 12.3

```
eai@eai-Z790-UD-AX:~$ nvidia-smi
Wed Dec 27 19:59:33 2023       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 545.29.06              Driver Version: 545.29.06    CUDA Version: 12.3     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA RTX A6000               Off | 00000000:01:00.0  On |                  Off |
| 30%   37C    P8              17W / 300W |    706MiB / 49140MiB |      5%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      2041      G   /usr/lib/xorg/Xorg                          152MiB |
|    0   N/A  N/A      2165      G   /usr/bin/gnome-shell                        225MiB |
|    0   N/A  N/A      2814      G   ...ures=SpareRendererForSitePerProcess       46MiB |
|    0   N/A  N/A     65181      C   python3                                     260MiB |
+---------------------------------------------------------------------------------------+
eai@eai-Z790-UD-AX:~$ 
eai@eai-Z790-UD-AX:~$ nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Aug_15_22:02:13_PDT_2023
Cuda compilation tools, release 12.2, V12.2.140
Build cuda_12.2.r12.2/compiler.33191640_0
eai@eai-Z790-UD-AX:~$ 
```

```
sudo apt install git-lfs
git clone https://github.com/NVlabs/curobo.git
cd curobo
git lfs pull
pip install -e . --no-build-isolation
```

```
eai@eai-Z790-UD-AX:~/rp/build_curobo/curobo$ python3 examples/ik_example.py
/usr/lib/python3/dist-packages/scipy/__init__.py:146: UserWarning: A NumPy version >=1.17.3 and <1.25.0 is required for this version of SciPy (detected version 1.26.2
  warnings.warn(f"A NumPy version >={np_minversion} and <{np_maxversion}"
Success, Solve Time(s), Total Time(s)  0.9992 1.1695911884307861 1789.24586747862 tensor(0.0107, device='cuda:0') tensor(0.1423, device='cuda:0')
Success, Solve Time(s), Total Time(s)  0.9972 0.2578887939453125 16037.759801289963 tensor(0.0233, device='cuda:0') tensor(0.3517, device='cuda:0')
Success, Solve Time(s), Total Time(s)  0.9942 0.25640439987182617 19382.611074552486 tensor(0.0702, device='cuda:0') tensor(0.5208, device='cuda:0')
Success, Solve Time(s), Total Time(s)  0.9946 0.2590043544769287 19212.834745867523 tensor(0.0736, device='cuda:0') tensor(0.4925, device='cuda:0')
Success, Solve Time(s), Total Time(s)  0.9944 0.25742006301879883 19333.383115260833 tensor(0.0466, device='cuda:0') tensor(0.5087, device='cuda:0')
Success, Solve Time(s), Total Time(s)  0.9938 0.258974552154541 19223.24294694696 tensor(0.0391, device='cuda:0') tensor(0.5017, device='cuda:0')
Success, Solve Time(s), Total Time(s)  0.993 0.25754809379577637 19317.9427375511 tensor(0.0525, device='cuda:0') tensor(0.6029, device='cuda:0')
Success, Solve Time(s), Total Time(s)  0.996 0.2578907012939453 19299.27630658686 tensor(0.0280, device='cuda:0') tensor(0.3821, device='cuda:0')
Success, Solve Time(s), Total Time(s)  0.9948 0.2580077648162842 19290.435691250303 tensor(0.0484, device='cuda:0') tensor(0.4698, device='cuda:0')
Success, Solve Time(s), Total Time(s)  0.9942 0.25784993171691895 19304.072799706915 tensor(0.0520, device='cuda:0') tensor(0.5610, device='cuda:0')
eai@eai-Z790-UD-AX:~/rp/build_curobo/curobo$
```
[... 263 lines cut ...]
  File "/home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/articulations/articulation_view.py", line 3243, in set_solver_velocity_iteration_counts
    set_prim_property(self.prim_paths[i], "physxArticulation:solverVelocityIterationCount", counts[read_idx])
  File "/home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/utils/prims.py", line 802, in set_prim_property
    prim.GetAttribute(property_name).Set(property_value)
pxr.Tf.ErrorException: 
	Error in 'pxrInternal_v0_22__pxrReserved__::UsdStage::_SetValueImpl' at line 6174 in file /buildAgent/work/ac88d7d902b57417/USD/pxr/usd/usd/stage.cpp : 'Empty typeName for </World/panda.physxArticulation:solverVelocityIterationCount>'
[... cut ...]
2023-12-27 11:26:53 [9,200ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2023-12-27 11:26:53 [9,202ms] [Warning] [carb.audio.context] 1 contexts were leaked
2023-12-27 11:26:53 [9,233ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
2023-12-27 11:26:53 [9,239ms] [Warning] [omni.core.ITypeFactory] Module /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.
eai@eai-Z790-UD-AX:~/rp/build_curobo/for_isaac_sim/curobo$
```

balakumar-s, 2023-12-27:
I am not sure what's happening here. Did you install `usd-core` in the isaac sim python environment (this can break things as Isaac Sim comes with it's own usd-core) ? 

Try one of the below:

Option 1:
We have not tested in Isaac Sim 2023.1.1. 

Can you comment out this line https://github.com/NVlabs/curobo/blob/c09d94908d8ad52f557ed59195c1ceb2d0434d65/examples/isaac_sim/helper.py#L115 
and also the next line and see if the errors go away?

Option 2:
We do not install a newer pytorch in an Isaac Sim env. Isaac sim comes with pytorch that was compiled with cuda 11.8. If you install cuda 11.8 in your machine that might fix the problem. Though I don't think this is the cause of the problem.

balakumar-s, 2023-12-28:
Regarding your pytest error, we have a known issue with pytest == 6.2.5 (https://curobo.org/source/getting_started/6_known_issues.html#python). Installing any other version of pytest will work.

im-renpei, 2023-12-28:
> Regarding your pytest error, we have a known issue with pytest == 6.2.5 (https://curobo.org/source/getting_started/6_known_issues.html#python). Installing any other version of pytest will work.

Hi @balakumar-s , this solution works. The following output proves that pytest can be passed, right?

```
eai@eai-Z790-UD-AX:~/rp/build_curobo/curobo$ pip list | grep pytest
launch-pytest                        1.0.4
pytest                               6.2.5
pytest-cov                           3.0.0
pytest-mock                          3.6.1
pytest-timeout                       2.1.0
eai@eai-Z790-UD-AX:~/rp/build_curobo/curobo$ pip install pytest==7.1.1
Defaulting to user installation because normal site-packages is not writeable
Collecting pytest==7.1.1
  Downloading pytest-7.1.1-py3-none-any.whl (297 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 297.0/297.0 kB 1.2 MB/s eta 0:00:00
Requirement already satisfied: attrs>=19.2.0 in /home/eai/.local/lib/python3.10/site-packages (from pytest==7.1.1) (23.1.0)
Requirement already satisfied: iniconfig in /usr/lib/python3/dist-packages (from pytest==7.1.1) (1.1.1)
Requirement already satisfied: packaging in /usr/lib/python3/dist-packages (from pytest==7.1.1) (21.3)
Requirement already satisfied: pluggy<2.0,>=0.12 in /usr/lib/python3/dist-packages (from pytest==7.1.1) (0.13.0)
Requirement already satisfied: py>=1.8.2 in /usr/lib/python3/dist-packages (from pytest==7.1.1) (1.10.0)
Requirement already satisfied: tomli>=1.0.0 in /usr/lib/python3/dist-packages (from pytest==7.1.1) (1.2.2)
Installing collected packages: pytest
Successfully installed pytest-7.1.1
eai@eai-Z790-UD-AX:~/rp/build_curobo/curobo$ pip list | grep pytest
launch-pytest                        1.0.4
pytest                               7.1.1
pytest-cov                           3.0.0
pytest-mock                          3.6.1
pytest-timeout                       2.1.0
eai@eai-Z790-UD-AX:~/rp/build_curobo/curobo$ python3 -m pytest .
=========================================================================== test session starts ===========================================================================
platform linux -- Python 3.10.12, pytest-7.1.1, pluggy-0.13.0
rootdir: /home/eai/rp/build_curobo/curobo, configfile: pyproject.toml
plugins: ament-mypy-0.12.9, ament-lint-0.12.9, launch-pytest-1.0.4, ament-flake8-0.12.9, ament-copyright-0.12.9, launch-testing-1.0.4, ament-pep257-0.12.9, launch-testing-ros-0.19.6, ament-xmllint-0.12.9, torchtyping-0.1.4, anyio-4.2.0, typeguard-4.1.5, mock-3.6.1, timeout-2.1.0, cov-3.0.0, colcon-core-0.15.1
collected 0 items / 1 skipped                                                                                                                                             

============================================================================ warnings summary =============================================================================
../../../../../usr/lib/python3/dist-packages/scipy/__init__.py:146
  Warning: A NumPy version >=1.17.3 and <1.25.0 is required for this version of SciPy (detected version 1.26.2

<unknown>:3
  Warning: invalid escape sequence '\p'

tests/ik_config_test.py:149
  Warning: "is not" with a literal. Did you mean "!="?

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
===================================================================== 1 skipped, 3 warnings in 1.57s ======================================================================
eai@eai-Z790-UD-AX:~/rp/build_curobo/curobo$
```

balakumar-s, 2023-12-28:
No, it didn't find any tests. Can you try with `python3 -m pytest tests/` ?

im-renpei, 2023-12-28:
> I am not sure what's happening here. Did you install `usd-core` in the isaac sim python environment (this can break things as Isaac Sim comes with it's own usd-core) ?
>
> Try one of the below:
>
> Option [...]

Hi @balakumar-s , I just install `usd-core` in the Isaac Sim Python environment, but it does not work.

```
eai@eai-Z790-UD-AX:~/rp/build_curobo/for_isaac_sim/curobo$ omni_python -m pip install usd-core
Collecting usd-core
  Downloading usd_core-23.11-cp310-none-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (25.8 MB)
     |████████████████████████████████| 25.8 MB 600 kB/s 
Installing collected packages: usd-core
Successfully installed usd-core-23.11
WARNING: You are using pip version 21.2.1+nv1; however, version 23.3.2 is available.
You should consider upgrading via the '/home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/bin/python3 -m pip install --upgrade pip' command.
eai@eai-Z790-UD-AX:~/rp/build_curobo/for_isaac_sim/curobo$ omni_python examples/isaac_sim/multi_arm_reacher.py --robot dual_ur10e.yml
Starting kit application with the following args:  ['/home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1920', '--/app/renderer/resolution/height=1080', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/exts', '--ext-folder', '/home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  ['--robot', 'dual_ur10e.yml']
[Info] [carb] Logging to file: /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/logs/Kit/Isaac-Sim/2023.1/kit_20231228_090404.log
2023-12-28 01:04:04 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.027s] [ext: omni.kit.async_engine-0.0.0] startup
[0.160s] [ext: omni.activity.core-1.0.1] startup
[0.162s] [ext: omni.assets.plugins-0.0.0] startup
[0.163s] [ext: omni.stats-0.0.0] startup
[0.163s] [ext: omni.client-1.0.1] startup
[0.167s] [ext: omni.activity.profiler-1.0.2] startup
[0.168s] [ext: omni.gpu_foundation-0.0.0] startup
[0.173s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.174s] [ext: carb.windowing.plugins-1.0.0] startup
[0.178s] [ext: omni.kit.renderer.init-0.0.0] startup
2023-12-28 01:04:04 [433ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) Graphics (RPL-S)
2023-12-28 01:04:04 [433ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) Graphics (RPL-S)

|---------------------------------------------------------------------------------------------|
| Driver Version: 545.29.06     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA RTX A6000                 | Yes: 0 |     | 49386   MB | 10de      | 0          |
|     |                                  |        |     |            | 2230      | 14a8ce36.. |
|     |                                  |        |     |            | 1         |            |
|---------------------------------------------------------------------------------------------|
| 1   | Intel(R) Graphics (RPL-S)        |        |     | 96430   MB | 8086      | 0          |
|     |                                  |        |     |            | a780      | 21a1fe4f.. |
|     |                                  |        |     |            | 0         |            |
|=============================================================================================|
| OS: 22.04.3 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.3, Kernel: 6.2.0-39-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: 13th Gen Intel(R) Core(TM) i9-13900K | Cores: 24 | Logical: 48
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 128573 | Free Memory: 121345
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
[0.686s] [ext: omni.kit.pipapi-0.0.0] startup
[0.687s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.688s] [ext: omni.pip.compute-1.2.0] startup
[0.688s] [ext: omni.pip.cloud-1.0.1] startup
[0.688s] [ext: omni.isaac.core_archive-2.2.1] startup
[0.689s] [ext: omni.pip.torch-2_0_1-2.0.2] startup
[0.706s] [ext: omni.kit.telemetry-0.5.0] startup
[0.724s] [ext: omni.kit.loop-isaac-1.1.0] startup
[0.725s] [ext: omni.kit.test-0.0.0] startup
[0.726s] [ext: omni.isaac.ml_archive-1.1.3] startup
[0.726s] [ext: omni.mtlx-0.1.0] startup
[... cut ...]
2023-12-28 01:10:45 [161,014ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2023-12-28 01:10:45 [161,016ms] [Warning] [carb.audio.context] 1 contexts were leaked
2023-12-28 01:10:45 [161,039ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
2023-12-28 01:10:45 [161,046ms] [Warning] [omni.core.ITypeFactory] Module /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.
eai@eai-Z790-UD-AX:~/rp/build_curobo/for_isaac_sim/curobo$
```

balakumar-s, 2023-12-28:
I wanted to check if you installed usd-core which you should not. Can you uninstall usd-core?

im-renpei, 2023-12-28:
> I wanted to check if you installed usd-core which you should not. Can you uninstall usd-core?

Hi @balakumar-s , `usd-core` has been remove from Isaac Sim Python environment, but the problem is still the same (`omni_python examples/isaac_sim/multi_arm_reacher.py --robot dual_ur10e.ym`).

```
eai@eai-Z790-UD-AX:~/rp/build_curobo/for_isaac_sim/curobo$ omni_python -m pip uninstall usd-core
Found existing installation: usd-core 23.11
Uninstalling usd-core-23.11:
  Would remove:
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/pxr/*
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core-23.11.dist-info/*
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libboost_python310-4bf84c61.so.1.78.0
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libtbb-3671e415.so.2
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_ar-f362f49a.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_arch-1f71a0ba.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_gf-5f8a4fe0.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_js-0d9c36ea.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_kind-03377bb6.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_ndr-e96f1be7.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_pcp-fe277e2d.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_plug-d7ac7d2b.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_sdf-ec6b5caf.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_sdr-f9e9ed64.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_tf-74256c82.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_trace-03be69d7.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_usd-fc6e088b.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_usdGeom-98925d03.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_usdHydra-32f39d72.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_usdLux-f018e4c3.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_usdMedia-19200e15.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_usdPhysics-feaf1a3b.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_usdProc-f9b0f2b2.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_usdRender-30f27dff.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_usdRi-01a85c2c.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_usdShade-09400a59.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_usdSkel-8529d519.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_usdUI-85a91a62.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_usdUtils-1cef7cf7.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_usdVol-d2f19872.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_vt-e5fa06d1.so
    /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/site-packages/usd_core.libs/libusd_work-feb657a8.so
Proceed (Y/n)? Y
  Successfully uninstalled usd-core-23.11
```

```
[5.582s] [ext: omni.kit.stage_column.payload-1.0.12] startup
[5.583s] [ext: omni.graph.ui_nodes-1.5.10] startup
[5.603s] [ext: omni.sensors.nv.common-1000.0.0-isaacsim.2023.1.1] startup
[5.609s] [ext: omni.sensors.nv.materials-1000.0.0-isaacsim.2023.1.1] startup
[5.610s] [ext: omni.sensors.nv.wpm-1000.0.0-isaacsim.2023.1.1] startup
2023-12-28 01:20:27 [5,603ms] [Error] [carb] [Plugin: libnvwpm2_factory.plugin.so] Could not load the dynamic library from /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/extscache/omni.sensors.nv.wpm-1000.0.0-isaacsim.2023.1.1+lx64.r/bin/libnvwpm2_factory.plugin.so. Error: libnvrtc.so.11.2: cannot open shared object file: No such file or directory (Additional information may be available by running the process with the LD_D [... cut ...]
2023-12-28 01:22:58 [156,628ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2023-12-28 01:22:58 [156,630ms] [Warning] [carb.audio.context] 1 contexts were leaked
2023-12-28 01:22:58 [156,668ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
2023-12-28 01:22:58 [156,675ms] [Warning] [omni.core.ITypeFactory] Module /home/eai/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.
eai@eai-Z790-UD-AX:~/rp/build_curobo/for_isaac_sim/curobo$
```

balakumar-s, 2023-12-28:
I don't know what's happening in your machine. We won't be able to test on Isaac Sim 2023.1.1 soon.

I would advise to use Isaac sim 2023.1.0 where we know for sure that everything works. 

We also advise to use a docker following https://curobo.org/source/getting_started/5_docker_development.html as that helps us reproduce any errors you get.

closed 2023-12-28 by balakumar-s