NVlabs/curobo #94: Using MotionGen.plan_batch() crashes

cremebrule, 2023-12-23:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): isaac sim
2. python version: 3.10.13
3. Isaac Sim version (if using): 2023.1.0

------
## Issue Details
Running `MotionGen.plan_single(cu_js, ik_goal, plan_cfg)`; however, `MotionGen.plan_batch(cu_js_batch, ik_goal_batch, plan_cfg)` crashes, where `cu_js_batch` and `ik_goal_batch` are simply the stacked versions of `cu_js` and `ik_goal` (e.g.: `Pose` consists of `(5,3)`-shaped positions and `(5,4)`-shaped quaternions. The parameters used for instantiating `MotionGen` and `plan_cfg` are identical in both cases.

Is there an example use case in IsaacSim that I can compared / debug against? There's `batch_motion_gen_reacher.py` but that utilizes multiple environments. I have a single environment and would just like to evaluate multiple goal poses in parallel for that single world.

Huge thanks!

balakumar-s, 2023-12-23:
Here is an example: https://github.com/NVlabs/curobo/blob/c09d94908d8ad52f557ed59195c1ceb2d0434d65/examples/motion_gen_example.py#L222

If you want to generate one trajectory that reaches one of the goals (goalset planning), you can also try motion_gen.plan_goalset: https://github.com/NVlabs/curobo/blob/c09d94908d8ad52f557ed59195c1ceb2d0434d65/tests/motion_gen_module_test.py#L109

What does the crash report?

cremebrule, 2023-12-23:
Huge thanks for the rapid response! I'll try out these examples, compare, and report back.

Here is the current error I've been getting:

```
terminate called after throwing an instance of 'c10::Error'
  what():  uc >= 0 INTERNAL ASSERT FAILED at "../c10/cuda/CUDACachingAllocator.cpp":1367, please report a bug to PyTorch. 
Exception raised from notifyCaptureDestroy at ../c10/cuda/CUDACachingAllocator.cpp:1367 (most recent call first):
frame #0: c10::Error::Error(c10::SourceLocation, std::string) + 0x57 (0x7fabab1484d7 in /scr/jdwong/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/lib/libc10.so)
frame #1: c10::detail::torchCheckFail(char const*, char const*, unsigned int, char const*) + 0x68 (0x7fabab112434 in /scr/jdwong/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/lib/libc10.so)
frame #2: <unknown function> + 0x27f6f (0x7fabab1c0f6f in /scr/jdwong/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/lib/libc10_cuda.so)
frame #3: at::cuda::CUDAGraph::reset() + 0x3f (0x7fabac4e73df in /scr/jdwong/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/lib/libtorch_cuda.so)
frame #4: at::cuda::CUDAGraph::~CUDAGraph() + 0xe (0x7fabac4e7c7e in /scr/jdwong/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/lib/libtorch_cuda.so)
frame #5: std::_Sp_counted_ptr<at::cuda::CUDAGraph*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() + 0x12 (0x7fac12186a62 in /scr/jdwong/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/lib/libtorch_python.so)
frame #6: <unknown function> + 0xb17b40 (0x7fac12186b40 in /scr/jdwong/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/lib/libtorch_python.so)
frame #7: <unknown function> + 0x57c2a9 (0x7facceabe2a9 in /cvgl2/u/jdwong/miniforge3/envs/s4r/lib/python3.10/site-packages/open3d/cpu/pybind.cpython-310-x86_64-linux-gnu.so)
<omitting python frames>
frame #52: __libc_start_main + 0xf3 (0x7facdae34083 in /lib/x86_64-linux-gnu/libc.so.6)

Fatal Python error: Aborted

Thread 0x00007f9b8ffff700 (most recent call first):
  <no Python frame>

Thread 0x00007fa95d3cd700 (most recent call first):
  File "/cvgl2/u/jdwong/miniforge3/envs/s4r/lib/python3.10/concurrent/futures/thread.py", line 81 in _worker
  File "/cvgl2/u/jdwong/miniforge3/envs/s4r/lib/python3.10/threading.py", line 953 in run
  File "/cvgl2/u/jdwong/miniforge3/envs/s4r/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
  File "/cvgl2/u/jdwong/miniforge3/envs/s4r/lib/python3.10/threading.py", line 973 in _bootstrap

Current thread 0x00007facdae03740 (most recent call first):
  File "/cvgl2/u/jdwong/PAIR/curobo/src/curobo/opt/newton/newton_base.py", line 537 in _create_opt_iters_graph
  File "/cvgl2/u/jdwong/PAIR/curobo/src/curobo/opt/newton/newton_base.py", line 491 in _initialize_opt_iters_graph
  File "/cvgl2/u/jdwong/PAIR/curobo/src/curobo/opt/newton/newton_base.py", line 142 in _optimize
  File "/cvgl2/u/jdwong/PAIR/curobo/src/curobo/opt/opt_base.py", line 96 in optimize
  File "/cvgl2/u/jdwong/PAIR/curobo/src/curobo/wrap/wrap_base.py", line 71 in optimize
  File "/cvgl2/u/jdwong/PAIR/curobo/src/curobo/wrap/wrap_base.py", line 139 in solve
  File "/cvgl2/u/jdwong/PAIR/curobo/src/curobo/wrap/reacher/ik_solver.py", line 675 in solve_from_solve_state
  File "/cvgl2/u/jdwong/PAIR/curobo/src/curobo/wrap/reacher/ik_solver.py", line 537 in solve_batch
  File "/cvgl2/u/jdwong/PAIR/curobo/src/curobo/wrap/reacher/ik_solver.py", line 406 in solve_any
  File "/cvgl2/u/jdwong/PAIR/curobo/src/curobo/wrap/reacher/motion_gen.py", line 882 in _solve_ik_from_solve_state
  File "/cvgl2/u/jdwong/miniforge3/envs/s4r/lib/python3.10/contextlib.py", line 79 in inner
  File "/cvgl2/u/jdwong/PAIR/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1812 in _plan_from_solve_state_batch
  File "/cvgl2/u/jdwong/PAIR/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1142 in _plan_batch_attempts
  File "/cvgl2/u/jdwong/PAIR/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1261 in plan_batch
  File "/cvgl2/u/jdwong/PAIR/doppelmaker/doppelmaker/utils/curobo_utils.py", line 267 in compute_trajectory
  File "/cvgl2/u/jdwong/PAIR/doppelmaker/tests/test_curobo_with_grasp.py", line 267 in <module>

Extension modules: numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, markupsafe._speedups, sklearn.__check_build._check_build, scipy._lib._ccallback_c, scipy.sparse._sparsetools, _csparsetools, scipy.sparse._csparsetools, scipy.sparse.linalg._isolve._iterative, scipy.linalg._fblas, scipy.linalg._flapack, scipy.linalg._cythonized_array_utils, scipy.linalg._flinalg, scipy.linalg._solve_toeplitz, scipy.linalg._matfuncs_sqrtm_triu, scipy.linalg.cython_lapack, scipy.linalg.cython_blas, scipy.linalg._matfuncs_expm, scipy.linalg._decomp_update, scipy.sparse.linalg._dsolve._superlu, scipy.sparse.linalg._eigen.arpack._arpack, scipy.sparse.csgraph._tools, scipy.sparse.csgraph._shortest_path, scipy.sparse.csgraph._traversal, scipy.sparse.csgraph._min_spanning_tree, scipy.sparse.csgraph._flow, scipy.sparse.csgraph._matching, scipy.sparse.csgraph._re [... cut ...] ffi_backend, frozenlist._frozenlist, scipy.io.matlab._mio_utils, scipy.io.matlab._streams, scipy.io.matlab._mio5_utils, xxhash._xxhash, embreex.rtcore, embreex.rtcore_scene, embreex.mesh_construction, lxml._elementpath, lxml.etree, shapely.lib, shapely._geos, shapely._geometry_helpers, numba.core.typeconv._typeconv, numba._helperlib, numba._dynfunc, numba._dispatcher, numba.core.runtime._nrt_python, numba.np.ufunc._internal, numba.experimental.jitclass._box (total: 253)
Aborted (core dumped)
```

balakumar-s, 2023-12-23:
Can you comment out motion_gen.warmup() in your code? This happens when you change the batch size or planning mode after warmup

cremebrule, 2023-12-23:
Ahhh interesting -- just tried now, and it seems to be working! Huge thanks for the fix.

So when should `motion_gen.warmup()` be used? I assumed it was necessary all the time, but perhaps it should only be used for single env / single batches?

If I don't call warmup, can I still call different motion_gen methods, e.g.: `plan_single` and `plan_batch` and `plan_goalset` within the same runtime?

Huge thanks for the assistance!!

balakumar-s, 2023-12-27:
`motion_gen.warmup()` can be used for single env or batch. But the batch size needs to be known before calling warmup(). 

cuRobo uses cuda graphs to compute motions. Hence, we cannot currently change the problem type or tensor shapes once initialized. You can use cuRobo without cuda graph by initializing `motion_gen.load_from_robot_config(use_cuda_graph=False, ..)`. This would allow you to change the problem type and batch size between calls. However, the motion generation compute time could be 10x slower.

closed 2023-12-27 by balakumar-s