NVlabs/curobo #106: plan_config's clone() function will let plan_single to crush and leaving no error trace or clue

zoctipus, 2023-12-30:
It took me entire day to finally find out it was the clone() function of motion_gen.plan_single that caused the entire script to shut done completely without returning any backtrace or error message. This left me clueless for most of day and wanted to give up several times..... But luckily I found it

The same error can be reproduced in shipped script examples/isaac_sim/motion_gen_reacher.py by simplely changing

```
result = motion_gen.plan_single(cu_js.unsqueeze(0), ik_goal, plan_config) ->
```
```
result = motion_gen.plan_single(cu_js.unsqueeze(0), ik_goal, plan_config.clone())
```

I have also tested this for plan_batch_env, it seems like for plan_batch_env, either self.plan_config or self.plan_config.clone() is fine.

balakumar-s, 2024-01-12:
I could not reproduce this in Isaac Sim 2023.1.0.

What version of Isaac sim did you find this issue? Also did you use this inside a docker?

zoctipus, 2024-01-15:
Thanks for getting back, I tested again,
the version of isaac sim I used is 2022.2.0, 
I did not use in a docker.

balakumar-s, 2024-01-15:
We only recommend 2022.2.1+ versions for isaac sim.

balakumar-s, 2024-01-23:
Thanks for reporting this, I actually found the issue. It was because not all elements were cloned in MotionGenPlanConfig.clone() function. I will push a fix for this.

balakumar-s, 2024-01-25:
We pushed a fix for this in the latest commit.

closed 2024-01-25 by balakumar-s