NVlabs/curobo #109: 'cudafe++' died due to signal 9 (Kill signal) when building x86 docker

BolunDai0216, 2024-01-03:
1. cuRobo installation mode (docker python):
2. python version: System Python is 3.10
3. Isaac Sim version (if using): 

------
Issue Details

When running

```bash
bash build_docker.sh x86
```

for the step

```bash
RUN cd /pkgs &&  git clone https://github.com/valtsblukis/nvblox.git && \
    cd nvblox && cd nvblox && mkdir build && cd build && \
    cmake .. -DPRE_CXX11_ABI_LINKABLE=ON && \
    make -j32 && \
    make install
```

I get the error:

```bash
25.88 [ 23%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/integrators/projective_occupancy_integrator.cu.o
25.88 [ 24%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/integrators/esdf_slicer.cu.o
25.89 [ 24%] Building CXX object CMakeFiles/nvblox_lib.dir/src/io/layer_cake_io.cpp.o
25.89 [ 25%] Building CXX object CMakeFiles/nvblox_lib.dir/src/io/ply_writer.cpp.o
26.26 [ 26%] Building CXX object CMakeFiles/nvblox_lib.dir/src/io/pointcloud_io.cpp.o
26.53 [ 26%] Building CXX object CMakeFiles/nvblox_lib.dir/src/io/image_io.cpp.o
27.07 [ 27%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/mesh/mesh_block.cu.o
27.46 [ 27%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/mesh/mesh_integrator_color.cu.o
28.05 [ 28%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/mesh/mesh_integrator.cu.o
28.31 [ 28%] Building CXX object CMakeFiles/nvblox_lib.dir/src/mesh/mesh.cpp.o
28.38 [ 29%] Building CXX object CMakeFiles/nvblox_lib.dir/src/mesh/mesh_streamer.cpp.o
31.55 nvcc error   : 'cudafe++' died due to signal 9 (Kill signal)
31.58 make[2]: *** [CMakeFiles/nvblox_lib.dir/build.make:366: CMakeFiles/nvblox_lib.dir/src/integrators/tsdf_decay_integrator.cu.o] Error 9
31.58 make[2]: *** Waiting for unfinished jobs....
32.33 nvcc error   : 'cudafe++' died due to signal 9 (Kill signal)
32.34 make[2]: *** [CMakeFiles/nvblox_lib.dir/build.make:150: CMakeFiles/nvblox_lib.dir/src/map/blox.cu.o] Error 9
32.77 nvcc error   : 'cudafe++' died due to signal 9 (Kill signal)
32.77 make[2]: *** [CMakeFiles/nvblox_lib.dir/build.make:222: CMakeFiles/nvblox_lib.dir/src/sensors/pointcloud.cu.o] Error 9
33.20 nvcc error   : 'cudafe++' died due to signal 9 (Kill signal)
33.20 make[2]: *** [CMakeFiles/nvblox_lib.dir/build.make:336: CMakeFiles/nvblox_lib.dir/src/integrators/view_calculator.cu.o] Error 9
50.82 nvcc error   : 'cudafe++' died due to signal 9 (Kill signal)
50.83 make[2]: *** [CMakeFiles/nvblox_lib.dir/build.make:381: CMakeFiles/nvblox_lib.dir/src/integrators/projective_occupancy_integrator.cu.o] Error 9
61.17 nvcc error   : 'cudafe++' died due to signal 9 (Kill signal)
61.18 make[2]: *** [CMakeFiles/nvblox_lib.dir/build.make:396: CMakeFiles/nvblox_lib.dir/src/integrators/projective_tsdf_integrator.cu.o] Error 9
70.05 nvcc error   : 'cudafe++' died due to signal 9 (Kill signal)
70.06 make[2]: *** [CMakeFiles/nvblox_lib.dir/build.make:351: CMakeFiles/nvblox_lib.dir/src/integrators/occupancy_decay_integrator.cu.o] Error 9
91.35 nvcc error   : 'cudafe++' died due to signal 9 (Kill signal)
91.36 make[2]: *** [CMakeFiles/nvblox_lib.dir/build.make:411: CMakeFiles/nvblox_lib.dir/src/integrators/projective_color_integrator.cu.o] Error 9
106.4 nvcc error   : 'cudafe++' died due to signal 9 (Kill signal)
106.4 make[2]: *** [CMakeFiles/nvblox_lib.dir/build.make:237: CMakeFiles/nvblox_lib.dir/src/sensors/image.cu.o] Error 9
124.3 nvcc error   : 'cudafe++' died due to signal 9 (Kill signal)
124.4 make[2]: *** [CMakeFiles/nvblox_lib.dir/build.make:165: CMakeFiles/nvblox_lib.dir/src/map/layer.cu.o] Error 9
166.9 nvcc error   : 'cudafe++' died due to signal 9 (Kill signal)
166.9 make[2]: *** [CMakeFiles/nvblox_lib.dir/build.make:471: CMakeFiles/nvblox_lib.dir/src/rays/sphere_tracer.cu.o] Error 9
170.1 In file included from /pkgs/nvblox/nvblox/include/nvblox/core/unified_ptr.h:217,
170.1                  from /pkgs/nvblox/nvblox/include/nvblox/sensors/image.h:22,
170.1                  from /pkgs/nvblox/nvblox/include/nvblox/io/image_io.h:21,
170.1                  from /pkgs/nvblox/nvblox/src/io/image_io.cpp:1:
170.1 In member function ‘nvblox::unified_ptr<T>& nvblox::unified_ptr<T>::operator=(nvblox::unified_ptr<T>&&) [with T = unsigned char []]’,
170.1     inlined from ‘nvblox::Image<_ElementType>& nvblox::Image<_ElementType>::operator=(nvblox::Image<_ElementType>&&) [with _ElementType = unsigned char]’ at /pkgs/nvblox/nvblox/include/nvblox/sensors/internal/impl/image_impl.h:48:15,
170.1     inlined from ‘bool nvblox::io::writeToPngTemplate(const string&, const ImageType&) [with ImageType = nvblox::Image<unsigned char>]’ at /pkgs/nvblox/nvblox/src/io/image_io.cpp:41:9:
170.1 /pkgs/nvblox/nvblox/include/nvblox/core/internal/impl/unified_ptr_impl.h:194:17: warning: ‘<unnamed>.nvblox::Image<unsigned char>::owned_data_.nvblox::unified_ptr<unsigned char []>::size_’ may be used uninitialized [-Wmaybe-uninitialized]
170.1   194 |   size_ = other.size_;
170.1       |           ~~~~~~^~~~~
170.1 /pkgs/nvblox/nvblox/src/io/image_io.cpp: In function ‘bool nvblox::io::writeToPngTemplate(const string&, const ImageType&) [with ImageType = nvblox::Image<unsigned char>]’:
170.1 /pkgs/nvblox/nvblox/src/io/image_io.cpp:41:11: note: ‘<anonymous>’ declared here
170.1    41 |     tmp = ImageType(MemoryType::kHost);
170.1       |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
170.1 In file included from /pkgs/nvblox/nvblox/include/nvblox/core/unified_ptr.h:217,
170.1                  from /pkgs/nvblox/nvblox/include/nvblox/sensors/image.h:22,
170.1                  from /pkgs/nvblox/nvblox/include/nvblox/io/image_io.h:21,
170.1                  from /pkgs/nvblox/nvblox/src/io/image_io.cpp:1:
170.1 In member function ‘nvblox::unified_ptr<T>& nvblox::unified_ptr<T>::operator=(nvblox::unified_ptr<T>&&) [with T = nvblox::Color []]’,
170.1     inlined from ‘nvblox::Image<_ElementType>& nvblox::Image<_ElementType>::operator=(nvblox::Image<_ElementType>&&) [with _ElementType = nvblox::Color]’ at /pkgs/nvblox/nvblox/include/nvblox/sensors/internal/impl/image_impl.h:48:15,
170.1     inlined from ‘bool nvblox::io::writeToPngTemplate(const string&, const ImageType&) [with ImageType = nvblox::Image<nvblox::Color>]’ at /pkgs/nvblox/nvblox/src/io/image_io.cpp:41:9:
170.1 /pkgs/nvblox/nvblox/include/nvblox/core/internal/impl/unified_ptr_impl.h:194:17: warning: ‘<unnamed>.nvblox::Image<nvblox::Color>::owned_data_.nvblox::unified_ptr<nvblox::Color []>::size_’ may be used uninitialized [-Wmaybe-uninitialized]
170.1   194 |   size_ = other.size_;
170.1       |           ~~~~~~^~~~~
170.1 /pkgs/nvblox/nvblox/src/io/image_io.cpp: In function ‘bool nvblox::io::writeToPngTemplate(const string&, const ImageType&) [with ImageType = nvblox::Image<nvblox::Color>]’:
170.1 /pkgs/nvblox/nvblox/src/io/image_io.cpp:41:11: note: ‘<anonymous>’ declared here
170.1    41 |     tmp = ImageType(MemoryType::kHost);
170.1       |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
177.8 make[1]: *** [CMakeFiles/Makefile2:437: CMakeFiles/nvblox_lib.dir/all] Error 2
177.8 make: *** [Makefile:146: all] Error 2
------
x86.dockerfile:113
--------------------
 112 |     
 113 | >>> RUN cd /pkgs &&  git clone https://github.com/valtsblukis/nvblox.git && \
 114 | >>>     cd nvblox && cd nvblox && mkdir build && cd build && \
 115 | >>>     cmake .. -DPRE_CXX11_ABI_LINKABLE=ON && \
 116 | >>>     make -j32 && \
 117 | >>>     make install
 118 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c cd /pkgs &&  git clone https://github.com/valtsblukis/nvblox.git &&     cd nvblox && cd nvblox && mkdir build && cd build &&     cmake .. -DPRE_CXX11_ABI_LINKABLE=ON &&     make -j32 &&     make install" did not complete successfully: exit code: 2
```

Anyone knows what might be the issue?

Thanks in advance!

balakumar-s, 2024-01-03:
It looks like you don't have access to cuda runtime during docker build. Did you complete step 1 of https://curobo.org/source/getting_started/5_docker_development.html#building-your-own-docker-image-with-curobo ?

BolunDai0216, 2024-01-03:
Thanks for the quick response!

I did complete the first step, but it seems like the issue stems from how I installed docker. After I reinstalled Ubuntu and followed the steps to install docker engine (not docker desktop), I was able to build the docker image.

I will close this issue.

closed 2024-01-03 by BolunDai0216