NVlabs/curobo #112: Issue with Updating Locked Joint State in motion_gen.py

Hermanye996, 2024-01-08:
1. cuRobo installation mode: python
2. python version:Python 3.8.10
3. curobo v0.6.2

------
```bash
    def update_locked_state(js: JointState):
        # update the joint values of locked joints:

        pass
```
I am currently working with a robotic arm that is mounted on a sliding joint. For my use case, I need to lock the sliding joint while executing the motion planning for the robotic arm. 

However, I also need to update the value of the sliding joint when I want to move the position of the robotic arm, and then lock it again.

I noticed the `update_locked_state` function in the `motion_gen.py` file, which seems to be intended for this purpose. However, it appears that this function is not yet implemented.

Could you please provide some guidance on how to properly update the locked state of the sliding joint? Any help would be greatly appreciated.

Thank you for your time and consideration.

balakumar-s, 2024-01-10:
We have had several users request this feature. We are currently implementing this so that you can dynamically change the joint angle of a locked joint. We will have it out by the end of this week.

Hermanye996, 2024-01-11:
Thanks a lot!

Hermanye996, 2024-02-21:
> We have had several users request this feature. We are currently implementing this so that you can dynamically change the joint angle of a locked joint. We will have it out by the end of this week.

```python
    def update_locked_state(js: JointState):
        # update the joint values ​​of locked joints:

        pass
```
I found that the current version of curobo still does not support dynamic adjustment of fixed joint values. When will this function be implemented? Thanks.

balakumar-s, 2024-02-23:
Here is an example: https://github.com/NVlabs/curobo/blob/main/tests/motion_gen_api_test.py#L64

closed 2024-02-23 by balakumar-s