NVlabs/curobo #113: [BUG] MotionGen.plan_batch() crashed when failed to find an IK solution

TooSchoolForCool, 2024-01-15:
1. cuRobo installation mode: I have tried python, docker python, docker issac sim. All have this issue.
2. python version: python 3.8 (w.o docker), python 3.10 (with docker)
3. Isaac Sim version (if using): 2023.1.0

------

## Reproduce Issue

Take the script [examples/motion_gen_example.py](https://github.com/NVlabs/curobo/blob/main/examples/motion_gen_example.py) as an example: change [line 249](https://github.com/NVlabs/curobo/blob/c09d94908d8ad52f557ed59195c1ceb2d0434d65/examples/motion_gen_example.py#L249) to:
```python
retract_pose.position[0, 0] = -30.3
```
This will have the first planner problem fail to find a IK solution, then the program will crash.

Then run `python examples/motion_gen_example.py`, we will get
```text
Traceback (most recent call last):
  File "examples/motion_gen_example.py", line 377, in <module>
    demo_motion_gen_batch()
  File "examples/motion_gen_example.py", line 250, in demo_motion_gen_batch
    result = motion_gen.plan_batch(
  File "/home/zeyu/Libs/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1261, in plan_batch
    result = self._plan_batch_attempts(
  File "/home/zeyu/Libs/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1142, in _plan_batch_attempts
    result = self._plan_from_solve_state_batch(
  File "/home/zeyu/Libs/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1978, in _plan_from_solve_state_batch
    traj_result = self._solve_trajopt_from_solve_state(
  File "/usr/lib/python3.8/contextlib.py", line 75, in inner
    return func(*args, **kwds)
  File "/home/zeyu/Libs/curobo/src/curobo/wrap/reacher/motion_gen.py", line 912, in _solve_trajopt_from_solve_state
    traj_result = trajopt_instance.solve_any(
  File "/home/zeyu/Libs/curobo/src/curobo/wrap/reacher/trajopt.py", line 510, in solve_any
    return self.solve_batch(
  File "/home/zeyu/Libs/curobo/src/curobo/wrap/reacher/trajopt.py", line 670, in solve_batch
    return self.solve_from_solve_state(
  File "/home/zeyu/Libs/curobo/src/curobo/wrap/reacher/trajopt.py", line 571, in solve_from_solve_state
    seed_traj = self.get_seed_set(
  File "/home/zeyu/Libs/curobo/src/curobo/wrap/reacher/trajopt.py", line 1069, in get_seed_set
    seed_traj = seed_traj.view(
RuntimeError: shape '[2, 28, 7]' is invalid for input of size 196
```

## My investigation

This happened only when `enable_graph` is `True`. Although in the example script, `enable_graph` is set to `False`, the `enable_graph_attempt = 1` will have the planner use the graph when the first trail of optimization failed. Change the `MotionGenPlanConfig` will not have this issue:
```python 
MotionGenPlanConfig(
    max_attempts=5, enable_graph=False, enable_graph_attempt=None, enable_opt=True
)
```
The program will failed to find a solution for the first problem, and will not crash.

The core problem is that when graph is enabled, if the some problem in a batch failed to find a solution, the dimension of the whole batch does not match the dimension of `trajopt_seed` generated from the graph. See [src/curobo/wrap/reach/motion_gen.py#line1874](https://github.com/NVlabs/curobo/blob/c09d94908d8ad52f557ed59195c1ceb2d0434d65/src/curobo/wrap/reacher/motion_gen.py#L1874), in the example I gave, when the first goal fail to find an IK solution, the value of `graph_success` is 1 which does not equal to the `total_seeds` (which is 2) for this problem.

j-hae1, 2024-03-08:
Did you solve this problem?

TooSchoolForCool, 2024-03-11:
> Did you solve this problem?

Yes, I do.

j-hae1, 2024-03-12:
You solved the problem by setting the parameter as above 

```
MotionGenPlanConfig(
    max_attempts=5, enable_graph=False, enable_graph_attempt=None, enable_opt=True
)
```

is this right?

balakumar-s, 2024-03-12:
@j-hae1  are you seeing this issue in the latest commit?

j-hae1, 2024-03-14:
yes. I still watch the same issue

balakumar-s, 2024-03-18:
We pushed a fix. Can you check if it works for you?

closed 2024-03-23 by balakumar-s