NVlabs/curobo #120: Cannot generate collision spheres using built-in franka URDF

BolunDai0216, 2024-01-20:
IPlease provide the below information in addition to your issue:

1. cuRobo installation mode: isaac sim
2. python version: 3.10
3. Isaac Sim version (if using): 2023.1.1

------
Issue Details

I was trying to learn how to configure a new robot, since I could not run the provided URDF to USD conversion script (see #119), I wanted to manually convert a URDF to USD and then perform the [collision sphere generation](https://curobo.org/source/tutorials/1_robot_configuration.html#robot-collision-representation). However, when playing around with the built-in franka URDF from Isaac Sim, after generating the USD, I cannot generate the collision spheres.

I noticed that instead of `/panda` being the articulation, I am getting `/panda/base_link`. The select link drop-down menu is empty. Does the URDF have to follow some convention that Isaac Sim itself is even not following?

Any suggestions would be greatly appreciated! Thanks in advance!

balakumar-s, 2024-01-20:
Did you click play after loading the sphere generation extension loads?

BolunDai0216, 2024-01-20:
Yes I did.

balakumar-s, 2024-01-20:
Can you try with isaac sim 2023.1.0?

BolunDai0216, 2024-01-20:
Is it an issue to install different versions of Isaac Sim on the same computer?

balakumar-s, 2024-01-20:
No issues, they will be installed in separate locations.

BolunDai0216, 2024-01-20:
Should I try the hotfix or 2023.1.0?

BolunDai0216, 2024-01-20:
Isaac Sim 2023.1.0 does not have this issue.

balakumar-s, 2024-01-20:
Okay, we haven't yet tested on 2023.1.1. We will add a note in the instructions.

BolunDai0216, 2024-01-20:
Thank you so much for the help!

closed 2024-01-20 by BolunDai0216