NVlabs/curobo #126: Failed to acquire IWindowing interface when Running CuRobo in Docker

emfebert, 2024-01-22:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode: docker isaac sim:
2. python version: /isaac-sim/python.sh  which is Python 3.10.13
3. Isaac Sim version (if using):  isaac_sim_2023.1.0

------

Running the example with the command `"/pkgs/curobo# /isaac-sim/python.sh examples/isaac_sim/motion_gen_reacher.py"` fails with the error:

```
2024-01-22 21:58:42 [1,162ms] [Error] [omni.appwindow.plugin] Failed to acquire IWindowing interface
[1.173s] [ext: omni.kit.renderer.core-0.0.0] startup
Fatal Python error: Segmentation fault

Current thread 0x00007feb9e5451c0 (most recent call first):
  File "/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 303 in _start_app
  File "/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 192 in __init__
  File "/pkgs/curobo/examples/isaac_sim/motion_gen_reacher.py", line 61 in <module>
```

I'd like to run issac_sim with a native gui from this docker container if possible.

Thanks!

balakumar-s, 2024-01-22:
Try:
1. `xhost +` outside your docker
2. Inside docker, `cd /isaac-sim` and then `./runapp.sh`

Does isaac sim open after running this?

emfebert, 2024-01-22:
After starting the native gui successfully, I opened another terminal in the same docker and used `"/pkgs/curobo# /isaac-sim/python.sh examples/isaac_sim/motion_gen_reacher.py"` again, but it's now running out of RAM (I have 32GB total RAM, and 24GB GPU-VRAM).

balakumar-s, 2024-01-22:
Can you close the native gui and try only running the motion_gen_reacher example?

emfebert, 2024-01-22:
That worked. You don't need do "./runapp.sh".
It took about 2-3 minutes to load is that normal?

balakumar-s, 2024-01-22:
It should not take more than 1 minute to load. Maybe because it's the first time you are launching and assets need to be downloaded.

closed 2024-01-22 by balakumar-s

emfebert, 2024-01-23:
Just timed it, it's 3 minutes every time.

balakumar-s, 2024-01-23:
What CPU and GPU are you using?

emfebert, 2024-01-23:
CPU: AMD Ryzen 9 5900X 12-core, 24-Thread
GPU: GeForce RTX 3090

emfebert, 2024-01-23:
Looks like the cache is disabled in the docker container, how can I turn it on?