NVlabs/curobo #130: RuntimeError: element 0 of tensors does not require grad and does not have a grad_fn

zoctipus, 2024-01-23:
1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]):
   Isaac sim
2. python version:
   python 3.10
3. Isaac Sim version (if using): 
   Isaac sim 23.1.0

------

I entered some very strange bug, I will try my best to continue debug and make reproducible

The strange thing is that I wrote a simple environment to test that plan_batch works, it worked.
Then I made a more complex environment, and copied and pasted the curobo part of the code exactly the same.
I check all the input to curobo motion_gen the correct as what I had for the simpler environment, and made sure all the 
state input into curobo are tensor copies, not references.
But for some reason
'''
File "~/Projects/Orbit2310/rlrf/curobo_controller.py", line 375, in _plan
    result = self.motion_gen.plan_batch_env(cu_js, ik_goal, self.plan_config)
'''

fails because
```
 File "~/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/autograd/__init__.py", line 200, in backward
    Variable._execution_engine.run_backward(  # Calls into the C++ engine to run the backward pass
RuntimeError: element 0 of tensors does not require grad and does not have a grad_fn
```
While I will also keep look into it. It will be greatly appreciated if you can provide insight what may be the potential issue. 
Big thanks, as always.

```
2024-01-23 18:29:09 [128,568ms] [Error] [__main__] element 0 of tensors does not require grad and does not have a grad_fn
2024-01-23 18:29:10 [129,467ms] [Error] [__main__] Traceback (most recent call last):
  File "~/Projects/Orbit2310/rlrf/rsl_rl/train.py", line 138, in <module>
    main()
  File "~/Projects/Orbit2310/rlrf/rsl_rl/train.py", line 129, in main
    runner.learn(num_learning_iterations=agent_cfg.max_iterations, init_at_random_ep_len=True)
  File "~/anaconda3/envs/orbit2310/lib/python3.10/site-packages/rsl_rl/runners/on_policy_runner.py", line 112, in learn
    obs, rewards, dones, infos = self.env.step(actions)
  File "~/Projects/Orbit2310/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/utils/wrappers/rsl_rl/vecenv_wrapper.py", line 162, in step
    obs_dict, rew, terminated, truncated, extras = self.env.step(actions)
  File "~/anaconda3/envs/orbit2310/lib/python3.10/site-packages/gymnasium/wrappers/order_enforcing.py", line 56, in step
    return self.env.step(action)
  File "~/Projects/Orbit2310/rlrf/rl_controller_task_env.py", line 152, in step
    action = self.curobo_controller.compute(
  File "~/Projects/Orbit2310/rlrf/curobo_controller.py", line 428, in compute
    result = self._plan(
  File "~/Projects/Orbit2310/rlrf/curobo_controller.py", line 375, in _plan
    result = self.motion_gen.plan_batch_env(cu_js, ik_goal, self.plan_config)
  File "~/Projects/Orbit2310/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1314, in plan_batch_env
    result = self._plan_batch_attempts(
  File "~/Projects/Orbit2310/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1142, in _plan_batch_attempts
    result = self._plan_from_solve_state_batch(
  File "~/Projects/Orbit2310/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1812, in _plan_from_solve_state_batch
    ik_result = self._solve_ik_from_solve_state(
  File "~/anaconda3/envs/orbit2310/lib/python3.10/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "~/Projects/Orbit2310/curobo/src/curobo/wrap/reacher/motion_gen.py", line 882, in _solve_ik_from_solve_state
    ik_result = self.ik_solver.solve_any(
  File "~/Projects/Orbit2310/curobo/src/curobo/wrap/reacher/ik_solver.py", line 427, in solve_any
    return self.solve_batch_env(
  File "~/Projects/Orbit2310/curobo/src/curobo/wrap/reacher/ik_solver.py", line 607, in solve_batch_env
    return self.solve_from_solve_state(
  File "~/Projects/Orbit2310/curobo/src/curobo/wrap/reacher/ik_solver.py", line 675, in solve_from_solve_state
    result = self.solver.solve(goal_buffer, coord_position_seed)
  File "~/Projects/Orbit2310/curobo/src/curobo/wrap/wrap_base.py", line 141, in solve
    act_seq = self.optimize(seed, shift_steps=0)
  File "~/Projects/Orbit2310/curobo/src/curobo/wrap/wrap_base.py", line 71, in optimize
    act_seq = opt.optimize(act_seq, shift_steps)
  File "~/Projects/Orbit2310/curobo/src/curobo/opt/opt_base.py", line 96, in optimize
    out = self._optimize(opt_tensor, shift_steps, n_iters)
  File "~/Projects/Orbit2310/curobo/src/curobo/opt/newton/newton_base.py", line 144, in _optimize
    best_q, best_cost, q, grad_q = self._call_opt_iters_graph(q, grad_q)
  File "~/Projects/Orbit2310/curobo/src/curobo/opt/newton/newton_base.py", line 521, in _call_opt_iters_graph
    return self._opt_iters(q, grad_q)
  File "~/Projects/Orbit2310/curobo/src/curobo/opt/newton/newton_base.py", line 170, in _opt_iters
    cost_n, q, grad_q = self._opt_step(q.detach(), grad_q.detach())
  File "~/Projects/Orbit2310/curobo/src/curobo/opt/newton/newton_base.py", line 182, in _opt_step
    q_n, cost_n, grad_q_n = self._approx_line_search(q, grad_q)
  File "~/Projects/Orbit2310/curobo/src/curobo/opt/newton/newton_base.py", line 425, in _approx_line_search
    return self._wolfe_line_search(x, step_direction)
  File "~/Projects/Orbit2310/curobo/src/curobo/opt/newton/newton_base.py", line 252, in _wolfe_line_search
    c, g_x = self._compute_cost_gradient(x_set)
  File "~/Projects/Orbit2310/curobo/src/curobo/opt/newton/newton_base.py", line 233, in _compute_cost_gradient
    g_x = cost.backward(gradient=self.l_vec, retain_graph=False)
  File "~/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/_tensor.py", line 487, in backward
    torch.autograd.backward(
  File "~/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/autograd/__init__.py", line 200, in backward
    Variable._execution_engine.run_backward(  # Calls into the C++ engine to run the backward pass
RuntimeError: element 0 of tensors does not require grad and does not have a grad_fn
```

Thank you!
Octi.

zoctipus, 2024-01-24:
After some testing,
I found out the issue not only exist when plan_single or plan_batch, it also exits in warmup 
it seems to come from 
file: **newton_base.py**  
function: **_compute_cost_gradient(self, x):**

in the debugger, 
inspecting x.requires_grad = True
inspecting x_n.requires_grad = True
inspecting x_in.requires_grad = **False**

while the success example has :
inspecting x.requires_grad = True
inspecting x_n.requires_grad = True
inspecting x_in.requires_grad = True

It is very curious that view function will strip away the requires_grad propery of the variable under some unsure conditions
Below is the gif demostration comparing the failure case vs success case

fail:
[image]

success:
[image]

zoctipus, 2024-01-25:
I don't know what exactly caused it, but luckily I found a work around solution that seems to avoid such issue. I will keep here updated if any more news comes

zoctipus, 2024-01-25:
Is there some function that can print the state of curobo?
It is really peculiar that x_in's requires_grad becomes false with x_n.view(). If there are function that let me inspect the current state such that I can compare the state difference between success example and failure example, and perhaps understand the bug at a better details..

balakumar-s, 2024-01-25:
Not sure what the issue is. One thing could be because you didn't initialize torch before isaac sim. In your main file, can you add these two lines at the very beginning: 

https://github.com/NVlabs/curobo/blob/88eac64edca2d2b641c1622e63f3e750157596c4/examples/isaac_sim/motion_gen_reacher.py#L14

zoctipus, 2024-01-25:
I think I know The reason now!!

RSL_RL(a rl package) has the structure of 
```
with torch.inference_mode():
   ....
   obs, rewards, dones, infos = self.env.step(goal_pose)
   ....
```
and I had `curobo.plan_batch_env()` within the `env.step` function, meaning that curobo's planning is done in inference mode.

I removed `with torch.inference_mode():`
Is there a way to enable 'grad_enabled_mode' for the curobo's planning within torch's inference_mode?

I have tried
```
with torch.inference_mode():
   ....
   with torch.enable_grad():
        obs, rewards, dones, infos = self.env.step(goal_pose)
   ....
```
but this doesn't work, the grad will be lost during the process like from x_n to x_in

zoctipus, 2024-01-25:
I unelegantly solved with 
with torch.inference_mode():
       ....#other code
obs, rewards, dones, infos = self.env.step(goal_pose)
with torch.inference_mode():
       ....#other code

closed 2024-01-25 by zoctipus

leoliumessi, 2024-05-17:
Have you found a good way to slove this problem? I also meet this problem.