NVlabs/curobo #132: Enabling Cache for IssacSim in Docker Container

emfebert, 2024-01-23:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode: docker isaac sim
2. python version:  /isaac-sim/python.sh which is Python 3.10.13
3. Isaac Sim version (if using):   isaac_sim_2023.1.0

------
Hi,

I'm using the docker container curobo_docker:user_isaac_sim_2023.1.0 and the cache is disabled as indicated by the message in the top-right corner of the isaac-sim gui.

How can I enable the cache?
The loading times for the curobo examples are extremely slow (3-4 minutes) every time.

I'm using isaac-sim as a native gui in the docker.

I'm cross-linking the same question int the discussion here: https://github.com/NVlabs/curobo/discussions/131

balakumar-s, 2024-01-23:
If you run the standard isaac sim docker, does cache work then?

emfebert, 2024-01-26:
I followed the instructions for the issac sim container here:
https://docs.omniverse.nvidia.com/isaacsim/latest/installation/install_container.html 
But the cache isn't working there either.

The cache folders in this command don't exist on my computer:
docker run --name isaac-sim --entrypoint bash -it --gpus all -e "ACCEPT_EULA=Y" --rm --network=host \
    -e "PRIVACY_CONSENT=Y" \
    -v ~/docker/isaac-sim/cache/kit:/isaac-sim/kit/cache:rw \
    -v ~/docker/isaac-sim/cache/ov:/root/.cache/ov:rw \
    -v ~/docker/isaac-sim/cache/pip:/root/.cache/pip:rw \
    -v ~/docker/isaac-sim/cache/glcache:/root/.cache/nvidia/GLCache:rw \
    -v ~/docker/isaac-sim/cache/computecache:/root/.nv/ComputeCache:rw \
    -v ~/docker/isaac-sim/logs:/root/.nvidia-omniverse/logs:rw \
    -v ~/docker/isaac-sim/data:/root/.local/share/ov/data:rw \
    -v ~/docker/isaac-sim/documents:/root/Documents:rw \
    nvcr.io/nvidia/isaac-sim:2023.1.1

How can I locate the cache on my machine and properly mount it in the docker?

balakumar-s, 2024-01-28:
If cache is not working in isaac sim container, can you ask experts here: https://forums.developer.nvidia.com/c/omniverse/simulation/69

closed 2024-03-23 by balakumar-s