NVlabs/curobo #134: Unit tests fail (v0.62 and main:  c09d949)

gavincangan, 2024-01-24:
Please provide the below information in addition to your issue:

1. cuRobo installation mode: python
2. python version: 3.11.7
3. Isaac Sim version (if using): N/A

------

### Issue Details

I ran the unit tests after `pip install`ing `cuRobo` and they error out. Here's the output:

```bash
gavin@omen:~/repos/curobo$ python3 -m pytest .
================== test session starts ====================
platform linux -- Python 3.11.7, pytest-7.4.4, pluggy-1.3.0
rootdir: /home/gavin/repos/curobo
configfile: pyproject.toml
plugins: anyio-4.2.0, torchtyping-0.1.4, typeguard-4.1.5
collected 3 items / 27 errors / 1 skipped

========================== ERRORS ===================
___________________________________________________________________________________________________________ ERROR collecting tests/cost_test.py ____________________________________________________________________________________________________________
tests/cost_test.py:15: in <module>
    from curobo.geom.sdf.world import WorldCollisionConfig, WorldPrimitiveCollision
src/curobo/geom/sdf/world.py:22: in <module>
    from curobo.geom.types import Cuboid, Mesh, Obstacle, WorldConfig, batch_tensor_cube
src/curobo/geom/types.py:36: in <module>
    @dataclass
../../miniforge3/envs/torch11.8/lib/python3.11/dataclasses.py:1230: in dataclass
    return wrap(cls)
../../miniforge3/envs/torch11.8/lib/python3.11/dataclasses.py:1220: in wrap
    return _process_class(cls, init, repr, eq, order, unsafe_hash,
../../miniforge3/envs/torch11.8/lib/python3.11/dataclasses.py:958: in _process_class
    cls_fields.append(_get_field(cls, name, type, kw_only))
../../miniforge3/envs/torch11.8/lib/python3.11/dataclasses.py:815: in _get_field
    raise ValueError(f'mutable default {type(f.default)} for field '
E   ValueError: mutable default <class 'curobo.geom.types.Material'> for field material is not allowed: use default_factory
```

I tried some hacky fixes:
 ```bash
# in src/curobo/geom/types.py
-    material: Material = Material()
+    material: Material = field(default_factory=Material)
```

```bash
# in src/curobo/rollout/rollout_base.py
-    goal_pose: Pose = Pose()
+    goal_pose: Pose = field(default_factory=Pose)
```

```bash
# in src/curobo/wrap/reacher/trajopt.py
-    traj_evaluator_config: TrajEvaluatorConfig = TrajEvaluatorConfig()
+    traj_evaluator_config: TrajEvaluatorConfig = field(default_factory=TrajEvaluatorConfig)
```

But now, I have this `warp` error

```bash
E   warp.codegen.WarpCodegenError: Error while parsing function "compute_matrix_to_quat" at /home/gavin/repos/curobo/src/curobo/geom/transform.py:401:
E       out_v[0] = wp.index(out_q, 3)
E   ;Couldn't find function overload for 'index' that matched inputs with types: [warp.types.quatf, int32]
```

gavincangan, 2024-01-24:
Thanks for the response! So I did `pip install warp-lang==0.9.0`, but needed to make a few other changes for the unit tests to run: https://github.com/gavincangan/curobo/tree/fix/unit_tests

balakumar-s, 2024-01-25:
The library has not been tested on python>3.10. 

Regarding your change to torch.Size(https://github.com/gavincangan/curobo/commit/ff64469cdb33490a458b1ced921f48a6235624f8) What pytorch version are you using? 

The issue with wp.index is fixed in the latest commit, so newer warp should also work.

gavincangan, 2024-01-26:
> The library has not been tested on python>3.10.
> 
> Regarding your change to torch.Size([gavincangan@ff64469](https://github.com/gavincangan/curobo/commit/ff64469cdb33490a458b1ced921f48a6235624f8)) What pytorch version are you using?

I'm using v2.1.2

> The issue with wp.index is fixed in the latest commit, so newer warp should also work.
Ah, thank you, I'll switch to the newer version then :)

closed 2024-01-26 by balakumar-s