NVlabs/curobo #138: How do I fix it?

SshiJwu, 2024-01-25:
Building wheels for collected packages: nvidia-curobo
  Building editable for nvidia-curobo (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building editable for nvidia-curobo (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [154 lines of output]
      /home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle/setuptools/dist.py:745: SetuptoolsDeprecationWarning: Invalid dash-separated options
      !!
      
              ********************************************************************************
              Usage of dash-separated 'long-description-content-type' will not be supported in future
              versions. Please use the underscore name 'long_description_content_type' instead.
      
              This deprecation is overdue, please update your project and remove deprecated
              calls to avoid build errors in the future.
      
              See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
              ********************************************************************************
      
      !!
        opt = self.warn_dash_deprecation(opt, section)
      /home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle/setuptools/config/setupcfg.py:293: _DeprecatedConfig: Deprecated config in `setup.cfg`
      !!
      
              ********************************************************************************
              The license_file parameter is deprecated, use license_files instead.
      
              This deprecation is overdue, please update your project and remove deprecated
              calls to avoid build errors in the future.
      
              See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
              ********************************************************************************
      
      !!
        parsed = self.parsers.get(option_name, lambda x: x)(value)
      *********************************
      CuRobo: Compiling CUDA kernels...
      This will take 20+ minutes, to speedup compilation set TORCH_CUDA_ARCH_LIST={X}+PTX to  compile only for a specific architecture. e.g., export TORCH_CUDA_ARCH_LIST=7.0+PTX
      running editable_wheel
      creating /tmp/pip-wheel-8b62s67g/.tmp-wb0qhzvx/nvidia_curobo.egg-info
      writing /tmp/pip-wheel-8b62s67g/.tmp-wb0qhzvx/nvidia_curobo.egg-info/PKG-INFO
      writing dependency_links to /tmp/pip-wheel-8b62s67g/.tmp-wb0qhzvx/nvidia_curobo.egg-info/dependency_links.txt
      writing requirements to /tmp/pip-wheel-8b62s67g/.tmp-wb0qhzvx/nvidia_curobo.egg-info/requires.txt
      writing top-level names to /tmp/pip-wheel-8b62s67g/.tmp-wb0qhzvx/nvidia_curobo.egg-info/top_level.txt
      writing manifest file '/tmp/pip-wheel-8b62s67g/.tmp-wb0qhzvx/nvidia_curobo.egg-info/SOURCES.txt'
      reading manifest template 'MANIFEST.in'
      adding license file 'LICENSE'
      writing manifest file '/tmp/pip-wheel-8b62s67g/.tmp-wb0qhzvx/nvidia_curobo.egg-info/SOURCES.txt'
      creating '/tmp/pip-wheel-8b62s67g/.tmp-wb0qhzvx/nvidia_curobo-0.6.2.post1.dev5.dist-info'
      creating /tmp/pip-wheel-8b62s67g/.tmp-wb0qhzvx/nvidia_curobo-0.6.2.post1.dev5.dist-info/WHEEL
      running build_py
      running build_ext
      Traceback (most recent call last):
        File "/home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle/setuptools/command/editable_wheel.py", line 155, in run
          self._create_wheel_file(bdist_wheel)
        File "/home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle/setuptools/command/editable_wheel.py", line 344, in _create_wheel_file
          files, mapping = self._run_build_commands(dist_name, unpacked, lib, tmp)
        File "/home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle/setuptools/command/editable_wheel.py", line 267, in _run_build_commands
          self._run_build_subcommands()
        File "/home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle/setuptools/command/editable_wheel.py", line 294, in _run_build_subcommands
          self.run_command(name)
        File "/home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle/setuptools/dist.py", line 1234, in run_command
          super().run_command(command)
        File "/home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle/setuptools/command/build_ext.py", line 84, in run
          _build_ext.run(self)
[... 73 lines cut ...]
        File "/home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle/setuptools/dist.py", line 1234, in run_command
          super().run_command(command)
        File "/home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle/setuptools/command/build_ext.py", line 84, in run
          _build_ext.run(self)
        File "/home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle/setuptools/_distutils/command/build_ext.py", line 345, in run
          self.build_extensions()
        File "/home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/utils/cpp_extension.py", line 499, in build_extensions
          _check_cuda_version(compiler_name, compiler_version)
        File "/home/wsj/.local/share/ov/pkg/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/utils/cpp_extension.py", line 387, in _check_cuda_version
          raise RuntimeError(CUDA_MISMATCH_MESSAGE.format(cuda_str_version, torch.version.cuda))
      RuntimeError:
      The detected CUDA version (10.1) mismatches the version that was used to compile
      PyTorch (11.8). Please make sure to use the same CUDA versions.
      
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building editable for nvidia-curobo
Failed to build nvidia-curobo
ERROR: Could not build wheels for nvidia-curobo, which is required to install pyproject.toml-based projects
There was an error running python

balakumar-s, 2024-01-25:
What command did you run?

SshiJwu, 2024-01-26:
I run the command "omni_python -m pip install -e .[isaacsim] --no-build-isolation".

SshiJwu, 2024-01-26:
I'm using the IsaacSim 2023.1.0 and CUDA version 10.1.

balakumar-s, 2024-01-26:
The log reports the error: 
```

RuntimeError:
  The detected CUDA version (10.1) mismatches the version that was used to compile
  PyTorch (11.8). Please make sure to use the same CUDA versions.
  

```

Install cuda 11.8 and this will work.

crlz182, 2024-01-27:
I was running into that issue as well. The problem is that the cuda version of your PyTorch does not align with the cuda toolkit version installed on your pc. I solved it by manually installing cuda-toolkit as well as the cuda screen drivers in combination with the correct pytorch version.

1. First you have to remove the installed cuda drivers and the cuda toolkit:
```
  #remove manual installation
  sudo /usr/local/cuda-X.Y/bin/cuda-uninstaller
  sudo /usr/bin/nvidia-uninstall
  
  #remove install packages installed with package manager
  sudo apt-get remove --purge nvidia\*
  sudo apt-get remove --purge \*nvidia\*
  sudo apt-get remove --purge \*cuda\*
  sudo apt-get remove --purge \*nvrtc\*
  sudo apt-get autoremove
      
  
  #download and install new version (11.7 in this case)
  wget https://developer.download.nvidia.com/compute/cuda/11.7.0/local_installers/cuda_11.7.0_515.43.04_linux.run
  sudo sh cuda_11.7.0_515.43.04_linux.run
```

2. In my case the screen turned black after the screen drivers got installed. I restarted the PC, ran the command again and only selected cuda toolkit this time. After these steps you should see cuda 11.7 when you run `nvidia-smi` and `nvcc --version`. 
3. You now have to install the correct pytorch version that matches cuda 11.7 from this site: https://pytorch.org/get-started/previous-versions/
4. Now you can install curobo again and it should compile

SshiJwu, 2024-01-28:
When I was running “omni_python examples/isaac_sim/motion_gen_reacher.py”,  I encountered this problem：
  File "/home/user/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/utils/prims.py", line 802, in set_prim_property
    prim.GetAttribute(property_name).Set(property_value)
  File "/home/user/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/articulations/articulation_view.py", line 3243, in set_solver_velocity_iteration_counts
    set_prim_property(self.prim_paths[i], "physxArticulation:solverVelocityIterationCount", counts[read_idx])
  File "/home/user/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/articulations/articulation.py", line 981, in set_solver_velocity_iteration_count
    self._articulation_view.set_solver_velocity_iteration_counts(count)
  File "/home/user/.local/share/ov/pkg/isaac_sim-2023.1.1/curobo/examples/isaac_sim/helper.py", line 115, in add_robot_to_scene
    robot_p.set_solver_velocity_iteration_count(4)
  File "/home/user/.local/share/ov/pkg/isaac_sim-2023.1.1/curobo/examples/isaac_sim/motion_gen_reacher.py", line 153, in main
    robot, robot_prim_path = add_robot_to_scene(robot_cfg, my_world)
  File "/home/user/.local/share/ov/pkg/isaac_sim-2023.1.1/curobo/examples/isaac_sim/motion_gen_reacher.py", line 376, in <module>
    main()
pxr.Tf.ErrorException: 
	Error in 'pxrInternal_v0_22__pxrReserved__::UsdStage::_SetValueImpl' at line 6174 in file /buildAgent/work/ac88d7d902b57417/USD/pxr/usd/usd/stage.cpp : 'Empty typeName for </World/panda.physxArticulation:solverVelocityIterationCount>'

balakumar-s, 2024-01-28:
Can you comment out these lines: https://github.com/NVlabs/curobo/blob/88eac64edca2d2b641c1622e63f3e750157596c4/examples/isaac_sim/helper.py#L115

prajapatisarvesh, 2024-02-10:
Hi @balakumar-s it seems that there's some api changes that is causing issues to import robot from an `urdf` file. I've implemented the load_from_usd function in helper.py that works for almost all examples except dual_arm_reach as there's no usd file for the same.
@SshiJwu you can test the implementation here --
https://github.com/prajapatisarvesh/curobo/tree/fix-2023.1.1

balakumar-s, 2024-03-23:
We added partial support for isaac sim 2023.1.1. Examples should work when you pass `--robot ur5e.yml`

closed 2024-03-23 by balakumar-s