NVlabs/curobo #144: Error building docker image with issac sim 2022.2.1

D-1shu, 2024-01-29:
1. cuRobo installation mode : docker isaac sim
2. python version:
3. Isaac Sim version (if using): 2022.2.1

------
Issue Details
Error building in docker image from `bash build_docker.sh isaac_sim_2022.2.1` command at `step 41`

```
Step 41/43 : RUN cd /pkgs && git clone https://github.com/nvlabs/nvblox_torch.git &&     cd /pkgs/nvblox_torch &&     sh install_isaac_sim.sh $($omni_python -c 'import torch.utils; print(torch.utils.cmake_prefix_path)') &&     $omni_python -m pip install -e .
 ---> Running in 9d87423c797d
Cloning into 'nvblox_torch'...
-- The CXX compiler identification is GNU 9.4.0
-- The CUDA compiler identification is NVIDIA 11.4.152
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting CUDA compiler ABI info
-- Detecting CUDA compiler ABI info - done
-- Check for working CUDA compiler: /usr/local/cuda/bin/nvcc - skipped
-- Detecting CUDA compile features
-- Detecting CUDA compile features - done
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda (found version "11.4") 
-- Caffe2: CUDA detected: 11.4
-- Caffe2: CUDA nvcc is: /usr/local/cuda/bin/nvcc
-- Caffe2: CUDA toolkit directory: /usr/local/cuda
-- Caffe2: Header version is: 11.4
-- Could NOT find CUDNN (missing: CUDNN_LIBRARY_PATH CUDNN_INCLUDE_PATH) 
CMake Warning at /isaac-sim/extscache/omni.pip.torch-1_13_1-0.1.4+104.2.lx64/torch-1-13-1/torch/share/cmake/Caffe2/public/cuda.cmake:120 (message):
  Caffe2: Cannot find cuDNN library.  Turning the option off
Call Stack (most recent call first):
  /isaac-sim/extscache/omni.pip.torch-1_13_1-0.1.4+104.2.lx64/torch-1-13-1/torch/share/cmake/Caffe2/Caffe2Config.cmake:92 (include)
  /isaac-sim/extscache/omni.pip.torch-1_13_1-0.1.4+104.2.lx64/torch-1-13-1/torch/share/cmake/Torch/TorchConfig.cmake:68 (find_package)
  CMakeLists.txt:32 (find_package)

CMake Warning at /isaac-sim/extscache/omni.pip.torch-1_13_1-0.1.4+104.2.lx64/torch-1-13-1/torch/share/cmake/Caffe2/public/cuda.cmake:214 (message):
  Failed to compute shorthash for libnvrtc.so
Call Stack (most recent call first):
  /isaac-sim/extscache/omni.pip.torch-1_13_1-0.1.4+104.2.lx64/torch-1-13-1/torch/share/cmake/Caffe2/Caffe2Config.cmake:92 (include)
  /isaac-sim/extscache/omni.pip.torch-1_13_1-0.1.4+104.2.lx64/torch-1-13-1/torch/share/cmake/Torch/TorchConfig.cmake:68 (find_package)
  CMakeLists.txt:32 (find_package)

CMake Warning at /isaac-sim/extscache/omni.pip.torch-1_13_1-0.1.4+104.2.lx64/torch-1-13-1/torch/share/cmake/Caffe2/public/utils.cmake:385 (message):
  In the future we will require one to explicitly pass TORCH_CUDA_ARCH_LIST
  to cmake instead of implicitly setting it as an env variable.  This will
  become a FATAL_ERROR in future version of pytorch.
Call Stack (most recent call first):
  /isaac-sim/extscache/omni.pip.torch-1_13_1-0.1.4+104.2.lx64/torch-1-13-1/torch/share/cmake/Caffe2/public/cuda.cmake:437 (torch_cuda_get_nvcc_gencode_flag)
  /isaac-sim/extscache/omni.pip.torch-1_13_1-0.1.4+104.2.lx64/torch-1-13-1/torch/share/cmake/Caffe2/Caffe2Config.cmake:92 (include)
  /isaac-sim/extscache/omni.pip.torch-1_13_1-0.1.4+104.2.lx64/torch-1-13-1/torch/share/cmake/Torch/TorchConfig.cmake:68 (find_package)
  CMakeLists.txt:32 (find_package)

-- Added CUDA NVCC flags for: -gencode;arch=compute_70,code=sm_70;-gencode;arch=compute_70,code=compute_70
CMake Error at /isaac-sim/extscache/omni.pip.torch-1_13_1-0.1.4+104.2.lx64/torch-1-13-1/torch/share/cmake/Caffe2/Caffe2Config.cmake:100 (message):
  Your installed Caffe2 version uses cuDNN but I cannot find the cuDNN
  libraries.  Please set the proper cuDNN prefixes and / or install cuDNN.
Call Stack (most recent call first):
  /isaac-sim/extscache/omni.pip.torch-1_13_1-0.1.4+104.2.lx64/torch-1-13-1/torch/share/cmake/Torch/TorchConfig.cmake:68 (find_package)
  CMakeLists.txt:32 (find_package)

-- Configuring incomplete, errors occurred!
The command '/bin/sh -c cd /pkgs && git clone https://github.com/nvlabs/nvblox_torch.git &&     cd /pkgs/nvblox_torch &&     sh install_isaac_sim.sh $($omni_python -c 'import torch.utils; print(torch.utils.cmake_prefix_path)') &&     $omni_python -m pip install -e .' returned a non-zero code: 1

```

balakumar-s, 2024-02-08:
Do you have NVIDIA driver 535 or above?

D-1shu, 2024-02-11:
@balakumar-s Driver Version: 525.60.11    CUDA Version: 12.0

Car38639, 2024-03-06:
I hit the same issue. I tried both 525 and 535 and neither of them worked.

anubhav-dogra, 2024-03-29:
Same here, Error is same.
My Nvidia Driver Version: 535.161.07   CUDA Version: 12.2

balakumar-s, 2024-08-19:
We have updated cuRobo to support Isaac Sim 4.0 as 2022.2.1 is no longer available.

closed 2024-08-19 by balakumar-s