NVlabs/curobo #152: CUDA error: invalid configuration argument for geom_cu.self_collision_distance when configuring new robot.

fi6, 2024-02-08:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): python
2. python version: 3.8
3. Isaac Sim version (if using): N/A

------
Issue Details

Runtime error when using new robot. The robot is configured following the steps described in the documentation.

Call stack:  
[image]
https://github.com/NVlabs/curobo/blob/88eac64edca2d2b641c1622e63f3e750157596c4/src/curobo/curobolib/geom.py#L58

Error information:  
```text
Exception has occurred: RuntimeError
CUDA error: invalid configuration argument
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.
```

balakumar-s, 2024-02-08:
Do you have at least two links in `collision_link_names:[]`? and also do you have at least a pair of links that are not allowed to self collide?

Can you paste the values for your robot for the below keys:

1. `collision_link_names`
2. `self_collision_ignore`

fi6, 2024-02-19:
The `collision_link_names` and `self_collision_ignore` is shown below:
```yaml
    collision_link_names: [
        'body_link',
        'rear_left_wheel_link',
        'rear_right_wheel_link',
        'front_right_wheel_link',
        'front_left_wheel_link',
        'chassis_link',
        'right_arm_base_link_inertia',
        'right_arm_shoulder_link',
        'right_arm_upper_arm_link',
        'right_arm_forearm_link',
        'right_arm_wrist_1_link',
        'right_arm_wrist_2_link',
        'right_arm_wrist_3_link',
        'right_gripper_palm',
        'right_gripper_finger_1_link_3',
        'right_gripper_finger_2_link_3',
        'right_gripper_finger_middle_link_3',
        'right_gripper_finger_2_link_2',
        'right_gripper_finger_1_link_2',
        'right_gripper_finger_middle_link_2',
        'closet_bottom_right_handle',
        'closet_bottom_right_door_link',
        'closet_base_link',
      ] # List[str]
```
```yaml
    self_collision_ignore: {
        'base_link': ['body_link', 'right_arm_base_link_inertia'],
        'body_link': ['right_arm_base_link_inertia', 'right_arm_shoulder_link'],
        'right_arm_base_link_inertia': ['right_arm_shoulder_link'],
        'right_arm_forearm_link':
          [
            'right_arm_upper_arm_link',
            'right_arm_wrist_1_link',
            'right_arm_wrist_2_link',
          ],
        'right_arm_shoulder_link': ['right_arm_upper_arm_link'],
        'right_arm_wrist_1_link':
          [
            'right_arm_wrist_2_link',
            'right_arm_wrist_3_link',
            'right_gripper_finger_1_link_0',
            'right_gripper_finger_1_link_1',
            'right_gripper_finger_1_link_2',
            'right_gripper_finger_1_link_3',
            'right_gripper_finger_2_link_0',
            'right_gripper_finger_2_link_1',
            'right_gripper_finger_2_link_2',
            'right_gripper_finger_2_link_3',
            'right_gripper_finger_middle_link_0',
            'right_gripper_finger_middle_link_1',
            'right_gripper_finger_middle_link_2',
            'right_gripper_finger_middle_link_3',
            'right_gripper_flange',
            'right_gripper_palm',
          ],
        'right_arm_wrist_2_link':
          [
            'right_arm_wrist_3_link',
            'right_gripper_finger_1_link_0',
            'right_gripper_finger_1_link_1',
            'right_gripper_finger_1_link_2',
            'right_gripper_finger_1_link_3',
            'right_gripper_finger_2_link_0',
            'right_gripper_finger_2_link_1',
            'right_gripper_finger_2_link_2',
            'right_gripper_finger_2_link_3',
[... 101 lines cut ...]
        'right_gripper_finger_middle_link_0':
          [
            'right_gripper_finger_middle_link_1',
            'right_gripper_finger_middle_link_2',
            'right_gripper_finger_middle_link_3',
            'right_gripper_flange',
            'right_gripper_palm',
          ],
        'right_gripper_finger_middle_link_1':
          [
            'right_gripper_finger_middle_link_2',
            'right_gripper_finger_middle_link_3',
            'right_gripper_flange',
            'right_gripper_palm',
          ],
        'right_gripper_finger_middle_link_2':
          [
            'right_gripper_finger_middle_link_3',
            'right_gripper_flange',
            'right_gripper_palm',
          ],
        'right_gripper_finger_middle_link_3': ['right_gripper_flange'],
        'right_gripper_flange': ['right_gripper_palm'],
      } # Dict[str, List[str]]
```

balakumar-s, 2024-02-20:
Can you share your robot urdf and configuration file?

We have tested on up to 1000 spheres so maybe there is something specific to your robot causing the issue.

fi6, 2024-02-20:
Thanks for the reply! May I know what GPU you tested on? We encountered GPU memory issue on ik example with collisions, so we think this may because we are using 3070 which does not have enough threads, or some other parameters may exceeded the limit.

balakumar-s, 2024-02-20:
Launching more threads than available will not cause a runtime error. We tested on a 3090 and 4090. What batch size were you using for IK?

closed 2024-03-23 by balakumar-s