NVlabs/curobo #159: The rigid body at /World/panda/ee_link has a possibly invalid inertia tensor

ravijo, 2024-02-18:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode: python
  ```console
  $ git clone https://github.com/NVlabs/curobo.git
  $ pip install -e . --no-build-isolation
  ```
2. python version
  ```console
  $ python3 --version
  Python 3.10.12
  
  $ omni_python --version
  Python 3.7.13
  ```
3. Isaac Sim version: 2022.2.1

------
Issue Details

Based on [the official documentation](https://curobo.org/source/getting_started/1_install_instructions.html#install-for-use-in-isaac-sim), I am trying to run a motion generation example after installing  Isaac Sim 2022.2.1 using the following command:
```console
ravi@asus:~/curobo$ omni_python examples/isaac_sim/motion_gen_reacher.py
```

After few minutes, it reports the following warning in the terminal:
```
...
**** Click Play to start simulation *****
**** Click Play to start simulation *****
2024-02-18 04:55:20 [258,688ms] [Warning] [omni.physx.plugin] The rigid body at /World/panda/base_link has a possibly invalid inertia tensor of {1.0, 1.0, 1.0} and a negative mass, small sphere approximated inertia was used. Either specify correct values in the mass properties, or add collider(s) to any shape(s) that you wish to automatically compute mass properties for. If you do not want the objects to collide, add colliders regardless then disable the 'enable collision' property.
2024-02-18 04:55:20 [258,688ms] [Warning] [omni.physx.plugin] The rigid body at /World/panda/ee_link has a possibly invalid inertia tensor of {1.0, 1.0, 1.0} and a negative mass, small sphere approximated inertia was used. Either specify correct values in the mass properties, or add collider(s) to any shape(s) that you wish to automatically compute mass properties for. If you do not want the objects to collide, add colliders regardless then disable the 'enable collision' property.
Updating world, reading w.r.t. /World/panda
2
Updated World
...
```

The same warning is reported within Isaac Sim console as shown below:
```
2024-02-18 04:51:24  [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2024-02-18 04:51:24  [Warning] [omni.ext.plugin] [ext: omni.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/media/hdd/Omniverse/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.lidar' or '/media/hdd/Omniverse/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.lidar/config'
2024-02-18 04:51:24  [Warning] [omni.ext.plugin] [ext: omni.sensors.nv.radar] Extensions config 'extension.toml' doesn't exist '/media/hdd/Omniverse/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.radar' or '/media/hdd/Omniverse/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.radar/config'
2024-02-18 04:51:24  [Warning] [carb.graphics-vulkan.plugin] No command queue family supports flags: 0x100, queue type: 3. No queues of this type will be created
2024-02-18 04:51:24  [Warning] [omni.kit.viewport.menubar.core.menu_item.viewport_menu_item] ViewportMenuItem order argument is deprecated, use order_setting_path
2024-02-18 04:51:24  [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
2024-02-18 04:51:24  [Warning] [omni.ext.plugin] [ext: omni.isaac.sensor-5.6.3] Native plugin search dir contains several files: /media/hdd/Omniverse/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.isaac.sensor/bin . It makes dir walk slow, replace with the exact path to the plugin instead of a wildcard.
2024-02-18 04:51:24  [Warning] [omni.isaac.isaac_sensor] omni.isaac.isaac_sensor is deprecated and was renamed; use omni.isaac.sensor
2024-02-18 04:51:24  [Warning] [omni.kit.widget.live.cache_state_menu] Unable to detect Omniverse Cache Server. Consider installing it for better IO performance.
2024-02-18 04:52:09  [Warning] [gpu.foundation.plugin] Waiting for compilation of ray tracing shaders by GPU driver: 30 seconds so far
2024-02-18 04:52:39  [Warning] [gpu.foundation.plugin] Waiting for compilation of ray tracing shaders by GPU driver: 60 seconds so far
2024-02-18 04:53:09  [Warning] [gpu.foundation.plugin] Waiting for compilation of ray tracing shaders by GPU driver: 90 seconds so far
2024-02-18 04:53:31  [Warning] [gpu.foundation.plugin] Ray tracing shader compilation finished after 112 seconds
2024-02-18 04:54:26  [Warning] [omni.client.plugin]  Tick: authentication: Discovery(ws://localhost/omni/discovery): Error creating Auth/Credentials search: Not connected
2024-02-18 04:54:26  [Warning] [omni.client.plugin]  Tick: authentication: Discovery(ws://localhost/omni/discovery): Error creating Auth/Tokens search: Not connected
2024-02-18 04:54:26  [Warning] [omni.client.plugin]  Tick: authentication: Discovery(ws://localhost/omni/discovery): Error creating Api/Connection search: Not connected
2024-02-18 04:54:26  [Warning] [omni.isaac.core.utils.nucleus] /persistent/app/omniverse/mountedDrives setting not found
2024-02-18 04:54:28  [Warning] [omni.hydra] Mesh '/World/obstacles/table_mesh' has corrupted data in primvar 'displayColor': buffer size 8 doesn't match expected size 36 in faceVarying primvars
2024-02-18 04:55:20  [Warning] [omni.physx.plugin] The rigid body at /World/panda/base_link has a possibly invalid inertia tensor of {1.0, 1.0, 1.0} and a negative mass, small sphere approximated inertia was used. Either specify correct values in the mass properties, or add collider(s) to any shape(s) that you wish to automatically compute mass properties for. If you do not want the objects to collide, add colliders regardless then disable the 'enable collision' property.
2024-02-18 04:55:20  [Warning] [omni.physx.plugin] The rigid body at /World/panda/ee_link has a possibly invalid inertia tensor of {1.0, 1.0, 1.0} and a negative mass, small sphere approximated inertia was used. Either specify correct values in the mass properties, or add collider(s) to any shape(s) that you wish to automatically compute mass properties for. If you do not want the objects to collide, add colliders regardless then disable the 'enable collision' property.
2024-02-18 04:56:17  [Warning] [omni.physx.plugin] The rigid body at /World/panda/base_link has a possibly invalid inertia tensor of {1.0, 1.0, 1.0} and a negative mass, small sphere approximated inertia was used. Either specify correct values in the mass properties, or add collider(s) to any shape(s) that you wish to automatically compute mass properties for. If you do not want the objects to collide, add colliders regardless then disable the 'enable collision' property.
2024-02-18 04:56:17  [Warning] [omni.physx.plugin] The rigid body at /World/panda/ee_link has a possibly invalid inertia tensor of {1.0, 1.0, 1.0} and a negative mass, small sphere approximated inertia was used. Either specify correct values in the mass properties, or add collider(s) to any shape(s) that you wish to automatically compute mass properties for. If you do not want the objects to collide, add colliders regardless then disable the 'enable collision' property.
```

Please see attached screenshot below:

[image]

After clicking on "Play" button, nothing happens in the viewport. How to run examples inside Isaac Sim? Thanks

balakumar-s, 2024-02-18:
Did you try moving the red cube as shown here? https://curobo.org/source/getting_started/2b_isaacsim_examples.html#motion-generation

ravijo, 2024-02-18:
@balakumar-s 

Thanks! Perfect. It works.

closed 2024-02-18 by ravijo