NVlabs/curobo #163: Trouble installing curobo to work with Isaac Sim

SLrepo, 2024-02-24:
I am trying to use curobo with Isaac Sim, because I need to configure a new robot. I'm using RTX 4070, Debian 12, and Cuda 11.7. I have tried the following
1. With Isaac Sim 2022.2.1, I can install curobo, but every time I try to load the .usd file, Isaac Sim crashes. Other versions of Isaac Sim can load the same .usd file. 
2. With Isaac Sim 2023.1.0, I cannot install curobo, it always reports ```command '/usr/local/cuda-11.7/bin/nvcc' failed with exit code 1``` during installation, because there is a fatal error ```Unsupported gpu architecture 'compute_89'```, which is the compute capability of 4070. 

What do you suggest I do at this point? Or it just won't work with 4070? 
Are there ways to configure a new robot without Isaac Sim?

balakumar-s, 2024-02-25:
Can you try cuda 11.8 which should support your gpu in isaac sim 2023.1.0?

SLrepo, 2024-02-26:
It works. Thanks!

closed 2024-02-27 by balakumar-s