NVlabs/curobo #167: URDF Importer fails for all robots on isaac-sim-2023.1.0-hotfix.1

sjauhri, 2024-02-26:
Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): Isaac Sim
2. python version: 3.10.13
3. Isaac Sim version (if using): isaac-sim-2023.1.0-hotfix.1

------
Issue Details

URDF import does not work for the franka and most other robots (when calling either `examples/isaac_sim/motion_gen_reacher.py` or `examples/isaac_sim/load_all_robots.py`). Isaac sim complains that the files are broken:
**[Warning] [omni.importer.urdf] Asset convert failed as asset file is broken.**
Specifically, the error comes at line 96 in helper.py: https://github.com/NVlabs/curobo/blob/286b3820a5a018f9ff1a4df8d7b998dc3e456791/examples/isaac_sim/helper.py#L96

It's a very strange error since nothing seems out of place here.
I've also verified that the urdf importer does successfully import a panda when using the default isaac sim importer and its files:
```
omni.kit.commands.execute(
                "URDFParseAndImportFile",
                urdf_path=self._extension_path + "/data/urdf/robots/franka_description/robots/panda_arm_hand.urdf",
                import_config=import_config,
            )
```

balakumar-s, 2024-02-26:
The meshes in curobo are stored in lfs, can you check if the mesh files are not just pointers and contain mesh data?

If they are pointers, try `git lfs pull`

sjauhri, 2024-02-26:
[Update, SOLVED]: I was able to solve the issue by replacing the mesh files at "curobo/src/curobo/content/assets/robot/franka_description/meshes" with the mesh files provided by standalone isaac sim at "isaac_sim-2023.1.0-hotfix.1/extscache/omni.importer.urdf-1.1.3+105.1.lx64.r.cp310/data/urdf/robots/franka_description".

It seems that these mesh files in curobo are deprecated.

sjauhri, 2024-02-26:
> The meshes in curobo are stored in lfs, can you check if the mesh files are not just pointers and contain mesh data?
> 
> If they are pointers, try `git lfs pull`

Ah thanks! This seems to be the problem. They were just pointers...

closed 2024-02-26 by sjauhri