NVlabs/curobo #176: 'devices_without_uva' error

fnuabhimanyu8713, 2024-03-05:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): Python
2. python version: 3.9
3. Isaac Sim version (if using):  N/A

------
While running "python3 -m pytest .", I encountered the error:  
UnboundLocalError: local variable 'devices_without_uva' referenced before assignment. Not sure how to proceed

balakumar-s, 2024-03-05:
The Warp team (https://github.com/NVIDIA/warp) is aware of this issue and will push a fix. In the mean time, downgrade to an older version with `pip install warp-lang==0.13.0`. 

We will add a upper bound on warp version in our installation to avoid issues like this in the future.

closed 2024-03-05 by balakumar-s

fnuabhimanyu8713, 2024-03-05:
Thanks for prompt response!