NVlabs/curobo #180: I got errer when installing a nvblox

songwookim, 2024-03-10:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): isaac sim
2. python version: 3.7
3. Isaac Sim version (if using):  22.2.1

------
Issue Details

Thanks for your great work :)

I don't know which part is the problem... 

Following curobo install instruction, i got error when executing below command.

```
cd ${PKGS_PATH} &&  git clone https://github.com/valtsblukis/nvblox.git && cd ${PKGS_PATH}/nvblox/nvblox && mkdir build && cd build && \
cmake ..  -DBUILD_REDISTRIBUTABLE=ON \
-DCMAKE_CXX_FLAGS=-D_GLIBCXX_USE_CXX11_ABI=0  -DPRE_CXX11_ABI_LINKABLE=ON \
-DSQLITE3_BASE_PATH="${PKGS_PATH}/sqlite/install/" -DGLOG_BASE_PATH="${PKGS_PATH}/glog/install/" \
-DGFLAGS_BASE_PATH="${PKGS_PATH}/gflags/install/" -DCMAKE_CUDA_FLAGS=-D_GLIBCXX_USE_CXX11_ABI=0 && \
make -j32 && \
sudo make install
```

The log is as follows.

```
[ 14%] Completed 'ext_eigen'
[ 14%] Built target ext_eigen
[ 16%] Building CXX object CMakeFiles/nvblox_gpu_hash.dir/src/utils/timing.cpp.o
[ 17%] Building CXX object CMakeFiles/nvblox_gpu_hash.dir/src/utils/nvtx_ranges.cpp.o
[ 17%] Building CUDA object CMakeFiles/nvblox_gpu_hash.dir/src/core/error_check.cu.o
[ 18%] Building CUDA object CMakeFiles/nvblox_gpu_hash.dir/src/gpu_hash/gpu_layer_view.cu.o
[ 20%] Building CUDA object CMakeFiles/nvblox_gpu_hash.dir/src/gpu_hash/gpu_set.cu.o
[ 21%] Linking CXX static library libnvblox_gpu_hash.a
[ 21%] Built target nvblox_gpu_hash
[ 22%] Building CXX object CMakeFiles/nvblox_lib.dir/src/core/cuda_stream.cpp.o
[ 24%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/core/error_check.cu.o
[ 24%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/core/warmup.cu.o
[ 25%] Building CXX object CMakeFiles/nvblox_lib.dir/src/core/parameter_tree.cpp.o
[ 27%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/map/blox.cu.o
[ 27%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/dynamics/dynamics_detection.cu.o
[ 28%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/map/layer.cu.o
[ 30%] Building CXX object CMakeFiles/nvblox_lib.dir/src/sensors/depth_preprocessing.cpp.o
[ 31%] Building CXX object CMakeFiles/nvblox_lib.dir/src/sensors/connected_components.cpp.o
[ 34%] Building CXX object CMakeFiles/nvblox_lib.dir/src/sensors/camera.cpp.o
[ 34%] Building CXX object CMakeFiles/nvblox_lib.dir/src/sensors/color.cpp.o
[ 34%] Building CXX object CMakeFiles/nvblox_lib.dir/src/sensors/npp_image_operations.cpp.o
[ 36%] Building CXX object CMakeFiles/nvblox_lib.dir/src/geometry/bounding_boxes.cpp.o
[ 37%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/sensors/pointcloud.cu.o
[ 37%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/sensors/image.cu.o
[ 40%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/integrators/view_calculator.cu.o
[ 41%] Building CXX object CMakeFiles/nvblox_lib.dir/src/mapper/multi_mapper.cpp.o
[ 42%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/integrators/occupancy_decay_integrator.cu.o
[ 42%] Building CXX object CMakeFiles/nvblox_lib.dir/src/geometry/bounding_spheres.cpp.o
[ 45%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/integrators/projective_occupancy_integrator.cu.o
[ 45%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/integrators/projective_color_integrator.cu.o
[ 45%] Building CXX object CMakeFiles/nvblox_lib.dir/src/mapper/mapper.cpp.o
[ 46%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/integrators/tsdf_decay_integrator.cu.o
[ 47%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/integrators/projective_tsdf_integrator.cu.o
[ 51%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/integrators/esdf_integrator.cu.o
[ 51%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/integrators/esdf_slicer.cu.o
[ 51%] Building CXX object CMakeFiles/nvblox_lib.dir/src/io/mesh_io.cpp.o
[ 53%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/integrators/freespace_integrator.cu.o
[ 53%] Building CXX object CMakeFiles/nvblox_lib.dir/src/interpolation/interpolation_3d.cpp.o
[ 55%] Building CXX object CMakeFiles/nvblox_lib.dir/src/io/ply_writer.cpp.o
[ 55%] Building CUDA object CMakeFiles/nvblox_lib.dir/src/rays/sphere_tracer.cu.o
[... 182 lines cut ...]
make[2]: *** [examples/CMakeFiles/load_map_and_mesh.dir/build.make:105: examples/load_map_and_mesh] error 1
make[1]: *** [CMakeFiles/Makefile2:547: examples/CMakeFiles/load_map_and_mesh.dir/all] error 2
[ 90%] Linking CXX executable esdf_query
/usr/bin/ld: /home/songwoo/pkgs/glog/install//lib/libglog.a(utilities.cc.o): in function `google::GetStackTrace(void**, int, int)':
utilities.cc:(.text+0x9e5): undefined reference to `_Ux86_64_getcontext'
/usr/bin/ld: utilities.cc:(.text+0x9fe): undefined reference to `_ULx86_64_init_local'
/usr/bin/ld: utilities.cc:(.text+0xa62): undefined reference to `_ULx86_64_get_reg'
/usr/bin/ld: utilities.cc:(.text+0xabb): undefined reference to `_ULx86_64_step'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/CMakeFiles/esdf_query.dir/build.make:106: examples/esdf_query] error 1
make[1]: *** [CMakeFiles/Makefile2:518: examples/CMakeFiles/esdf_query.dir/all] error 2
[ 91%] Linking CXX shared library libnvblox_datasets.so
[ 91%] Built target nvblox_datasets
make: *** [Makefile:136: all] error 2

```

---
Additionally, It's small, but, I think there are some typing error at "7.install nvblox" of install instruction
I think it is needed to add '&&' between 'cd' command and  'mkdir' command

```
cd ${PKGS_PATH} &&  git clone https://github.com/valtsblukis/nvblox.git && cd ${PKGS_PATH}/nvblox/nvblox mkdir build && ...
# cd ${PKGS_PATH}/nvblox/nvblox mkdir build   && ->  cd ${PKGS_PATH}/nvblox/nvblox && mkdir build &&
```

balakumar-s, 2024-03-23:
It looks like glog was not correctly linked from this path: `home/songwoo/pkgs/glog/install//lib/libglog.a`, can you check if this path is valid?

songwookim, 2024-03-23:
Yes. `libglog.a` file exists in that path.

songwookim, 2024-03-24:
> It looks like glog was not correctly linked from this path: `home/songwoo/pkgs/glog/install//lib/libglog.a`, can you check if this path is valid?
You mean check whether exist the file in that path?
Yes. libglog.a file exists in that.

balakumar-s, 2024-03-24:
There is an extra `/` in the path here `install//lib/libglog.a`. Can you remove that and check if that fixes the error?

songwookim, 2024-03-24:
> There is an extra `/` in the path here `install//lib/libglog.a`. Can you remove that and check if that fixes the error?

Even if I erase and run again, I get the same error except for `/` at path

balakumar-s, 2024-03-24:
Your glog install has issues. Specifically, it's trying to link to libunwind. Can you try installing libunwind-dev and then recompile glog?

songwookim, 2024-03-25:
> Your glog install has issues. Specifically, it's trying to link to libunwind. Can you try installing libunwind-dev and then recompile glog?
> 
> More info: [google/glog#629](https://github.com/google/glog/issues/629)

I re-install glog and nvblox but got same error above

balakumar-s, 2024-03-25:
Are you using glog 0.6.0? An alternative is to try this docker: https://github.com/NVlabs/curobo/blob/b481ee201a2754c072ba9bcd80bd5d327fddc599/docker/isaac_sim.dockerfile#L213

songwookim, 2024-03-25:
> Are you using glog 0.6.0? An alternative is to try this docker:
> 
> https://github.com/NVlabs/curobo/blob/b481ee201a2754c072ba9bcd80bd5d327fddc599/docker/isaac_sim.dockerfile#L213

Yes glog version is 0.6.0.

I will try setting curobo at Docker.  

Sincerely thanks for your help :)

balakumar-s, 2024-04-01:
I am closing this issue as we don't know how to reproduce on our machine. If the docker fails, please re-open and we can help.

GracefulMan, 2024-04-15:
i have the same error when executing `make`, the glog branch is v0.6.0:
[image]

a-nooj, 2024-04-19:
Removing libunwind-dev worked for me

`sudo apt remove libunwind-dev`
Delete glog build folder, recreate, cmake, make, make install
Delete nvblox build folder, recreate, cmake, make, make install

zhengkai666666, 2025-01-08:
Is your library（glog,gflags） installed in a conda environment? I had a similar problem after installing in the Conda environment.Then I quit the Conda environment.Delete build folder, recreate, cmake, make, make install.That is OK.

songwookim, 2025-01-14:
> Is your library（glog,gflags） installed in a conda environment? I had a similar problem after installing in the Conda environment.Then I quit the Conda environment.Delete build folder, recreate, cmake, make, make install.That is OK.

thanks for your help :)