NVlabs/curobo #183: IK uses incorrect rotation for given ee-frame when it is not tool0

Fzaero, 2024-03-12:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): python
2. python version: 3.8

------
Issue Details

When I use a frame other than tool0 frame in IK, robot still uses rotation of tool0 frame even if rotation of end-effector frame is different. I am not sure if this is specific to tool0 frame.

To better explain, check the following images:

[image]
This image shows the tool0 and grasp_center frames. I tested using both of these frames for planning with curobo:

[image]

This is the plan for tool0 frame for given pose.

[image]

This is the plan for grasp_center frame for given pose.

As you see, curobo uses correct translations for grasp_center and tool0 frames, but still same rotation.

Fzaero, 2024-03-12:
Yes, I am using that format. 

I believe way that I give coordinate is correct as IK works correctly for tool0 frame, but not for grasp_center frame, as you can see in the images.

balakumar-s, 2024-03-12:
I'm assuming that you are sending the same goal pose in these two cases. Then the end effector will reach that same position and orientation. 

Can you render the goal frame, gripper frame and tool0 frame in the two cases?

Fzaero, 2024-03-12:
I think I found the issue. It was on my side. Sorry about this.