NVlabs/curobo #198: AttributeError: 'NoneType' object has no attribute 'GetPath'

starnetB, 2024-03-23:
cuRobo installation mode: isaac sim
version: cuda11.7 and isaacsim2022.2.1; cuda11.8 and isaacsim2023.1.1

After setup environment and run example/isaacsim scripts, error occured.

[image]

starnetB, 2024-03-23:
isaacsim2023 pip list:
[image]
[image]
[image]
[image]

balakumar-s, 2024-03-23:
Does the example work if you pass `--robot ur5e.yml`?

starnetB, 2024-03-24:
> Does the example work if you pass `--robot ur5e.yml`?

[image]

balakumar-s, 2024-03-24:
Did you pip install `usd-core`? There will be one already in the isaac sim environment. If you installed a new version, that can cause issues.

starnetB, 2024-03-25:
> Did you pip install `usd-core`? There will be one already in the isaac sim environment. If you installed a new version, that can cause issues.
> 
> In addition, is this a remote machine or local machine?

yes, when i install curobo, it said that 'usd-core' missed, so i installed a new one
It's local machine i think

balakumar-s, 2024-03-25:
Can you uninstall with `pip uninstall usd-core`?

starnetB, 2024-03-25:
> Can you uninstall with `pip uninstall usd-core`?

im reinstalling isaacsim now, this time it didnt show any message of usd-core. Maybe for the reason of ros environment (this time, i remove ros activate command in ~/.bashrc

starnetB, 2024-03-25:
[image]
run with omni_python example/isaac_sim/motion_gen_reacher.py --robot ur5e.yml

starnetB, 2024-03-25:
I tried install curobo without isaacsim and ran test code, maybe it will give some tips:

[image]
[image]

starnetB, 2024-03-25:
Idk the details, but it's actually a network problem

closed 2024-03-25 by starnetB

chiawenwang, 2024-07-20:
> Idk the details, but it's actually a network problem

Hi! I got the same problem about `'NoneType' object has no attribute 'GetPath'`. How did you fix it?

MiangChen, 2025-03-19:
> > Idk the details, but it's actually a network problem
> 
> Hi! I got the same problem about `'NoneType' object has no attribute 'GetPath'`. How did you fix it?

This may be helpful: https://github.com/isaac-sim/IsaacLab/issues/1814#issuecomment-2661293571

explore-code-world7, 2025-05-21:
> Idk the details, it's actually a network problem

Yes, I print the prim_path and config, 
then the terrain was imported rightly.

[image]