NVlabs/curobo #199: Loading problem when using CuRobo with Isaac Sim

Follograph, 2024-03-24:
1. cuRobo installation mode : isaac sim
2. Isaac Sim version : 2022.2.1 
------
Hello, when I run the given examples like simple_stacking.py, the simulation will be stuck in a few steps and the GUI cannot respond. After waiting for a period, warnings below will occur.  It seems due to loading the Isaac assets (beta) slowly. 
```
2024-03-21 00:37:29 [194,120ms] [Warning] [omni.client.plugin]  HTTP Client: provider_http: CC-493: Request through cache failed. Retrying without cache for http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/O3dyn/Parts/.thumbs/256x256/JFD_JFD_1200.usd.png
2024-03-21 00:37:29 [194,120ms] [Warning] [omni.client.plugin]  HTTP Client: omniclient: CC-873: Bypassing cache until the application is restarted
2024-03-21 00:37:29 [194,120ms] [Warning] [omni.client.plugin]  HTTP Client: provider_http: CC-493: Request through cache failed. Retrying without cache for http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/O3dyn/Parts/.thumbs/256x256/JFD_JFD_1147.usd.png
2024-03-21 00:37:29 [194,120ms] [Warning] [omni.client.plugin]  HTTP Client: provider_http: CC-493: Request through cache failed. Retrying without cache for http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/Robotiq/2F-C2/.thumbs/256x256/2fc2_instanceable.usd.png
2024-03-21 00:37:29 [194,120ms] [Warning] [omni.client.plugin]  HTTP Client: provider_http: CC-493: Request through cache failed. Retrying without cache for http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/O3dyn/Parts/.thumbs/256x256/JFD_JFD_1179.usd.png
2024-03-21 00:37:29 [194,120ms] [Warning] [omni.client.plugin]  HTTP Client: provider_http: CC-493: Request through cache failed. Retrying without cache for http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/Robotiq/2F-140/.thumbs/256x256/2f140_instanceable.usd.png
2024-03-21 00:37:30 [195,119ms] [Warning] [omni.client.plugin]  HTTP Client: provider_http: CC-493: Request through cache failed. Retrying without cache for http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/Robotiq/2F-85/.thumbs/256x256/2f85_instanceable.usd.png
2024-03-21 00:38:46 [270,887ms] [Warning] [omni.client.plugin]  HTTP Client: provider_http: CC-493: Request through cache failed. Retrying without cache for http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/Robotiq/2F-85/.thumbs/256x256/2f85_instanceable.usd.png
2024-03-21 00:39:31 [316,285ms] [Error] [gpu.foundation.plugin] Failed to read texture file http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/Robotiq/2F-85/.thumbs/256x256/2f85_instanceable.usd.png or file is empty
2024-03-21 00:39:31 [316,285ms] [Error] [gpu.foundation.plugin] Failed to load texture http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/Robotiq/2F-85/.thumbs/256x256/2f85_instanceable.usd.png (reason: 3)
2024-03-21 00:39:31 [316,285ms] [Warning] [omni.ui] Failed to upload UI Image 'http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/Robotiq/2F-85/.thumbs/256x256/2f85_instanceable.usd.png'
2024-03-21 00:40:31 [375,718ms] [Warning] [omni.client.plugin]  HTTP Client: provider_http: CC-493: Request through cache failed. Retrying without cache for http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/O3dyn/Parts/.thumbs/256x256/JFD_JFD_1147.usd.png
2024-03-21 00:40:31 [375,718ms] [Warning] [omni.client.plugin]  HTTP Client: provider_http: CC-493: Request through cache failed. Retrying without cache for http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/Robotiq/2F-140/.thumbs/256x256/2f140_instanceable.usd.png
2024-03-21 00:40:31 [375,718ms] [Warning] [omni.client.plugin]  HTTP Client: provider_http: CC-493: Request through cache failed. Retrying without cache for http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/O3dyn/Parts/.thumbs/256x256/JFD_JFD_1200.usd.png
2024-03-21 00:40:31 [375,718ms] [Warning] [omni.client.plugin]  HTTP Client: provider_http: CC-493: Request through cache failed. Retrying without cache for http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/O3dyn/Parts/.thumbs/256x256/JFD_JFD_1179.usd.png
2024-03-21 00:40:31 [375,718ms] [Warning] [omni.client.plugin]  HTTP Client: provider_http: CC-493: Request through cache failed. Retrying without cache for http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/Robotiq/2F-C2/.thumbs/256x256/2fc2_instanceable.usd.png
2024-03-21 00:41:27 [431,996ms] [Error] [gpu.foundation.plugin] Failed to read texture file http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/Robotiq/2F-C2/.thumbs/256x256/2fc2_instanceable.usd.png or file is empty
2024-03-21 00:41:27 [431,996ms] [Error] [gpu.foundation.plugin] Failed to read texture file http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/O3dyn/Parts/.thumbs/256x256/JFD_JFD_1179.usd.png or file is empty
2024-03-21 00:41:27 [431,996ms] [Error] [gpu.foundation.plugin] Failed to read texture file http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/O3dyn/Parts/.thumbs/256x256/JFD_JFD_1200.usd.png or file is empty
2024-03-21 00:41:27 [431,996ms] [Error] [gpu.foundation.plugin] Failed to read texture file http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/O3dyn/Parts/.thumbs/256x256/JFD_JFD_1147.usd.png or file is empty
2024-03-21 00:41:27 [431,996ms] [Error] [gpu.foundation.plugin] Failed to read texture file http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/Robotiq/2F-140/.thumbs/256x256/2f140_instanceable.usd.png or file is empty
2024-03-21 00:41:27 [431,996ms] [Error] [gpu.foundation.plugin] Failed to load texture http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/Robotiq/2F-C2/.thumbs/256x256/2fc2_instanceable.usd.png (reason: 3)
2024-03-21 00:41:27 [431,996ms] [Error] [gpu.foundation.plugin] Failed to load texture http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/O3dyn/Parts/.thumbs/256x256/JFD_JFD_1147.usd.png (reason: 3)
2024-03-21 00:41:27 [431,996ms] [Error] [gpu.foundation.plugin] Failed to load texture http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/O3dyn/Parts/.thumbs/256x256/JFD_JFD_1200.usd.png (reason: 3)
2024-03-21 00:41:27 [431,996ms] [Error] [gpu.foundation.plugin] Failed to load texture http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/O3dyn/Parts/.thumbs/256x256/JFD_JFD_1179.usd.png (reason: 3)
2024-03-21 00:41:27 [431,996ms] [Error] [gpu.foundation.plugin] Failed to load texture http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.2.1/Isaac/Robots/Robotiq/2F-140/.thumbs/256x256/2f140_instanceable.usd.png (reason: 3)
```
[image]

The problem had never occurred until last week. I had done nothing except running the scripts a little frequently. Do you know where the problem lies and whether it is feasible not to load the Isaac assets somehow?

Thanks!

balakumar-s, 2024-03-24:
Are you loading the robot from a local urdf?

Also does this happen when you use isaac sim from the launcher without cuRobo?

Follograph, 2024-03-24:
I load the robot from the local urdf. 

When using without cuRobo, the isaac sim will not load Isaac assets. But it actually loads saved .usd file slowly and the problem occurs the same day. The dev of Isaac Sim suggests updating to 2023.1.1, while cuRobo has not fully supported this version. So I wonder whether it is feasible to avoid loading the Isaac assets, as it is unnecessary for me.

Follograph, 2024-03-30:
update: The function add_extensions in helper.py loads the isaac assets. After I comment out the "omni.isaac.asset_browser", it returns to normal.

closed 2024-03-30 by Follograph