NVlabs/curobo #201: Ubuntu 20.04 with cuda 11.8 kernal failing to install and crashing after the installation is done.

MayurHulke, 2024-03-26:
The installation failed quite few times and I manged get it insattled now but its crashing sometimes for curobo even after following instructions provided by documentation. Also is Isaac sim supposed to take liek 20 min just to warm and even after that its crashing ? 

Is it possible to get a exact system specs and also the kernal specs on wihch the `curobo` has been tested ? It will be super helpful.  

Here is my system specs: 
1. cuRobo installation mode-  isaac sim
2. python version: 3.11.77
3. Isaac Sim version : 2022.2.1
4. cuda - 11.8
5. nvidia-smi - 535.161.07
6. PyTorch - 2.2.1+cu121
7. GPU - NVIDIA GeForce RTX 3080 

```
(base) mayur@mayur:~$ nvidia-smi
Tue Mar 26 16:32:57 2024       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.161.07             Driver Version: 535.161.07   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 3080 ...    Off | 00000000:01:00.0 Off |                  N/A |
| N/A   50C    P5              24W /  35W |    388MiB / 16384MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      1898      G   /usr/lib/xorg/Xorg                            4MiB |
|    0   N/A  N/A      2437      G   /usr/lib/xorg/Xorg                           28MiB |
|    0   N/A  N/A      4649    C+G   ...e/ov/pkg/isaac_sim-2022.2.1/kit/kit      332MiB |
+---------------------------------------------------------------------------------------+

```
```
(base) mayur@mayur:~$ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Wed_Sep_21_10:33:58_PDT_2022
Cuda compilation tools, release 11.8, V11.8.89
Build cuda_11.8.r11.8/compiler.31833905_0
```
```
(base) mayur@mayur:~$ python
Python 3.11.7 (main, Dec 15 2023, 18:12:31) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
>>> print(torch.__version__)
2.2.1+cu121
```
[image]
[image]

balakumar-s, 2024-03-26:
1. Isaac sim will compile shaders on first launch which can take 2-3 minutes.
2. How are you using python 3.11 with isaac sim 2022.2.1? Isaac sim 2022.2.1 supports python 3.7. Did you follow instructions here to setup conda env: https://docs.omniverse.nvidia.com/isaacsim/latest/installation/install_python.html#advanced-running-with-anaconda

MayurHulke, 2024-03-26:
Oh thats strange because Its working for me even for python 3.11

[image]

balakumar-s, 2024-03-26:
The environment.yml file is located in the isaac sim path: `~/.local/share/ov/pkg/isaac_sim-2022.2.1/`

MayurHulke, 2024-03-26:
This is great, It is working way smoother than before 👍🏼 . Thanks a lot @balakumar-s.  Just one last question may I know what GPU were you using and RAM for like ideal perfromance for hardcore dev work (testing, experimentation etc ).

balakumar-s, 2024-03-26:
Isaac Sim requirements should also be more than enough for cuRobo: https://docs.omniverse.nvidia.com/isaacsim/latest/installation/requirements.html#system-requirements