NVlabs/curobo #205: Constrained Planning - High physics_dt crashes everything - Isaac 2023.1.1

eferreirafilho, 2024-04-01:
1. cuRobo installation mode: isaac sim
2. python version: 3.10.12
3. Isaac Sim version (if using): 2023.1.1 

------
Issue Details

High Physics_dt crashes everything, it was working fine in Isaac 2022.

When adding this line:
my_world = World(stage_units_in_meters=1.0, physics_dt=1/180)
to isaac_sim/constrained_reacher.py with an ur5e robot:

`2024-04-01 12:56:35 [80,240ms] [Warning] [omni.isaac.core.articulations.articulation_view] ArticulationView needs to be initialized.
2024-04-01 12:56:35 [80,240ms] [Warning] [omni.isaac.core.articulations.articulation_view] Physics Simulation View is not created yet in order to use get_joints_state
2024-04-01 12:56:35 [80,240ms] [Warning] [omni.isaac.core.articulations.articulation_view] ArticulationView needs to be initialized.
sim_js: None
sim_js_names: None
Traceback (most recent call last):
  File "/home/edson/curobo/examples/isaac_sim/constrained_reacher_ur5e.py", line 452, in <module>
    position=tensor_args.to_device(sim_js.positions),
AttributeError: 'NoneType' object has no attribute 'positions'`

It works fine with physics_dt=1/60 or 1/120

balakumar-s, 2024-04-01:
Looks like this is an issue with Isaac Sim. 
Can you create an issue here https://forums.developer.nvidia.com/c/omniverse/simulation/69 so that Isaac Sim experts can help?

balakumar-s, 2025-04-25:
This has been fixed in the latest release (v0.7.7)

closed 2025-04-25 by balakumar-s

eferreirafilho, 2025-04-25:
Keep up the great work @balakumar-s !