NVlabs/curobo #211: Potential issues with the latest release

shannonwarren, 2024-04-06:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): isaac sim
2. python version:
3. Isaac Sim version (if using): 2022.2.1

------
Issue Details

I'm currently evaluating cuRobo's planner and noticed a performance degradation with the latest release. I used the plan_single_js api on the feb 23rd commit and it works perfectly but it with the latest release it give me an error saying.
`Plan did not converge to a solution.  No action is being taken.` .  My planconfigs were not modified too and my world has no obstacles.

>  goal_js = JointState(
                    position=tensor_args.to_device([-1.588, -1.633, -2.104, -2.324, 1.661, 3.107]),
                    velocity=tensor_args.to_device(sim_js.velocities) * 0.0,
                    acceleration=tensor_args.to_device(sim_js.velocities) * 0.0,
                    jerk=tensor_args.to_device(sim_js.velocities) * 0.0,
                    joint_names=sim_js_names,
                )
                goal_js = goal_js.get_ordered_joint_state(motion_gen.kinematics.joint_names)
                print("Planning to 'home' position")
                print(sim_js.positions)
                result = motion_gen.plan_single_js(
                    cu_js.unsqueeze(0), goal_js.unsqueeze(0), plan_config)
 

>  plan_config = MotionGenPlanConfig(
        enable_graph=False, enable_graph_attempt=8, max_attempts=20, enable_finetune_trajopt=False
    )

shannonwarren, 2024-04-06:
Additionally, do you have any suggestions on how to debug when there is no plan generated? I know it could it be a resource limitation, MotionGenConfig or PlanConfig or just the collision environment. Is there a way of knowing what's causing this? Thank you

balakumar-s, 2024-04-06:
Can you share the start_js as well? And is it for the Franka Panda robot?

We have a logger that will create an animated usd of the solver iterations. Though we haven't tested with plan_single_js

Here is an example: https://github.com/NVlabs/curobo/blob/774dcfd609847ad98147f166ea2f5954516ff7a0/examples/usd_example.py#L168

shannonwarren, 2024-04-08:
I'm using a ur5e robot in sim. I've made some changes to constraint_reacher.py to use both plan_single_js and plan_single 
```
import torch

a = torch.zeros(4, device="cuda:0")
from omni.isaac.kit import SimulationApp

simulation_app = SimulationApp(
    {
        "headless": False,
        "width": "1920",
        "height": "1080",
    }
)
import numpy as np
import torch
from helper import add_extensions, add_robot_to_scene

from curobo.geom.sdf.world import CollisionCheckerType
from curobo.geom.types import Cuboid, WorldConfig
from curobo.types.base import TensorDeviceType
from curobo.types.math import Pose
from curobo.types.robot import JointState, RobotConfig
from curobo.types.state import JointState
from curobo.util_file import get_robot_configs_path, get_world_configs_path, join_path, load_yaml
from curobo.wrap.model.robot_world import RobotWorld, RobotWorldConfig
from curobo.wrap.reacher.motion_gen import (
    MotionGen,
    MotionGenConfig,
    MotionGenPlanConfig,
    PoseCostMetric,
)

simulation_app.update()
import argparse

import carb
from omni.isaac.core import World
from omni.isaac.core.materials import OmniGlass, OmniPBR
from omni.isaac.core.objects import cuboid, sphere
from omni.isaac.core.utils.types import ArticulationAction

from curobo.util.usd_helper import UsdHelper

parser = argparse.ArgumentParser()

parser.add_argument("--robot", type=str, default="ur5e.yml", help="robot configuration to load")
parser.add_argument(
    "--visualize_spheres",
    action="store_true",
    help="When True, visualizes robot spheres",
    default=True,
)

args = parser.parse_args()

def create_environment():
    # world_cfg1 = 
    pass

if __name__ == "__main__":
    my_world = World(stage_units_in_meters=1.0)
    stage = my_world.stage
    n_obstacle_cuboids = 10
    n_obstacle_mesh = 10

    stage = my_world.stage
    # my_world.scene.add_default_ground_plane()

    xform = stage.DefinePrim("/World", "Xform")
[... 395 lines cut ...]
        if cmd_plan is not None:
            cmd_state = cmd_plan[cmd_idx]

            # get full dof state
            art_action = ArticulationAction(
                cmd_state.position.cpu().numpy(),
                cmd_state.velocity.cpu().numpy(),
                joint_indices=idx_list,
            )
            # set desired joint angles obtained from IK:
            articulation_controller.apply_action(art_action)
            cmd_step_idx += 1
            if cmd_step_idx == 2:
                cmd_idx += 1
                cmd_step_idx = 0
            # for _ in range(2):
            #    my_world.step(render=False)
            if cmd_idx >= len(cmd_plan.position):
                cmd_idx = 0
                cmd_plan = None
    print("finished program")

    simulation_app.close()

```

shannonwarren, 2024-04-08:
This is what i have running in sim. when `target_idx == 6` I tell the robot to go home using the joint state api and it fails to plan.
[Screencast from 04-08-2024 12:56:47 PM.webm](https://github.com/NVlabs/curobo/assets/46359438/0329c8c2-1bae-40e2-88d4-827243d71537)

shannonwarren, 2024-04-08:
@balakumar-s I just tried the simple_stacking.py isaac example which has the usdhelper calling write_motion_gen_log api and switched on the save_log flag to True and i get this error.

> [7.206s] Simulation App Startup Complete
> warming up...
> Traceback (most recent call last):
>   File "examples/isaac_sim/simple_stacking.py", line 435, in <module>
>     my_world=my_world, my_task=my_task, constrain_grasp_approach=args.constrain_grasp_approach
>   File "examples/isaac_sim/simple_stacking.py", line 161, in __init__
>     self.motion_gen.warmup(parallel_finetune=True)
>   File "curobo/src/curobo/wrap/reacher/motion_gen.py", line 2202, in warmup
>     self.plan_single_js(start_state, goal_state, MotionGenPlanConfig(max_attempts=1))
>   File "curobo/src/curobo/wrap/reacher/motion_gen.py", line 2372, in plan_single_js
>     solve_state, start_state, goal_state, plan_config=plan_config
>   File "curobo/src/curobo/wrap/reacher/motion_gen.py", line 1842, in _plan_js_from_solve_state
>     result.debug_info["trajopt_result"] = traj_result
> TypeError: 'NoneType' object does not support item assignment
> 2024-04-08 20:54:16 [14,746ms] [Warning] [carb.audio.context] 1 contexts were leaked
> /home/shannon/.local/share/ov/pkg/isaac_sim-2022.2.1/python.sh: line 41: 593397 Segmentation fault      (core dumped) $python_exe "$@" $args
> There was an error running python

shannonwarren, 2024-04-09:
An update on 2) regarding navigating to the two joint states. There was a self collision in my tool joint causing the failure to plan. I reduced the sphere size to a very low value and it works reliably. I think I'll need to add this to the `self_collision_ignore`. Would be helpful to see a debug message when self collision happens.

balakumar-s, 2024-04-11:
We have joint space planner working better in the latest release. Let us know if you have issues.

closed 2024-04-11 by balakumar-s