NVlabs/curobo #212: Robot models overlapped in example batch_motion_gen_reacher

YANGHANJIANG, 2024-04-07:
1. cuRobo installation mode:isaac sim
2. python version:3.10
3. Isaac Sim version:2022.2.1
4. Cuda:11.8
5. Ubuntu:20.04

Issue Details
Robots are located at the same place when running the example batch_motion_gen_reacher.
It appears that two robots, named "panda" and "panda_0," are both located under the "World" prim in the Stage panel.
The screenshot of Isaac Sim is depicted below.
[image]

balakumar-s, 2024-04-08:
Thanks for finding this bug. We have a fix that will be pushed soon. In the meantime, to unblock you:

1. Remove this line https://github.com/NVlabs/curobo/blob/774dcfd609847ad98147f166ea2f5954516ff7a0/examples/isaac_sim/helper.py#L107 and replace it with:

```
    robot_p = Robot(
        prim_path=robot_path + "/" + base_link_name,
        name=robot_name,
    )
    from curobo.util.usd_helper import set_prim_transform
    robot_prim = robot_p.prim
    stage = robot_prim.GetStage()
    linkp = stage.GetPrimAtPath(robot_path)
    set_prim_transform(linkp, [position[0], position[1], position[2], 1, 0, 0, 0])
 ```

songlin, 2024-04-09:
@balakumar-s  I followed your modification. The robots are placed in the right positions now. However, Isaac-sim crashed when I moved the target cube.

```
[ERROR] [curobo] changing goal type, breaking previous cuda graph.
NoneType: None
Stack (most recent call last):
  File "examples/isaac_sim/batch_motion_gen_reacher.py", line 298, in <module>
    main()
  File "examples/isaac_sim/batch_motion_gen_reacher.py", line 252, in main
    result = motion_gen.plan_batch_env(full_js, ik_goal, plan_config.clone())
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1418, in plan_batch_env
    plan_config,
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1243, in _plan_batch_attempts
    solve_state, start_state, goal_pose, plan_config
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/motion_gen.py", line 2082, in _plan_from_solve_state_batch
    return_all_solutions=True,
  File "/home/songlin/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/contextlib.py", line 74, in inner
    return func(*args, **kwds)
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/motion_gen.py", line 991, in _solve_trajopt_from_solve_state
    newton_iters=newton_iters,
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/trajopt.py", line 560, in solve_any
    newton_iters=newton_iters,
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/trajopt.py", line 762, in solve_batch_env
    newton_iters=newton_iters,
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/trajopt.py", line 590, in solve_from_solve_state
    goal_buffer = self.update_goal_buffer(solve_state, goal)
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/trajopt.py", line 492, in update_goal_buffer
    log_error("changing goal type, breaking previous cuda graph.")
  File "/home/songlin/Projects/curobo/src/curobo/util/logger.py", line 44, in log_error
    logger.error(txt, exc_info=exc_info, stack_info=stack_info, *args, **kwargs)
Traceback (most recent call last):
  File "examples/isaac_sim/batch_motion_gen_reacher.py", line 298, in <module>
    main()
  File "examples/isaac_sim/batch_motion_gen_reacher.py", line 252, in main
    result = motion_gen.plan_batch_env(full_js, ik_goal, plan_config.clone())
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1418, in plan_batch_env
    plan_config,
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1243, in _plan_batch_attempts
    solve_state, start_state, goal_pose, plan_config
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/motion_gen.py", line 2082, in _plan_from_solve_state_batch
    return_all_solutions=True,
  File "/home/songlin/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/contextlib.py", line 74, in inner
    return func(*args, **kwds)
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/motion_gen.py", line 991, in _solve_trajopt_from_solve_state
    newton_iters=newton_iters,
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/trajopt.py", line 560, in solve_any
    newton_iters=newton_iters,
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/trajopt.py", line 762, in solve_batch_env
    newton_iters=newton_iters,
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/trajopt.py", line 590, in solve_from_solve_state
    goal_buffer = self.update_goal_buffer(solve_state, goal)
  File "/home/songlin/Projects/curobo/src/curobo/wrap/reacher/trajopt.py", line 492, in update_goal_buffer
    log_error("changing goal type, breaking previous cuda graph.")
  File "/home/songlin/Projects/curobo/src/curobo/util/logger.py", line 45, in log_error
    raise
RuntimeError: No active exception to reraise
/home/songlin/.local/share/ov/pkg/isaac_sim-2022.2.1/python.sh: line 41: 368592 Segmentation fault      (core dumped) $python_exe "$@" $args
There was an error running python

```
Can you help please? I am using Issac-sim 2022.2.1 and installed latest curobo commit

YANGHANJIANG, 2024-04-10:
> @balakumar-s I followed your modification. The robots are placed in the right positions now. However, Isaac-sim crashed when I moved the target cube.
>
> ```
> [ERROR] [curobo] changing goal type, [...]

Faced the same issue.

Workaround:
Replace this line
https://github.com/NVlabs/curobo/blob/774dcfd609847ad98147f166ea2f5954516ff7a0/examples/isaac_sim/batch_motion_gen_reacher.py#L149 with
`use_cuda_graph=False,`

songlin, 2024-04-10:
thanks

Best,
Songlin Wei

balakumar-s, 2024-04-11:
We fixed the robot location in latest release.

closed 2024-04-11 by balakumar-s