NVlabs/curobo #214: Error building docker image with issac sim 2023.1.0

makolon, 2024-04-09:
1. cuRobo installation mode: docker isaac sim
2. python version: 
3. Isaac Sim version (if using): 2023.1.0

------
Issue Details
Error building in docker image from bash build_docker.sh isaac_sim_2023.1.0.
```
753.5   copying src/curobo/content/configs/robot/spheres/iiwa_allegro.yml -> build/lib.linux-x86_64-cpython-310/curobo/content/configs/robot/spheres
753.5   copying src/curobo/content/configs/robot/spheres/kinova_gen3.yml -> build/lib.linux-x86_64-cpython-310/curobo/content/configs/robot/spheres
753.5   copying src/curobo/content/configs/robot/spheres/quad_ur10e.yml -> build/lib.linux-x86_64-cpython-310/curobo/content/configs/robot/spheres
753.5   copying src/curobo/content/configs/robot/spheres/single.yml -> build/lib.linux-x86_64-cpython-310/curobo/content/configs/robot/spheres
753.5   copying src/curobo/content/configs/robot/spheres/ur10e.yml -> build/lib.linux-x86_64-cpython-310/curobo/content/configs/robot/spheres
753.5   creating build/lib.linux-x86_64-cpython-310/curobo/curobolib/cpp
753.5   copying src/curobo/curobolib/cpp/geom_cuda.cpp -> build/lib.linux-x86_64-cpython-310/curobo/curobolib/cpp
753.5   copying src/curobo/curobolib/cpp/helper_math.h -> build/lib.linux-x86_64-cpython-310/curobo/curobolib/cpp
753.5   copying src/curobo/curobolib/cpp/kinematics_fused_cuda.cpp -> build/lib.linux-x86_64-cpython-310/curobo/curobolib/cpp
753.5   copying src/curobo/curobolib/cpp/kinematics_fused_kernel.cu -> build/lib.linux-x86_64-cpython-310/curobo/curobolib/cpp
753.5   copying src/curobo/curobolib/cpp/lbfgs_step_cuda.cpp -> build/lib.linux-x86_64-cpython-310/curobo/curobolib/cpp
753.5   copying src/curobo/curobolib/cpp/lbfgs_step_kernel.cu -> build/lib.linux-x86_64-cpython-310/curobo/curobolib/cpp
753.5   copying src/curobo/curobolib/cpp/line_search_cuda.cpp -> build/lib.linux-x86_64-cpython-310/curobo/curobolib/cpp
753.5   copying src/curobo/curobolib/cpp/line_search_kernel.cu -> build/lib.linux-x86_64-cpython-310/curobo/curobolib/cpp
753.5   copying src/curobo/curobolib/cpp/pose_distance_kernel.cu -> build/lib.linux-x86_64-cpython-310/curobo/curobolib/cpp
753.5   copying src/curobo/curobolib/cpp/self_collision_kernel.cu -> build/lib.linux-x86_64-cpython-310/curobo/curobolib/cpp
753.5   copying src/curobo/curobolib/cpp/sphere_obb_kernel.cu -> build/lib.linux-x86_64-cpython-310/curobo/curobolib/cpp
753.5   copying src/curobo/curobolib/cpp/tensor_step_cuda.cpp -> build/lib.linux-x86_64-cpython-310/curobo/curobolib/cpp
753.5   copying src/curobo/curobolib/cpp/tensor_step_kernel.cu -> build/lib.linux-x86_64-cpython-310/curobo/curobolib/cpp
753.5   copying src/curobo/curobolib/cpp/update_best_kernel.cu -> build/lib.linux-x86_64-cpython-310/curobo/curobolib/cpp
753.5   running build_ext
753.5   /isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/utils/cpp_extension.py:388: UserWarning: The detected CUDA version (11.4) has a minor version mismatch with the version that was used to compile PyTorch (11.8). Most likely this shouldn't be a problem.
753.5     warnings.warn(CUDA_MISMATCH_WARN.format(cuda_str_version, torch.version.cuda))
753.5   building 'curobo.curobolib.lbfgs_step_cu' extension
753.5   creating build/temp.linux-x86_64-cpython-310
753.5   creating build/temp.linux-x86_64-cpython-310/src
753.5   creating build/temp.linux-x86_64-cpython-310/src/curobo
753.5   creating build/temp.linux-x86_64-cpython-310/src/curobo/curobolib
753.5   creating build/temp.linux-x86_64-cpython-310/src/curobo/curobolib/cpp
753.5   gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include -I/isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include/torch/csrc/api/include -I/isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include/TH -I/isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include/THC -I/usr/local/cuda/include -I/isaac-sim/kit/python/include/python3.10 -c src/curobo/curobolib/cpp/lbfgs_step_cuda.cpp -o build/temp.linux-x86_64-cpython-310/src/curobo/curobolib/cpp/lbfgs_step_cuda.o -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE=\"_gcc\" -DPYBIND11_STDLIB=\"_libstdcpp\" -DPYBIND11_BUILD_ABI=\"_cxxabi1011\" -DTORCH_EXTENSION_NAME=lbfgs_step_cu -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++17
753.5   /usr/local/cuda/bin/nvcc -I/isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include -I/isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include/torch/csrc/api/include -I/isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include/TH -I/isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include/THC -I/usr/local/cuda/include -I/isaac-sim/kit/python/include/python3.10 -c src/curobo/curobolib/cpp/lbfgs_step_kernel.cu -o build/temp.linux-x86_64-cpython-310/src/curobo/curobolib/cpp/lbfgs_step_kernel.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options '-fPIC' --threads=8 -O3 --ftz=true --fmad=true --prec-div=false --prec-sqrt=false -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE=\"_gcc\" -DPYBIND11_STDLIB=\"_libstdcpp\" -DPYBIND11_BUILD_ABI=\"_cxxabi1011\" -DTORCH_EXTENSION_NAME=lbfgs_step_cu -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_70,code=compute_70 -gencode=arch=compute_70,code=sm_70 -std=c++17
753.5   /isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include/c10/util/irange.h(54): warning: pointless comparison of unsigned integer with zero
753.5             detected during:
753.5               instantiation of "__nv_bool c10::detail::integer_iterator<I, one_sided, <unnamed>>::operator==(const c10::detail::integer_iterator<I, one_sided, <unnamed>> &) const [with I=size_t, one_sided=false, <unnamed>=0]"
753.5   (61): here
753.5               instantiation of "__nv_bool c10::detail::integer_iterator<I, one_sided, <unnamed>>::operator!=(const c10::detail::integer_iterator<I, one_sided, <unnamed>> &) const [with I=size_t, one_sided=false, <unnamed>=0]"
753.5   /isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include/c10/core/TensorImpl.h(77): here
753.5   
753.5   /isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include/c10/util/irange.h(54): warning: pointless comparison of unsigned integer with zero
753.5             detected during:
753.5               instantiation of "__nv_bool c10::detail::integer_iterator<I, one_sided, <unnamed>>::operator==(const c10::detail::integer_iterator<I, one_sided, <unnamed>> &) const [with I=std::size_t, one_sided=true, <unnamed>=0]"
753.5   (61): here
753.5               instantiation of "__nv_bool c10::detail::integer_iterator<I, one_sided, <unnamed>>::operator!=(const c10::detail::integer_iterator<I, one_sided, <unnamed>> &) const [with I=std::size_t, one_sided=true, <unnamed>=0]"
753.5   /isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include/ATen/core/qualified_name.h(73): here
753.5   
753.5   g++ -pthread -shared -Wl,-rpath=$ORIGIN/../lib -L/builds/omniverse/externals/python-build/_build/target-deps/openssl/lib -L/builds/omniverse/externals/python-build/_build/target-deps/sqlite/lib -L/builds/omniverse/externals/python-build/_build/target-deps/libffi/lib build/temp.linux-x86_64-cpython-310/src/curobo/curobolib/cpp/lbfgs_step_cuda.o build/temp.linux-x86_64-cpython-310/src/curobo/curobolib/cpp/lbfgs_step_kernel.o -L/isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/lib -L/usr/local/cuda/lib64 -L/isaac-sim/kit/python/lib -lc10 -ltorch -ltorch_cpu -ltorch_python -lcudart -lc10_cuda -ltorch_cuda -o build/lib.linux-x86_64-cpython-310/curobo/curobolib/lbfgs_step_cu.cpython-310-x86_64-linux-gnu.so
753.5   building 'curobo.curobolib.kinematics_fused_cu' extension
753.5   gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include -I/isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include/torch/csrc/api/include -I/isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include/TH -I/isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/include/THC -I/usr/local/cuda/i [... cut ...]
754.0 You should consider upgrading via the '/isaac-sim/kit/python/bin/python3 -m pip install --upgrade pip' command.
754.4 There was an error running python
------
isaac_sim.dockerfile:179
--------------------
 177 |     
 178 |     
 179 | >>> RUN cd /pkgs/curobo && $omni_python -m pip install .[dev] --no-build-isolation
 180 |     
 181 |     WORKDIR /pkgs/curobo
--------------------
ERROR: failed to solve: process "/bin/sh -c cd /pkgs/curobo && $omni_python -m pip install .[dev] --no-build-isolation" did not complete successfully: exit code: 1
```

balakumar-s, 2024-04-10:
Thanks for reporting the issue. We are able to reproduce the issue. We will post an update once we have a fix.

balakumar-s, 2024-04-11:
We fixed in the latest release. Open this again if it doesn't work.

closed 2024-04-11 by balakumar-s